Kimplul/hid-tmff2 #5: No FFB for me

spocthier83, 2020-09-19:
Hello! I was about to sell my T300RS when I found this project.

I have successfully built and installed both modules in my Linux Mint 20. But, unfortunately, I have no FFB in any game (Assetto Corsa Competizione, Wrekfest, ...).

In other hand, _Oversteer_ software finds successfully the plugged T300RS with theses modules.

What are the requirements to make FFB work?
 
```
$ uname -a
Linux spocthier-PC 5.4.0-47-generic #51-Ubuntu SMP Fri Sep 4 19:50:52 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
And the last firmware is installed (v31).

I don't know what to say more to help for debugging.

Thx for this!

Kimplul, 2020-09-19:
Heyo, I'm guessing you've unplugged and plugged the wheel back in and possibly restarted your computer?

The wheel's FFB for some godawful reason initializes itself when a certain USBHID request is sent out. On Linux, this request isn't necessarily always sent out because of some driver/device data caching, I believe, so a reboot is usually needed. Windows seems to work somewhat differently, so I guess Thrustmaster didn't think of making the wheel initialize in a sane way.

spocthier83, 2020-09-19:
Indeed, I have rebooted my computer many times since installation. I followed the instructions of course.

In addition, I had to generate certificates (private and public keys) to sign modules in kernel and find the file System.map beforehand to install modules.

Is there a little tool to test FFB directly (i.e. without game)?

Kimplul, 2020-09-19:
Yeah, fftest, should be in the the package joystick. You use it with the command 
`fftest /dev/input/by-id/usb-Thrustmaster_Thrustmaster_T300RS_Racing_wheel-event-joystick`.

Effects 0-3 are currently supported, the last two I haven't really figured out yet.

Right, sort of interesting, I have the newest firmware and I recompiled the module from GitHub. Works on my machine™.

Anycase, could you paste the output in dmesg you get when plugging the wheel in? Also, please include the output of 
`cat /sys/kernel/debug/hid/0003\:044F\:B66E.0005/rdesc`
You'll probably need superuser privileges.

In addition to that, could you explain in detail what the steps you took to compile the kernel were?

For Ubuntu, which you seem to be running, the whole process for me was
+ `sudo apt install linux-$(uname -r)-headers`
+ `make`
+ `sudo make install`
+ `sudo depmod -a`
+  `sudo modprobe hid-tminit`
+ `sudo modprobe hid-tmt300rs`
And by this point I plugged in my wheel and it started working. The last three commands usually aren't necessary, but if any errors occur it'll be more obvious where and what might've gone wrong.

spocthier83, 2020-09-19:
Thx for your help.

1) Unfortunately, nothing happens (no FFB) using _fftest_ :
```
$ fftest /dev/input/by-id/usb-Thrustmaster_Thrustmaster_T300RS_Racing_wheel-event-joystick 
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/by-id/usb-Thrustmaster_Thrustmaster_T300RS_Racing_wheel-event-joystick opened
Features:
  * Absolute axes: X, Y, Z, RZ, Hat 0 X, Hat 0 Y, 
    [27 00 03 00 00 00 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: Constant, Periodic, Ramp, Spring, Friction, Damper, Rumble, Inertia, Gain, Autocenter, 
    Force feedback periodic effects: Square, Triangle, Sine, Saw up, Saw down, 
    [00 00 00 00 00 00 00 00 00 00 FF 1F 03 00 00 00 ]
  * Number of simultaneous effects: 16

Setting master gain to 75% ... OK
Uploading effect #0 (Periodic sinusoidal) ... OK (id 0)
Uploading effect #1 (Constant) ... OK (id 1)
Uploading effect #2 (Spring) ... OK (id 2)
Uploading effect #3 (Damper) ... OK (id 3)
Uploading effect #4 (Strong rumble, with heavy motor) ... OK (id 4)
Uploading effect #5 (Weak rumble, with light motor) ... OK (id 5)
Enter effect number, -1 to exit
0
Now Playing: Sine vibration
Enter effect number, -1 to exit
1
Now Playing: Constant Force
Enter effect number, -1 to exit
3
Now Playing: Damping Condition
Enter effect number, -1 to exit
```

2) The output of /sys/kernel/debug/hid/0003:044F:B66E.000A/rdesc is :
[0003:044F:B66E.000A_rdesc.log](https://github.com/Kimplul/hid-tmff2/files/5250615/0003.044F.B66E.000A_rdesc.log)
Note that the file name you describe is not exactly the same as mine.

3) Endly, how I have to built and install your software :
* built :
```
$ git clone https://github.com/Kimplul/hid-tmff2.git
$ cd hid-tmff2
$ make
```
* install :
```
$ sudo openssl req -verify -new -nodes -utf8 -sha512 -days 36500 -batch -x509 -config ~/x509.genkey -outform PEM -out /usr/src/linux-headers-5.4.0-47-generic/certs/signing_key.x509 -keyout /usr/src/linux-headers-5.4.0-47-generic/certs/signing_key.pem
$ sudo cp /boot/System.map-5.4.0-47-generic /usr/src/linux-headers-5.4.0-47-generic/System.map
$ sudo make install
```
At first glance, my system needs signed modules to be inserted into kernel (installation fail otherwise).

Thx a lot for your help!

Kimplul, 2020-09-19:
Are you using SELinux or something similar? You shouldn't have to generate certs or anything like that to insert a kernel module. That being said, certs shouldn't affect the behaviour of the module in any way, so it might be a red herring.

Thanks for the rdesc, yeah, sorry, the path can be a bit different, I should've mentioned that. Could you still show the output of `dmesg` when you insert the wheel?

Also, just to be on the safe side, you do have your wheel set to ps3?

berarma, 2020-09-19:
> $ sudo openssl req -verify -new -nodes -utf8 -sha512 -days 36500 -batch -x509 -config ~/x509.genkey -outform PEM -out /usr/src/linux-headers-5.4.0-47-generic/certs/signing_key.x509 -keyout [...]

These steps shouldn't be required. You may see a warning about the module not being signed but that's not an issue as far as I know. That message can be ignored.

The complete output of the commands you run could help on the diagnosing. Running `script mycommands` will write all your following console output to a file named mycommands until you type `exit`. Attaching this file would provide us with all that is happening in the process of building and installing the module.

The `dmesg` output after the module is loaded and the device connected has useful information too.

spocthier83, 2020-09-20:
I use Linux Mint 20 "out of box", I don't know if it is a SELinux. Maybe yes.
But I agree with you signing module or not shouldn't be the problem. But I was thinking it was necessary at this moment because it doesn't work. In the other hand, I have now no warning at installation. It's a good thing.

Here are my commands and the output :
[commands.log](https://github.com/Kimplul/hid-tmff2/files/5252152/commands.log)
There is nothing crazy.

And _dmesg_ output with T300RS plugged :
[dmesg.log](https://github.com/Kimplul/hid-tmff2/files/5252167/dmesg.log)

Hope it will help.

And yes, it is on PS3 side.

Kimplul, 2020-09-20:
Very strange. I just downloaded Mint to test it out and everything works just fine ootb. I did get the same warnings that you also got, but they were just warnings, and I could load the driver just fine. In addition, there's nothing weird or abnormal with the logs you've provided, at least as far as I can see.

I'm just going to fire off some suggestions that may or may not have anything to do with the issue.

+ USB ports. Try all of them. At least on Windows I remember having issues with the wheel whenever I plugged it into a USB2 port, and even their customer service recommended I stick to USB3 if possible. So far I haven't experienced similar issues on Linux, but it's probably worth trying it out.

+ Not a standard T300RS. I know that Thrustmaster has produced some 'special' versions of the T300RS, but I've understood that they all should use the same servo/base, so this might be a longshot. If you happen to have a Windows install nearby, could you show us what Thrustmaster's 'Control Panel' tells you about the wheel? Below is how it looks on my machine.
[image]

+ Hardware issue. When did you last use the wheel? Could it have become damaged since you last used it, etc? If you can, please check if the wheel works on Windows.

spocthier83, 2020-09-20:
The wheel works well under Windows. The forces tests are very good.
[image]

As you recommend, I have tried other USB plug too without improvement.

Maybe a difference : I don't plug any pedals to the wheel directly because I use an external controller for my old G25 pedals and shifter (which is not actually connected during my tests).

My T300RS is the only thing I miss from my former Windows life.

Kimplul, 2020-09-21:
The wheel should work with or without pedals just fine, at least it has for me.

Has everything else worked like it should? Can you set the wheel's rotation, the wheel is frictionless when opened in oversteer, but autocentering comes on when you close oversteer?

Kimplul, 2020-09-21:
Quick update, I bought Wreckfest to test out and that game does actually somehow manage to nullify the FFB, even in other games. I'll try to figure this out as soon as possible. I don't know if this is the whole issue or even related, but on the surface it seems like it.

Kimplul, 2020-09-21:
Okay, I pushed a fix for F1 2018 to the branch ramp, here
https://github.com/Kimplul/hid-tmff2/tree/ramp

Now F1 works like it should as does Wreckfest, at least on my machine. I still have no clue if this is even related to your original issue, but what the hey, a bugfix is better than nothing. Give the branch a try and I'll keep my fingers crossed.

spocthier83, 2020-09-21:
> Has everything else worked like it should? Can you set the wheel's rotation, the wheel is frictionless when opened in oversteer, but autocentering comes on when you close oversteer?

To reply to this, when I first plug the wheel (__after__ a fresh boot) :
* the wheel doesn't calibrate it-self (but it moves a very little hieratically)
* is "hard" to rotate manually, I fell resistance and "notches/steps".
* it doesn't auto-recenter after manually rotating,
* Oversteer displays no rotation and no button pressed

If I plug __and__ (re)boot computer the wheel :
* the wheel auto-calibrate automatically (as nominal case)
* the wheel now auto center itself when I rotate it manually (as nominal case).
* Oversteer displays rotation and buttons pressed (as nominal case)
* the range limitation doesn't seem to work
* but no FFB other than auto-recentering

And the wheel works like a charm under Windows.

I have built/reinstalled both modules without signing them, but it is exactly the same situation.
You are right : signing module or not doesn't change anything.

I really want to thank you for your support.

Kimplul, 2020-09-21:
Right, thanks for your patience. When the wheel goes "hard", it usually means that the wheel has crashed. It happened to me a couple of times when I was testing my wheel in games, but never when plugging it in, so I'm unfamiliar with this issue.

You could always try https://github.com/scarburato/hid-tminit, which is supposed to be a more generic Thrustmaster initialization module that @scarburato wrote a little while back. It just replaces the init that I wrote for this module. It's installed the same way this module is, just write `make` and `sudo make install`. If the wheel crashes during initialization, it could be an issue with my init, and this more generic init might make the wheel initialize like it should. A shot in the dark, but I'm sort of at my wits' end here.

As I've probably mentioned several times, everything works just fine on my end, and I can't for the life of me replicate your issues, so I have no real idea what the issue could be. Without your setup and possibly your wheel, debugging becomes incredibly hard and tedious, if not impossible.

spocthier83, 2020-09-23:
Hello,

I give up for now. I have no much time to spent to go deeper. Maybe some feedback from others users may help later.

I thank you for your help and to work on such a kind of project.

Kimplul, 2020-09-23:
I understand. Thank you for your patience.

Kimplul, 2020-11-06:
Hey, do you still have your wheel? We fixed an issue that migth've been messing up your wheel too.

spocthier83, 2020-11-07:
Guy, you rock! It works like a charm!
It's awesome to feel my T300RS again.

Now, i can sell my G29 little toy.

closed 2020-11-07 by spocthier83

Kimplul, 2020-11-07:
Glad it works. Obviously thanks to @dafteran4, @berarma and @scarburato as well.

mercster, 2025-07-14:
Unfortunately I am also seeing this behavior with my newly purchased T300RS.  Oversteer can change things on the wheel, but I enter a game (American Truck Simulator)... I have no effects, just autocenter.  fftest can do the 1, 4, and 5 "rumble" type effects, but the other ones do nothing.  (Again, oversteer CAN change force feedback strength, autocenter, etc.)  Even turning off FF in ATS completely, no change, just slightly stiff wheel with strong autocenter.

When I boot, hit-tminit-new and usb-tminit-new do not get loaded, only hid-tmff-new.  Not sure why.. also I have seen no mention from anyone of the usb-tminit-new module that gets built.. which of all these should be loaded and in what order?