Kimplul/hid-tmff2 #6: BUG: scheduling while atomic

dafteran4, 2020-10-31:
First, thanks for your work on this. It's much appreciated.

I am on arch linux, kernel Linux 5.9.1-arch1-1 #1 SMP PREEMPT.

I haven't managed to get force feedback working. Attached is the dmesg log when plugging in the wheel. There are some errors:
```
[   12.988878] BUG: scheduling while atomic: systemd-udevd/757/0x00000002
...
[   12.988968] Preemption disabled at:
[   12.988974] [<0000000000000000>] 0x0
...
[   12.988986] Call Trace:
[   12.988999]  dump_stack+0x6b/0x88
```
After plugging in, the wheel does its calibration thing and after that force feedback works to keep the wheel centered.

Oversteer shows working axes and buttons, but the force feedback setting seem to have no effect. fftest reports OK for all effects, but nothing happens when actually running an effect.

It may be of significance that wheel's firmware has been last updated a couple of years ago. I haven't had access to Windows machine since, but I'll try to find one to update to recent firmware to see if that helps. 

Otherwise, I 'd appreciate any pointers on what to try to make this work. Thanks.

[dmesg.txt](https://github.com/Kimplul/hid-tmff2/files/5469691/dmesg.txt)

berarma, 2020-10-31:
I've just taken a quick look to the code and it seems there's function calls to kernel code inside spinlock blocks.

The code inside a spinlock should be very simple and fast without a chance to sleep. Normally just setting variables that need to be protected from being changed simultaneously by different threads.

Kimplul, 2020-10-31:
Yeah, @berarma is right. From the dmesg log it seems like `hid_hw_start` inside a big spinlock is the main point of interest here. I could always take out the spinlock surrounding it, I mainly wrote it as a quick and dirty way to make sure that all data is initialized before any program attempts to communicate with the wheel.

I created a new branch for testing out spinlocks and possibly other locking mechanisms if they're necessary, try this out for a start and tell me if the dmesg logs change at all.
https://github.com/Kimplul/hid-tmff2/tree/spinlocks

dafteran4, 2020-11-01:
Yup, that took care of the errors, dmesg now looks fine. However, force feedback still doesn't work, everything seems the same as I described previously, buttons and axes showing, but no force except autocentering.

Now it seems somewhat similar to issue #5. I am attaching rdesc as suggested in that issue. 
[rdesc.txt](https://github.com/Kimplul/hid-tmff2/files/5470366/rdesc.txt)

The wheel is a variant T300 Ferrari Alcantara Edition.

Kimplul, 2020-11-01:
Well, I've been looking through the pcaps found over at https://gitlab.com/her0/tmdrv/-/issues/4 and while I can't really tell for sure, but it seems that different T300RS wheels have slightly different init procedures, which is odd. Maybe Thrustmaster has changed the controllers for the wheels over the years? Don't know, but it sure is annoying.

Anycase, I would probably need some more information of your specific wheel. Is it at all possible for you to capture some packets with Wireshark? You would need a Windows virtual machine to begin with.

dafteran4, 2020-11-01:
I'll need some time to setup the necessary, but I am glad to provide more data. What kind of virtual machine would do - qemu?

berarma, 2020-11-01:
> I'll need some time to setup the necessary, but I am glad to provide more data. What kind of virtual machine would do - qemu?

It's very easy to setup VMs with Gnome Boxes or even libvirt (the backend to Boxes).

dafteran4, 2020-11-01:
Indeed, it was surprisingly pleasant and hassle free. I have now a win10 installation inside a Gnome Box. Not sure how to add USB devices and what to do next with Wireshark.

Kimplul, 2020-11-01:
I personally use virt-manager as a frontend, but https://help.gnome.org/users/gnome-boxes/stable/usb-redirection.html.en should (hopefully) work.

For the most part I would need the packets from the initialization of the wheel, so you'll probably have to delete the `hid-tmt300rs.ko` and `hid-tminit.ko` files from `/lib/modules/$(uname -r)/extra`. This essentially is removing my driver from your system, so now when you plug in your wheel it should show up as 'Generic Thrustmaster FFB Wheel' or just 'Thrustmaster, Inc.'.

With the generic wheel, connect it to your virtual machine, and install the Thrustmaster drivers and firmware. The wheel will probably reconnect itself under a new name, and you will probably have to connect the wheel again to the virtual machine to get the driver/firmware installation to finish properly.

With everything up to date, we can start capturing packets with Wireshark. To run Wireshark, you'll need to load the `usbmon` module. `sudo modprobe usbmon` should work. You'll probably also have to start Wireshark with root privileges.

With Wireshark running, open up usbmon0. It should be pretty much in the centre of the Wireshark window. From there, we should add a display filter so that we can focus on the wheel's traffic. Open up a teminal on your linux machine, and run `lsusb`. Inside Wireshark, write `usb.bus_id == X && usb.device_address == Y`, where X and Y come from lsusb: `Bus 00X Device 00Y: ID 044f:065d Thrustmaster, Inc.`

Wireshark should start recording packets automatically, but if you want to restart the recording, press the triangle in the upper-left corner of the window. The packets we want to record will be sent when you connect the 'generic' wheel to the virtual machine, and when you connect the 'proper' wheel again, after the generic wheel has initialized itself. 

I would also like to see some packets from the force feedback tests in Thrustmaster's control panel, if possible.

To save the packets, stop the recording with the red square, and open File > Export Specified Packets. This will export the packets caught by the filter.

If you have any questions, just ask.

dafteran4, 2020-11-02:
Here is Wireshark output. First event is the wheel plugged-in. Then at 56secs I redirected usb to windows machine. At that point the wheel reconfigured its usb bus/address to 2.5 so I had to adjust the filters. Then I opened up TS panel and ran a all the FF effects and moved the wheel around a bit.

[usbmon.zip](https://github.com/Kimplul/hid-tmff2/files/5474216/usbmon.zip)

I can provide more data if needed.

Kimplul, 2020-11-02:
Wonderful, thanks.

I've had a quick look at the packets and to be honest, couldn't find anything too obviously different. That being said, packet 506 in your pcaps is slighlty different from the one I got when testing my own wheel. I can't really tell why on earth it would be, since as far as I have so far seen, all preceding packets are pretty much identical. Anycase, this packet is sent when the wheel is opened for reading/writing by a program, such as the TS panel. Sort of interesting, when I modified my driver to send out the version of the packet your logs have, I didn't get any force feedback. However, not sending it out at all, does give me FF. Why? No clue. Thrustmaster, fucking explain, please.

Anycase, I pushed a dirty fix to https://github.com/Kimplul/hid-tmff2/tree/spinlocks , which (temporarily) removes the packet. This has the side-effect of disabling autocentering when the wheel is unused, but doesn't seem to affect any effects when in games. Try it out, but I suspect more testing and digging will be needed.

dafteran4, 2020-11-02:
Trying latest spinlocks brach, things look somewhat different now. In oversteer, sliders "Force feedback gain" and "Autocenter strength" both work and have the expected effect on wheel behavior. Other sliders (Spring, Damper and Friction) are disabled and cannot be moved. fftest, however, still has no effect on the wheel.

dafteran4, 2020-11-02:
I'll try to get some wine games up to do more testing, though I guess if fftest isn't working, the games wouldn't also.

dafteran4, 2020-11-02:
You mentioned packet 506 earlier. Is this the right number? In my file I see:

`506	154.915888	host	2.5.1	USB	128	URB_INTERRUPT out`

It doesn't seem relevant. Also this is pretty late in the process, I am quite sure at that time I was in the middle of clicking on various example effects.

Kimplul, 2020-11-02:
With the leftover capture data "6012bf04000003c32000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"?

Well, you're free to be skeptical, but the packets at 504, 506 and 520 are ones that are typically sent out when a program opens the wheel. Sort of interesting, is that 506 isn't always sent out, whereas 504 and 520 are. I don't really know what it's for, but it seems like it has something to do with autocentering. That being said, there is a second method that sets autocentering, which is what my driver uses. That's also the method that you see in effect when using oversteer.

Well, interesting that we have some progress. I'll continue looking into it, but this is puzzling.

dafteran4, 2020-11-02:
You're right. I tried again and such a packet appeared when I opened the panel. For clarity, I did another capture and noting what I was doing.
```
0s   - wheel plugged-in
43s  - usd redirect to win machine
95s  - another usb redirect since device address has changed
140s - opened wheel panel and moved the wheel a bit
175s - started running FF effects
```

[wheel.zip](https://github.com/Kimplul/hid-tmff2/files/5476550/wheel.zip)

Kimplul, 2020-11-02:
That's interesting. Now there isn't anything like packets 504 or 506, instead there's just one, at 634, that's like a weird combination of both 504 and 506? But 648 in `wheel.zip` is still the same packet as 520 in `usbmon.zip`.

That being said I'm starting to think that this is a red herring. I tried not sending out anything, no 504, 506, or 520, and I still get FF. 

So, what I've now done is essentially try to replicate the messages I see in your logs as closely as possible. hid-tminit seems to already be pretty much identical, so I'm focusing on the hid-tmt300rs module. The newest commit in https://github.com/Kimplul/hid-tmff2/tree/spinlocks still gives me FF, and I included some interrupts that I previously thought to be unnecessary. Maybe they still are, and I'm way off the actual source of the problem, but give the newest commit a try and if that fails, I'll at least know to look elsewhere.

berarma, 2020-11-02:
I hope I'm correctly following the conversation. The packets you're talking about might be configuration commands. You could try changing the settings in the panel, quit then open again and see how settings are being sent when changed and when starting the panel. There might be groups or individual bits dedicated to different settings.

scarburato, 2020-11-02:
Take a look at packet 52, what it seems to be the [model number](https://github.com/scarburato/hid-tminit/blob/6a120c044242872e1cdf92463e2fee1a85dbff66/hid-tminit.h#L84) it's different from yours [pcap](https://github.com/Kimplul/hid-tmff2/blob/pcaps/t300init.pcapng) (packet 3750). You had `0x0206` meanwhile he has `0x0204` and also the others unknown fields are different.

Either my assumption on the field meaning was wrong or @dafteran4 wheel is another revision

Kimplul, 2020-11-02:
@scarburato Heyo, thanks for joining. I wouldn't be surprised if it's a hardware revision. The problem is that the data leaving the computer is still, as far as I can tell, pretty much identical. At least all leftover capture data is, so if the latest commit doesn't work I'm trying to look at control urbs, and maybe interrupts. Something that would set my T300RS apart from @dafteran4 's, besides that one number.

@berarma You're most probably right, but even just copying the packets I can see from @dafteran4 's logs doesn't seem to work, which makes me really confused. I'd like to figure out what configuration works first, but if that isn't happening I suppose I might as well do as you suggested.

scarburato, 2020-11-02:
If you watch the request `73` (used to ask the wheel what is seems to be the model information) is made by the Windows driver multiple time even after the init procedure, for example after packet 190 and after packet 608 it's spammed 35+ times

Maybe you driver could make this request once one during the probe and keep something to discriminate the hardware revisions 

Anyway I'll add this new value to the hid-tminit as "T300 Ferrari Alcantara Edition"

dafteran4, 2020-11-02:
Tried latest spinlocks branch. Unfortunately, fftest still doesn't work. I tried another capture that might be helpful by using the module from this latest update. 

- First I attached the wheel, kernel module loads, 
- then I try fftest, doesn't work - wheel not moving, 
- then, with the wheel still connected, I redirect usb to windows (usb address now remains the same), try effects in windows, works as expected, 
- then turning off redirect, try again fftest, doesn't work,
- then turning on redirect, try again in windows, works.

[wheel2.zip](https://github.com/Kimplul/hid-tmff2/files/5477906/wheel2.zip)

Thanks to everybody for participating in this.

Kimplul, 2020-11-02:
Hmmmm. I'm assuming by latest you mean the 'Added mysterious control out'-commit? Anycase, could you download https://gimx.fr/download/b882e209a0ac023d03abbf560dfc3f25fe6367ca/fedit.zip to your Windows machine? It's a force feedback editor, I just want to make sure that the commands that produce the wanted effects aren't different between the two wheels.

With FEdit open, add in a `ConstantForce`, and move it up to the beginning of the line, like so:
[image]

Doubleclick on the added force, and go to 'Direction'. Type in 9000 in the text box, and press 'Play'. Capture the packets that this produces.

Thanks for your patience with this whole ordeal.

dafteran4, 2020-11-03:
> Hmmmm. I'm assuming by latest you mean the 'Added mysterious control out'-commit? 

Yes, that was the latest commit.

Here are the packets from FEdit: [wheel3.zip](https://github.com/Kimplul/hid-tmff2/files/5480779/wheel3.zip)

Kimplul, 2020-11-03:
Welp, good news and bad news, The good news is that the packets are the same, so I probably won't have to write a whole new set of instructions for your wheel specifically. Bad news, I still don't know what the core issue is.

berarma, 2020-11-03:
If the packets are the same maybe the timings could have an influence. I've seen `t300rs_init_interrupts` is sending a batch of unknown commands as fast as possible. Would inserting short waits (some tens of ms) between calls to `t300rs_send_int` make a difference?

Also, `fftest` might be doing too much and triggering multiple issues. It might be a bit too indeterministic for testing at this stage. You could try https://github.com/berarma/ffbtools. After building, you can play the effect files in the tests directory using this command:

`bin/ffbplay -d /dev/input/your_device tests/constant.ffb`

dafteran4, 2020-11-03:
I'm trying to get a sense on how the read these captures, I am still a bit in the dark. Can you perhapse help me get a better understanding of the elements? 

I figure "URB INTERRUPT out" are the packets where host send FFB command to the wheel. By monitoring fftest I see that it sends 60 00 ID 89 01. This corresponds to 'starting to play' in the documentation. ID is the number of the effect selected. This is pretty much everything that it sends. How does the wheel know what these IDs mean?

Is it possible somehow for me to issue these commands manually so I could try replaying the ones I see in windows?

dafteran4, 2020-11-03:
@berarma I'll try with ffbtools.

Kimplul, 2020-11-03:
@berarma Excellent idea, I added some short sleeps in between the interrupts. Newest commit in spinlocks as usua.

@dafteran4 Sure, you're correct about the starting of effect. The ID is given by the upload command, so for constant force it would be 60 00 NEW_ID 6a ..., in my documentation under 'Initialisation'.
A new ID can be given once an effect is stopped, with 60 00 ID 89 00 .... For some reason this command works both as a "stop the effect" and "unload the effect and free this ID".

To be honest, I don't think there is any way to send out individual commands. At least I haven't found any tool like that, but you're free to modify my code however you wish.

berarma, 2020-11-03:
> Is it possible somehow for me to issue these commands manually so I could try replaying the ones I see in windows?

I wrote a deceptively simple command to do this for the Logitech wheels. I may try to modify it so it can work on any device and add it to ffbtools.

dafteran4, 2020-11-03:
Commit 4356921 breaks oversteer which I previously reported had functioning Force and Autocenter sliders. With this commit oversteer opens at 40 deg angle and the wheel goes berserk :), also Force/Autocenter sliders don't work anymore. Reverting prior to this commit fixes that and sliders work again.

Kimplul, 2020-11-03:
That's odd. I added back the open/close functions, assuming that it was a red herring. Guess not? Anyway, I took them away again. 

I did notice that Oversteer doesn't work for me on Ubuntu, I get `ValueError: invalid literal for in() with base 10: '0.0'`. Would you like me to open up an issue @berarma ?

berarma, 2020-11-03:
> I did notice that Oversteer doesn't work for me on Ubuntu, I get `ValueError: invalid literal for in() with base 10: '0.0'`. Would you like me to open up an issue @berarma ?

Yes, please. I need the full error and the version you tried, master or the last release tag. You can see if the error goes away with any of them.

dafteran4, 2020-11-03:
A lot is going on. :) 

> EDIT: Oh yeah, and I also removed the command that sets the wheel's range from the startup, I noticed that it wasn't being sent out by the Windows driver.

`    t300rs_range_store(dev, &dev_attr_range, range, 10);`

Without this line oversteer goes mad as it tries to set a very narrow range. 

So with b092962f, oversteer works well if range_store is applied,
But with b1ae16b1, oversteer doesn't work even if range_store is applied.
Either way, I am not getting anything out of fftest and ffbplay.

Kimplul, 2020-11-03:
Right, so my opening is still somehow flawed. Interesting. Well, one other interesting thing is that your wheel apparently needs a 60 01 05 ... when closing, whereas mine apparently doesn't? I also noticed in your usbmon.pcapng that at one point, it's sent twice when opening the device. I wonder if it could be something like a switch?

Could you try the newest commit? I essentially just added a 60 01 05 ... command to the init sequence, imitating the packets in usbmon.pcapng. Not holding my breath, but, anything small like that could be the solution here.

Man, I wish I had your wheel here. This would be so much quicker and less nerve-wracking.

berarma, 2020-11-03:
> Man, I wish I had your wheel here. This would be so much quicker and less nerve-wracking.

Don't you know about USB over IP? I'm not saying this is the solution but it's there. :D 

@dafteran4: I've added `rawcmd` to ffbtools. I allows sending bytes to a device using the hidraw device. You can look for the hidraw device of your wheel in the `dmesg` output.

berarma, 2020-11-03:
You must already know this but in case you don't, this is the Logitech Force Feedback Protocol document. This protocol (Classic Format in the document) has been around since the early days of FFB so newer devices might own some debt to it, or not.

https://opensource.logitech.com/opensource/images/c/ce/Logitech_Force_Feedback_Protocol_V1.5.pdf

Kimplul, 2020-11-03:
@berarma I've heard of it, the company where my dad worked used one tool like that to try to develop some product, but all I heard was that it was awful, because you couldn't do everything needed over the internet. In this case, I probably won't have any clue if the wheel is reacting to my packets.

Also, hey, thanks, I knew that document existed but I could never find it.

dafteran4, 2020-11-03:
> Could you try the newest commit? I essentially just added a 60 01 05 ... command to the init sequence, imitating the packets in usbmon.pcapng. Not holding my breath, but, anything small like that [...]

@Kimplul Nope, didn't change anything. If we can't figure it out, I'll get a quote from dhl. :)

@berarma Thanks for `rawcmd`. I quickly tried sending force commands as seen in win, but that didn't work. Still it's very nice to have, so I can experiment further.

scarburato, 2020-11-03:
> I quickly tried sending force commands as seen in win, but that didn't work. Still it's very nice to have, so I can experiment further.

Did you check with Wireshark if the packets were in facts equal?

Kimplul, 2020-11-03:
I realized that I haven't actually shown what my pcaps look like, sorry. I uploaded them to the pcaps branch, the Generic -> T300RS pcaps are in `t300rsinit.pcapng` and T300RS -> are in `t300rs_kimplu.pcapng`. Now you can go through them yourselves and shout at me because I missed something obvious.

In t300rs_kimplu.pcapng, I open up the TS panel at packet 272, play the 'Engine' effect at 1252 and close the TS panel at 4732.

dafteran4, 2020-11-04:
> Did you check with Wireshark if the packets were in facts equal?

@berarma I did. I had to modify your utility to allow MAX_SIZE=64, otherwise I would get out 80 bytes requests instead of 128 bytes as seen in windows. I have  no clue if that matters at all though. I am still learning how to decipher all that usb traffic. I only sent the force command, I didn't do any initialization, so that was a bit naive, I guess. 

Am I right in thinking that by using `usb.endpoint_address.direction == 0` as a display filter in Wireshark, this shows only 
outbound traffic (from host to wheel)?

@Kimplul Thanks for all your work, please don't strain yourself too much on this. I'll try to look at the captures myself and see what could be different. As said above, I am still learning, so any general tips on how to read wireshark traffic are appreciated.

Kimplul, 2020-11-04:
Not quite, `usb.src == "host"` is more what you're looking for, I think. The main thing to look for is the source of the packet, and when looking at outbound packets, that would be the host.

This link will probably explain the different packets better than I can: https://beyondlogic.org/usbnutshell/usb4.shtml
But in short, the in/out in URB_INTERRUPT or URB_CONTROL refers more to if we want to send data or receive data. So for example, a URB_CONTROL in from host -> device roughly means "Hey, could you please send me some data that corresponds to my parameters?", whereas URB_CONTROL out is roughly "Hey, could you please do this for me?"

Approximately the same thing can be said for URB_INTERRUPTs, you've probably noticed that the force effects are uploaded(downloaded? Logitech uses downloaded) to the device with URB_INTERRUPT out packets. 

URB_INTERRUPTS in are commonly used for input by HID devices, so in this wheel's case, reporting the position of the wheel, buttons and pedals. The kernel usually handles asking for input automatically, and my driver doesn't touch these packets. If you look closely at the pcaps, you'll probably notice that there are some interrupts in, that start with 07 and some with 12 and a couple with 02. So far I've concluded that 07 is the input, buttons etc. but I have no clue what 12 or 02 means. So far I haven't needed them, and your 12/02 packets are the exact same as mine, so right now I'm not in a hurry to figure out what they mean, but at some point I probably will have to.

As far as I'm aware, yes, 128 bytes is what you want.

Kimplul, 2020-11-04:
What @scarburato said earlier is starting to really bug me, why would the driver need to spam the model packet(control 73) for 35+ times? I mean, surely it can't have anything to do with how the wheel reacts to force feedback requests? It just seems so incredibly dumb, I don't understand.

This came up when I tried sending out the interrupts I saw in the init stage, the interrupts in after 42 01 00 ..., packet 167 in `usbmon.pcapng`.  I found out that by sending them out manually, I got similar responses as in `usbmon.pcapng`. Packets 174, 178, 182, 186 and 190, that is. With the exception that I don't get 174, but it seems to be an input packet as described previously, so don't really know what's up with that.

Anyway, as always, newest commit in spinlock, you can try to compare the packets from my driver with your `usbmon.pcapng`. Personally I say that right now the biggest difference is the 35+ packet spam, but it is entirely possible that I'm just not seeing something.

dafteran4, 2020-11-04:
Finally some success. This is on 85715ce, I've been testing for hours now and haven't tried your newest commit yet.

After plugging in the wheel, I run the following command imitating behavior from win machine. At 3 the wheel start spinning. And after that both ffbplay and fftest work as expected. 

```
1. Pressing Button Defaults in Windows

6002bf00
600803b71e00

2. Requests sent out when opening the panel

60110404000003c32000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
6002bf00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0a041210000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0a04000c000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0a049003000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
60010500000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

3. Reques sent out when triggering engine effect

6000016b305cb9f90000180200805501b77971050000014f98080000000000000000000000000000000000000000000000000000000000000000000000000000
600001eb305cb9f90000180200805501b77971050000014f9808000000000000000102cb3364feff00006b000080034f740e0000000000000001000000000000
600002690c0a0000c0012918034f740e000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

```

Number 1 doesn't seem to be needed. But, what is very peculiar is that if I omit number 3, then fftest/ffbplay don't work. Will do some more testing and report back.

Kimplul, 2020-11-04:
Heck yeah dude!

Interesting, have you tried sending out either just packet 2 or packet 3 from 3? I don't have anything mentioning them in my documentation, and I don't know what they mean/do.

dafteran4, 2020-11-04:
Ok, here's the sequence imitating CF from FEdit. It works, exactly replicating win behavior. I can run it mutiple times in sequence or alternating with engine effects sequence from my last post and everything works fine.

However, I have been a bit premature calling fftest/ffbplay working... in both cases (engine and const force pattern) they behave erracitally, usually just replaying the last pattern or doing nothing at all - they definitely do not work as they should.

```
@ 0 secs - redirected wheel

42010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0a049003000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0a04000c000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0a041210000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0a040006000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

@ 2 secs - ???

0a040300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0a040300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
0a040300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

@ 7 secs - opened FEdit
60110404000003b71e00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
6002bf00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
60010500000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

@ 25 secs - started CF effect
6000016afd3f0000fc7f0000fc7f004f60180000000000ffff000000000000000000000000000000000000000000000000000000000000000000000000000000
60000189010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

@ 30 secs - stopped CF effect
60000189000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```

Kimplul, 2020-11-04:
How exactly did you test out fftest/ffbplay? Did you open up fftest/ffbplay, then send those signals? Or signals first, then open up something? Also, what happens if you leave out the second command at 7 secs, `60 02 bf 00 ... `?

dafteran4, 2020-11-04:
> How exactly did you test out fftest/ffbplay? Did you open up fftest/ffbplay, then send those signals? Or signals first, then open up something?

After. First I run the imitation sequence (cf or engine), then after that I try running fftest and ffbplay. The both seem to behave weirdly, not doing their own thing, but running the effect from previous imitation sequence or nothing at all.

> 
> Also, what happens if you leave out the second command at 7 secs, `60 02 bf 00 ... `? It should just be setting the gain, just weird that it shows up there of all places.

It still works just the same.

> EDIT: One more question, have you tried downloading multiple effects to the device at the same time? In FEdit, you can just add in another effect and imitate that. My thinking is that so far the tests have been with just one effect at a time, maybe the wheel behaves weird if we try to upload more than it can handle? What if the cap is something like four, fftest uploads five efffects I believe, which could mess the wheel up.

I'll try that later and report.

> EDIT EDIT: Oh, I just noticed. What duration does FEdit give you? For some reason the packet has a different duration than mine.
> And what happens if you set Duration to INFINITE?
I haven't looked at the timings tab then and didn't save to disk to check now. Default seems to be 4160000, though.

These are the INFINITE duration commands to start CF effect that I get from FEdit win session. I can replay on host and it works just as fine:
```
6000016afd3f0000fc7f0000fc7f004fffff0000200800ffff000000000000000000000000000000000000000000000000000000000000000000000000000000
600001c9004400000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```

dafteran4, 2020-11-04:
> > EDIT EDIT: Oh, I just noticed. What duration does FEdit give you? For some reason the packet has a different duration than mine.
> > And what happens if you set Duration to INFINITE?
> > I haven't looked at the timings tab then and didn't save to disk to check now. Default seems to be 4160000, though.
> 
Using stopwatch to measure the duration when replaying I get about 6 seconds. :)

Kimplul, 2020-11-04:
Thanks, seems fine. Sort of odd that your default is different from mine in FEdit, but hey, it's whatever.

I'll just quickly point out that you should remember to unload/free/stop the effects before opening up fftest/ffbplay, as uploading an effect before stopping the one in the ID slot is pretty much undefined behaviour. Seems like you did it with the constant force and FEdit, but just a reminder to be on the safe side.

Could you also try only sending out `60 01 05 ...`, and see if that has any effect? My assumption is that all `0a ...`and `42 01 ...` packets are unnecessary.

dafteran4, 2020-11-05:
I recorded the exact output from `ffbplay tests/constant.ffb` and started testing by replaying the usb interrupt out packets with a script which sends raw data to `hidraw` device. The packet `60 01 05` is a must - it has to be sent after initializing the wheel, otherwise there are no effects. Everything else is unnecessary. So I would send the following with the script and get the expected behavior:

```
# mystery line - necessary to call the first time after plugging-in the wheel,
# otherwise nothing is played. It can be ommited on subsequent runs.
60010500000000000000000000000000000000000000000000

# exact usb interrupt OUT from ffbplay tests/constant.ffb
6002ff00000000000000000000000000000000000000000000 # set gain
6000016ad8dc0000000000000000004fffff0000000000ffff # upload effect
60000189010000000000000000000000000000000000000000 # start playing
6000010a282300000000000000000000000000000000000000 # change to next effect 
6000010a589e00000000000000000000000000000000000000 # change to next effect
6000010aa86100000000000000000000000000000000000000 # change to next effect
60000189000000000000000000000000000000000000000000 # stop playing
# note that we wait 5 secs before changing and stopping effects
```
One would think this resolves the issue. Alas, there is a complication. This only works if I send it with the script directly to `hidraw`. If I use `ffbplay`, it simply repeats the last played effect and doesn't switch between effects every 5s. This is true even if I send `60 01 05` immediately before running `ffbplay`. The bizarre thing being, if I record all this in Wireshark, the outputs from `ffbplay` and the script match exactly.

Kimplul, 2020-11-05:
I wonder if the packets that get sent out by my driver are sent out too fast?

When you send out the packets manually, does the wheel respond to them in the order that they arrive, or does it receive some amount of packets and then sends a bunch of responses back?

dafteran4, 2020-11-05:
I am not sure what constitutes wheel 'response'. I am attaching wireshark recording from my session (unfortunately I had `usb.src == "host"` enabled, so it only half of the packets, but I guess it should be enough for the purpose).  

[ffbplay_vs_manual.zip](https://github.com/Kimplul/hid-tmff2/files/5496027/ffbplay_vs_manual.zip)

In the capture at packet 4 I manually send `60 01 05` then at 6 `ffbplay` starts. Then at 4693 I run the whole sequence manually. 

Looking at `ffbplay` and manual sequence, interrupt OUTs match exactly, but with manual sequence there are INs after every OUT and there are larger time gaps between OUTS.

Kimplul, 2020-11-05:
Right, sorry, I should've been more clear.

What I meant was that when a `60 00 01 ...` is sent out manually is the `[Response in: NUMBER]` part in wireshark immediately after the packet? So, if `60 00 01 ...` is packet no. 4, is the response packet number 5?

Here's a picture that shows where to find the section I'm talking about.
[image]

In the pcaps you provided that information is unfortunately lost because of the filtering.

Kimplul, 2020-11-05:
Heyo, sort of interesting, I looked at new-lg4ff and noticed that berarma used hid_hw_request, so I thought "well heck, I might as well" and I'm having similar symptoms. Autocentering works just fine, but otherwise I don't get any effects. It'd be pretty funny if this method worked for you, so try it out.

Newest commit in spinlocks. Don't mind the previous commits, I sort of messed around with mutexes and as it turns out, using them in an interrupt context crashes your machine.

dafteran4, 2020-11-05:
Testing f635e5d... `ffbplay` with all examples in tests dir works great and correctly as far as I can tell. That's out of the box, no need to send any manual commands. `fftest` also works 0-3, Mayber number 0 (periodic sine) seems a bit weird, vibrates and moves erratically, it is not like in `ffbplay`, but other seems fine. All in all, this looks great.

Can you recommend any games that should be good for testing force feedback?

dafteran4, 2020-11-05:
Not to be a party breaker, but while effects work now, I don't get any movement and button pushes registered in oversteer. :)

Kimplul, 2020-11-05:
Holy fucking shit what a weird experience this has been. Well, heck yeah dude, we dun did it.

Personally I'd recommend Wreckfest, Dirt Rally, Dirt Rally 2 and I guess rFactor 2. The F1 series works but feels somewhat 'floaty', but from what I've read on the internet that might be intentional? Don't know.

Dirt Rally one and two are the only two games that I have experience on windows with, and they're fairly close on Linux as well, in my opinion.

Buttons, yeah, hold on, I've had something similar before. Let me check if the solution is still the same.

Kimplul, 2020-11-05:
Right, test the newest commit. You might have to unplug your wheel and reboot, I at least had to.

berarma, 2020-11-05:
Aren't you skipping the first byte here?

https://github.com/Kimplul/hid-tmff2/blob/4621e44e97bd17e32c19af8383519d3d9669bfb1/hid-tmt300rs.c#L102

I've also noticed you're using `greport` but have `t300rs_device_entry->report` that isn't used, same with `t300rs_device_entry->ff_field'. 

The whole block at https://github.com/Kimplul/hid-tmff2/blob/4621e44e97bd17e32c19af8383519d3d9669bfb1/hid-tmt300rs.c#L1692 makes little sense to me but it could be my fault.

[10 more replies]

closed 2020-11-06 by dafteran4