Kimplul/hid-tmff2 #7: Range Change

dafteran4, 2020-11-13:
In the README, there are nstructions on how to change range by writing to file:
`/sys/bus/devices/XXXX:044F:B66E.XXX/range`

In my system (Arch Linux) there is no such file, instead the range can be changed at:
`/sys/devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5:1.0/0003:044F:B66E.0008/range`

The exact path varies and can be seen in `dmesg` when plugging-in the wheel. 

The driver seems to init 900 degrees range, aren't TM3000RS wheels all 1080? 

Btw, I've been testing the wheel in Dirt Rally and BeamNG and it works great.

Kimplul, 2020-11-13:
Seems like I have a typo, the path should be `/sys/bus/hid/devices`, good catch.

Yep, all TM wheels that I know of support 1080 degrees, but Oversteer seems to at the moment only go up to 900, which is what I set the default to. I can maybe open up a patch for 1080 support for TM in Oversteer, I'll see.

Good to hear that the driver works.

dafteran4, 2020-11-13:
> Seems like I have a typo, the path should be `/sys/bus/hid/devices`, good catch.
Ah, that's shorter and more practical. Thanks.

closed 2020-11-13 by dafteran4

berarma, 2020-11-14:
> Ah, that's shorter and more practical. Thanks.

It doesn't look shorter, it's actually longer.

> Yep, all TM wheels that I know of support 1080 degrees, but Oversteer seems to at the moment only go up to 900, which is what I set the default to. I can maybe open up a patch for 1080 support for TM in Oversteer, I'll see.

Feel free to open an issue for it and I'll work on it when I can. Implementing device dependent ranges is on my todo list, but I didn't notice it was important for TM support.