Kimplul/hid-tmff2 #8: Installation on pop os (ubuntu) 20.04

fcf1, 2020-11-15:
Never had a wheel, ordered T300RS GT today. I tried to install driver and got this:

> make -C /lib/modules/5.8.0-7630-generic/build M=/home/fcf/hid-tmff2 modules_install
> make[1]: Entering directory '/usr/src/linux-headers-5.8.0-7630-generic'
>   INSTALL /home/fcf/hid-tmff2/hid-tminit.ko
> At main.c:160:
> - SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
> - SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
> sign-file: certs/signing_key.pem: No such file or directory
>   INSTALL /home/fcf/hid-tmff2/hid-tmt300rs.ko
> At main.c:160:
> - SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
> - SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
> sign-file: certs/signing_key.pem: No such file or directory
>   DEPMOD  5.8.0-7630-generic
> Warning: modules_install: missing 'System.map' file. Skipping depmod.
> make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-7630-generic'

Kimplul, 2020-11-15:
The modules should be installed and ready to go, the errors you're seeing are telling you that you don't have a certificate to sign the module with. Linux should still load the modules, signed or not.

If you want to silence the errors, here's a SuperUser thread about it:
https://superuser.com/questions/1214116/no-openssl-sign-file-signing-key-pem-leads-to-error-while-loading-kernel-modules/1322832#1322832

fcf1, 2020-11-18:
OK, there is no more errors, still warning remains "missing 'System.map' file. Skipping depmod" and module does not install.

Kimplul, 2020-11-18:
What do you mean by 'module does not install'? Is `/lib/modules/$(uname -r)/extra` empty or how do you figure that it doesn't install?

fcf1, 2020-11-18:
Modprobe says module does not exist.

> modprobe hid_tmt300rs
> modprobe: FATAL: Module hid_tmt300rs not found in directory /lib/modules/5.8.0-7630-generic

 After some googling I run "sudo depmod -A" and now modprobe works and oversteer sees the wheel.

Kimplul, 2020-11-18:
Oh yeah, gotcha. That's sometimes needed, I suppose I could add it to the install.

fcf1, 2020-11-18:
If I understand correctly, module signing is not something you can correct on your end? It's not very beginner friendly.

Kimplul, 2020-11-18:
That's correct, everyone who compiles this module has to sign their module themselves, as a way to say "I compiled this". If they want/need to.

Yeah, technological security unfortunately can be rather confusing, but try to break it down and draw comparisons to the real world. For example, signing a module can be seen as literally putting your name on it. "I did this, and if it does something it shouldn't, I'm fully responsible." That kind of deal.

closed 2020-11-18 by fcf1