Kimplul/hid-tmff2 #9: No FFB archlinux

cloudstrife4213, 2020-12-02:
Hi,

First I would like to thanks you for your amazing job.

I installed your driver, my wheel is well recognized as T300rs, but I have no FFB, only autocenter.

When I try fftest, nothing happens, and in oversteer I only have force gain which is activated, nothing else.

Is there any solution to my problem?

Thanks in advance

Kimplul, 2020-12-02:
Heyo, this sounds like it could be similar to #6, but we thought we found the solution for the issue in that thread. 

Just to get the obvious things out of the way first:
1. Did you make sure the `PS3/PS4` switch is in the `PS3` position?
2. Is this a bog standard T300RS, or some special edition, like Alcantara or something like that?
3. Did you reboot your computer and/or remove and reinsert the USB plug?

cloudstrife4213, 2020-12-02:
Thanks for the fast reply.

1. Yep, the switch is on the PS3 position.
2. It's the T300RS GT, I think it's the standard edition.
3. I tried to reboot my computer a lot of times, and remove the USB plug too. 

The only thing my wheel do is autocalibrate at boot, or when i plug it, but that's all. I have no sensation in different games that I tried.

Kimplul, 2020-12-02:
Right, that's what I feared. As you can probably see, the last time this kind of issue arose, it needed quite a bit of data to figure out. To start off with, could you post your `dmesg` output when plugging in the wheel, and also when trying to play an effect with `fftest`?

cloudstrife4213, 2020-12-02:
this is the result of dmesg when I plug in my wheel

`dmesg -W
[   79.791683] usb 2-1: new full-speed USB device number 4 using xhci_hcd
[   79.933362] usb 2-1: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[   79.933366] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   79.933368] usb 2-1: Product: Thrustmaster FFB Wheel
[   79.933371] usb 2-1: Manufacturer: Thrustmaster
[   79.935688] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/0003:044F:B65D.0004/input/input25
[   79.936020] hid-generic 0003:044F:B65D.0004: input,hidraw3: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:00:14.0-1/input0
[   79.957863] hid_tminit: loading out-of-tree module taints kernel.
[   79.957915] hid_tminit: module verification failed: signature and/or required key missing - tainting kernel
[   80.025547] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/0003:044F:B65D.0004/input/input26
[   80.025753] thrustmaster init 0003:044F:B65D.0004: input,hidraw3: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:00:14.0-1/input0
[   80.045569] usb 2-1: urb status -71 received
[   80.045739] usb 2-1: USB disconnect, device number 4
[   80.698405] usb 2-1: new full-speed USB device number 5 using xhci_hcd
[   80.839444] usb 2-1: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[   80.839448] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   80.839451] usb 2-1: Product: Thrustmaster T300RS Racing wheel
[   80.839453] usb 2-1: Manufacturer: Thrustmaster
[   80.840905] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/0003:044F:B66E.0005/input/input27
[   80.841088] hid-generic 0003:044F:B66E.0005: input,hidraw3: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:00:14.0-1/input0
[   80.945601] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/0003:044F:B66E.0005/input/input28
[   80.945806] t300rs 0003:044F:B66E.0005: input,hidraw3: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:00:14.0-1/input0
[   80.945831] t300rs 0003:044F:B66E.0005: force feedback for T300RS
`

And when I'm triying to play an effect with fftest nothing happens, in dmesg there is no more message when I'm testing my wheel. This is the result of ffbtest

`fftest /dev/input/event22
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/event22 opened
Features:
  * Absolute axes: X, Y, Z, RZ, Hat 0 X, Hat 0 Y, 
    [27 00 03 00 00 00 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: Constant, Periodic, Ramp, Spring, Friction, Damper, Rumble, Inertia, Gain, Autocenter, 
    Force feedback periodic effects: Square, Triangle, Sine, Saw up, Saw down, 
    [00 00 00 00 00 00 00 00 00 00 FF 1F 03 00 00 00 ]
  * Number of simultaneous effects: 16

Setting master gain to 75% ... OK
Uploading effect #0 (Periodic sinusoidal) ... OK (id 0)
Uploading effect #1 (Constant) ... OK (id 1)
Uploading effect #2 (Spring) ... OK (id 2)
Uploading effect #3 (Damper) ... OK (id 3)
Uploading effect #4 (Strong rumble, with heavy motor) ... OK (id 4)
Uploading effect #5 (Weak rumble, with light motor) ... OK (id 5)
Enter effect number, -1 to exit
0
Now Playing: Sine vibration
Enter effect number, -1 to exit
1
Now Playing: Constant Force
Enter effect number, -1 to exit
2
Now Playing: Spring Condition
Enter effect number, -1 to exit
3
Now Playing: Damping Condition
Enter effect number, -1 to exit`

Kimplul, 2020-12-02:
Out of curiosity, could try out ce269e6? Last time the issue was that the commands that I send out to the wheel were I guess sent out the wrong way, so we changed it. However, the method I had at this commit still worked for me, and different T300RS wheels seem to react to the different methods in different ways.

Sorry if I sound patronizing, but do you know how to check out a specific commit?

cloudstrife4213, 2020-12-02:
I've just tried, and know in dmesg when I'm plugging my wheel there is this : 

> [  568.031308] BUG: scheduling while atomic: kworker/1:1/163/0x00000002
[  568.031369] Preemption disabled at:
[  568.031374] [<0000000000000000>] 0x0

and always no reaction when i'm tryin to use fftest

Kimplul, 2020-12-02:
Oh shit, sorry, I went too far back. dfd059d should be new enough that that issue is fixed, but we're still on the older method.

cloudstrife4213, 2020-12-02:
Once again, nothing happens when I'm using fftest. When I plug my wheel to the computer, the steering wheel starts to calibrate itself, locks to the right, then restarts its calibration, until the end this time.

Dmesg when I plug it

> [  619.224724] usb 2-1: new full-speed USB device number 12 using xhci_hcd
[  619.366355] usb 2-1: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[  619.366360] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  619.366362] usb 2-1: Product: Thrustmaster FFB Wheel
[  619.366364] usb 2-1: Manufacturer: Thrustmaster
[  619.370159] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/0003:044F:B65D.000C/input/input35
[  619.370360] thrustmaster init 0003:044F:B65D.000C: input,hidraw3: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:00:14.0-1/input0
[  619.370947] thrustmaster init 0003:044F:B65D.000C: bEndPointAddress: 1 1
[  624.384750] thrustmaster init 0003:044F:B65D.000C: **fuck: -110**
[  624.403131] usb 2-1: urb status -71 received
[  624.403263] usb 2-1: USB disconnect, device number 12
[  625.051385] usb 2-1: new full-speed USB device number 13 using xhci_hcd
[  625.195756] usb 2-1: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[  625.195761] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  625.195763] usb 2-1: Product: Thrustmaster T300RS Racing wheel
[  625.195765] usb 2-1: Manufacturer: Thrustmaster
[  625.197343] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/0003:044F:B66E.000D/input/input36
[  625.197501] t300rs 0003:044F:B66E.000D: input,hidraw3: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:00:14.0-1/input0
[  625.197520] t300rs 0003:044F:B66E.000D: force feedback for T300RS
[  625.197590] t300rs 0003:044F:B66E.000D: Response: 56 19 0 0

Kimplul, 2020-12-02:
Okay, one more thing to check, could you test out https://github.com/scarburato/hid-tminit? It's essentially a more generic init than what I have in this repo.

I did notice that the "Response" in your output is different from my wheel, if I recall correctly we assumed that it had something to do with the firmware version. When did you last update the wheel's firmware?

cloudstrife4213, 2020-12-02:
Oh wait, I've never done any update of its firmware. Is it possible that this is the source of the problem? I've been on linux for 12 years, my wheel was only used for ps4, and I never needed to update the firmware.

I'll try the generic driver.

Kimplul, 2020-12-02:
It could be that Thrustmaster has changed the USB api over the years, I don't really know. I've tested my wheel with firmware version 31, at least.

The fftest issue seems to be because the wheel isn't picked up by the `hid-tmt300rs` module, instead `hid-generic` kicks in. Did you remove the `hid-tmt300rs` module?

scarburato, 2020-12-02:
You have to `rmmod hid-tminit` before `insmod hid-tminit.ko` (or `modprobe hid-tminit` if you have installed it)

cloudstrife4213, 2020-12-02:
Yep, I removed it, so I retried with both but it's not working.

I'll boot on windows, and see what's my wheels firmware. Maybe it will help us.

arabek, 2020-12-20:
Had the same problems, wheel FW was at version 25. After finding a windows installation and updating to 31, ffb is now active and kicking.

Kimplul, 2020-12-21:
Thanks for the info. Good to know that firmware version plays a role, although sort of unfortunate that owners of the wheel might have to find a Windows machine to get their wheel working. I could always see if there is some way to update the firmware from Linux, but don't hold your breath. Anyway, I'll add a warning to the README about this.

I'd still like to hear from the OP of this thread before I close this.

berarma, 2020-12-21:
> Thanks for the info. Good to know that firmware version plays a role, although sort of unfortunate that owners of the wheel might have to find a Windows machine to get their wheel working. I could [...]

It should be possible to do from a Boxes/libvirt VM.

arabek, 2020-12-21:
Thanks @Kimplul for the excellent work you did with the driver. Much appreciated!

Kimplul, 2021-03-08:
Closing due to inactivity.

closed 2021-03-08 by Kimplul

exapsy, 2021-06-12:
I had "function not implemented" error as well.

I fixed it by updating the firmware to 33 (had 31). Now it works fine.