Kimplul/hid-tmff2 #10: Outpu queue full in kmesg/dmesg

arabek, 2020-12-21:
After game launch, this starts spamming the console:
```
[ 3034.516377] t300rs 0003:044F:B66E.000B: output queue full
[ 3034.533090] t300rs 0003:044F:B66E.000B: output queue full
[ 3034.533094] t300rs 0003:044F:B66E.000B: output queue full
[ 3034.533096] t300rs 0003:044F:B66E.000B: output queue full
[ 3034.551362] t300rs 0003:044F:B66E.000B: output queue full
[ 3034.551365] t300rs 0003:044F:B66E.000B: output queue full
```
Game: ETS2 on Steam (running the windows version via Proton Experimental)

berarma, 2020-12-21:
This is the game sending more effect updates than the device can handle. Limiting the frame rate can help with the issue.

It happens with other wheels too.

arabek, 2020-12-21:
I decided to hack a bit and changed the `DEFAULT_TIMER_PERIOD` to `2` instead of `8` in `hid-tmt300rs.h`. For now it seems stable and no msg spam even with unlocked framerate. Might be a placebo but the effects in ETS2 seem more defined / on point too.

Kimplul, 2020-12-21:
Might not be placebo, as you've essentially increased the effect's resolution. That being said I'm not really comfortable lowering the timer period down to 2 ms, because I experienced some issues with my wheel when I set the timer that low. The wheel itself would occasionally freeze up, and I'd have to unplug the wheel and plug it back in. Though that was before the resolution we found in #6, so the timing issue might've been also been affected by the changes.

arabek, 2020-12-22:
I've now played with the 2ms setting for quite some time and it didn't occur anymore. Neither with ETS2 nor Project CARS 2 under Steam/Proton.

Kimplul, 2020-12-22:
That sounds promising. I'd still like to test it out on my own wheel.

I could also just set the timer to be a module parameter, so users could set their own timers, if I find out that different wheels have different tolerances.

arabek, 2020-12-22:
So i've played Assetto Corsa Competizione, and this might give you a hint:
https://www.wolframalpha.com/input/?i=333hz+to+miliseconds
(game allows update rates for force feedback up to 333hz). Basically what i'm saying is - setting `2ms` (`3ms` might do too, idk, never tested) is the only option that makes sense to me (and leaves 1ms for other stuff the wheel might want to do). Since i've set it to `2ms` i never had any dmesg spam with queue (nor broken effects). Hope it helps.

Kimplul, 2020-12-23:
Yep, thanks, I'll try it out on my wheel as soon as possible. Not 100% confident in pushing such a change without some testing by myself.

Kimplul, 2021-01-08:
Hello again, sorry for the long delay. I added in functionality for setting the timer period when loading the module, either by `modprobe hid-tmt300rs timer_msecs=NUMBER` or by setting `options hid-tmt300rs timer_msecs=NUMBER` in `/etc/modprobe.d/hid-tmt300rs.conf`. As far as I could tell, setting the number to 2 ms didn't seem to crash my wheel as it had done previously, but I'm still going to keep the default period at 8 ms, just to be on the safe side.

Check that everything is alright and we can close this issue.

arabek, 2021-01-08:
I would do the opposite: set it to 2ms and add an adnotation in the readme
that the previous default was 8ms, and that in case it's unstable for
someone, they should tune it up higher. As i've stated in both issues (the
one about missing/invalid effects) the 2ms seems to be the more stable and
less problematic setting overall.

Kimplul, 2021-01-08:
I'm sort of on the fence about it.

You're absolutely right that changing the timer has cleaned up the console spam on your setup, and the increased resolution is definitely a plus. But I'm not really comfortable with the thought that some people's wheels MIGHT crash the first time they're trying it out, and then have to go back and see if there's something they can do about it. I'd prefer to make sure that the wheel works out of the box, and if the user feels the need to tune the wheel then they can do so.

The output you showed in this thread admittedly does speak for the 2 ms setting, but I don't have ETS2 and no other game has had similar issues. That I own, at least.

arabek, 2021-01-11:
lgtm to me

closed 2021-01-11 by arabek