Kimplul/hid-tmff2 #11: Invalid effect types

arabek, 2020-12-21:
While playing Project CARS 2 on Steam (via Proton Tricks) the following spams the console (minus the initialization bit):

```
[62004.545268] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:01.1/0000:01:00.0/usb1/1-5/1-5:1.0/0003:044F:B66E.000B/input/input36
[62004.545358] hid-generic 0003:044F:B66E.000B: input,hidraw9: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:01:00.0-5/input0
[62008.871269] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:01.1/0000:01:00.0/usb1/1-5/1-5:1.0/0003:044F:B66E.000B/input/input37
[62008.871334] t300rs 0003:044F:B66E.000B: input,hidraw9: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:01:00.0-5/input0
[62008.871347] t300rs 0003:044F:B66E.000B: force feedback for T300RS
[62291.174625] t300rs 0003:044F:B66E.000B: invalid effect type: 2038
[62291.174628] t300rs 0003:044F:B66E.000B: failed uploading effects
[62322.996295] t300rs 0003:044F:B66E.000B: invalid effect type: 2038
[62322.996299] t300rs 0003:044F:B66E.000B: failed uploading effects
[62451.303549] t300rs 0003:044F:B66E.000B: invalid effect type: 2031
[62451.303552] t300rs 0003:044F:B66E.000B: failed uploading effects
[62451.561965] t300rs 0003:044F:B66E.000B: invalid effect type: 2031
[62451.561968] t300rs 0003:044F:B66E.000B: failed uploading effects
```

Kimplul, 2020-12-21:
Yeah, I saw some similar error messages with DiRT Rally 2. The game itself worked fine, for some reason the first few effects just apparently had the effect type wrong. Did Project CARS 2 work alright as well?

Could you check with [ffbtools](https://github.com/berarma/ffbtools), namely ffbwrap, what it claims the effects are? If ffbwrap shows `UNKNOWN`, then the driver is working as intended and the game or proton is sending out garbage.

berarma, 2020-12-21:
> Could you check with [ffbtools](https://github.com/berarma/ffbtools), namely ffbwrap, what it claims the effects are? If ffbwrap shows `UNKNOWN`, then the driver is working as intended and the game or proton is sending out garbage.

I'd like to know if this happens, it might be a bug or unsupported feature in Proton.

arabek, 2020-12-22:
> I'd like to know if this happens, it might be a bug or unsupported feature in Proton.

I'd love to help, but is there a guide on how to check that? I've no clue about the tool.

Anyway, after a few more passes this does not seem to be the case anymore, so - maybe the lost packets/timing issue from https://github.com/Kimplul/hid-tmff2/issues/10 had fixed it.

Kimplul, 2020-12-22:
https://github.com/berarma/ffbtools/blob/master/docs/ffbwrap.md shows how to set up logging. So in our case you'd just paste 
`ffbwrap --logger=/home/user/myapp /dev/input/by-id/usb-Thrustmaster_Thrustmaster_T300RS_Racing_wheel-event-joystick -- %command%`

Just replace user with your user and myapp with anything, really.

This issue could of course be some kind of a race condition, and changing the timing could affect the it, but in that case we haven't "fixed" anything, just pushed the issue over to somewhere else.

arabek, 2020-12-22:
Makes sense (the bit about not fixing but pushing the issue somewhere else). I will test it… soon-ish. I'm having too much fun playing right now ;)

arabek, 2020-12-22:
So i've tried running pcars2 with the ffbwrap command enabled, and it came out empty:
```$ cat pcars2-ffbwrap.log-20201222184045.log 
000000000000 # DEVICE_NAME=Thrustmaster Thrustmaster T300RS Racing wheel, UPDATE_FIX=0, DIRECTION_FIX=0, FEATURES_HACK=0, FORCE_INVERSION=0, IGNORE_SET_GAIN=0, OFFSET_FIX=0, THROTTLING=0
```
It seems it initialized properly, but then again, it didn't spit any output. That's the desired behaviour when nothing is being logged out to dmesg too? I mean, there were no errors. Still running the 2ms patched version of the driver. Should i revert?

berarma, 2020-12-22:
Is Steam or the game running in a container?

In the ffbtools issues there are already logs for several games. I think none of them logged unknown effects.

arabek, 2020-12-22:
I'm closing this. I can't reproduce anymore at the `2ms` setting.

closed 2020-12-22 by arabek

Kimplul, 2020-12-23:
Strange, I'll probably investigate this a bit more after the holidays. Sort of odd that the ffbwrap log is empty, it really should be filled with stuff, but that's neither here nor there at this point.

Thanks for your bug reports, they're greatly appreciated.

Kimplul, 2021-01-08:
Sort of interesting, I looked into the unknown effect types but couldn't replicate them under any game/timer combination. There might still be a race condition somewhere, but I just can't find it. The games seem to work alright even when the issue arises, so I think I might leave this issue be for now.

If someone can come up with a reliable way to replicate the issue, I'll investigate further.