Kimplul/hid-tmff2 #12: tmff2 doesn't compile under vanilla kernel due to unknown function

arabek, 2020-12-21:
https://github.com/Kimplul/hid-tmff2/blob/3e03c3d7fe270cea549cac60b7885e1a4aeb75e0/hid-tminit.c#L44

```
$ make
make -C /lib/modules/5.10.1-gentoo-r1/build M=/usr/src/hid-tmff2 modules
make[1]: Entering directory '/usr/src/linux-5.10.1-gentoo-r1'
  CC [M]  /usr/src/hid-tmff2/hid-tminit.o
/usr/src/hid-tmff2/hid-tminit.c: In function ‘tminit_interrupts’:
/usr/src/hid-tmff2/hid-tminit.c:44:5: error: implicit declaration of function ‘kzfree’; did you mean ‘kfree’? [-Werror=implicit-function-declaration]
   44 |     kzfree(send_buf);
      |     ^~~~~~
      |     kfree
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:279: /usr/src/hid-tmff2/hid-tminit.o] Error 1
make[1]: *** [Makefile:1805: /usr/src/hid-tmff2] Error 2
make[1]: Leaving directory '/usr/src/linux-5.10.1-gentoo-r1'
make: *** [Makefile:5: all] Error 2
```

I had to manually change that to ```kfree(send_buf);``` to make it compile cleanly.
gcc:
```gcc version 9.3.0 (Gentoo 9.3.0-r2 p4) ```

Kimplul, 2020-12-21:
That's interesting. Does adding `#include <linux/slab.h>` in `hid-tmt300rs.h` fix the issue?

berarma, 2020-12-21:
`kzfree` has been renamed to `kfree_sensitive`. Isn't `kfree` enough?

arabek, 2020-12-22:
> That's interesting. Does adding `#include <linux/slab.h>` in `hid-tmt300rs.h` fix the issue?

Nope id didn't. It stil bails, at the same function. Indeed grepping for kzfree in `include/linux/slab.h` shows up empty in linux kernel source for 5.10 series. Haven't tested on older kernels.

Kimplul, 2020-12-22:
@berarma Good point, no clue what I was thinking using kzfree. Newest commit in master should work, I just changed kzfree to kfree like arabek.

closed 2020-12-22 by Kimplul