Kimplul/hid-tmff2 #14: can not install on Ubuntu 20.04.2 LTS x86_64

lauccode, 2021-04-05:
**Can not install on(Regolith)  OS: Ubuntu 20.04.2 LTS x86_64 / Kernel: 5.4.0-70-generic :**
It should be my mistake but I can not find it.

**sudo make install** 
make -C hid-tminit install
make[1] : on entre dans le répertoire « /home/coyote/hid-tmff2/hid-tminit »
make -C /lib/modules/5.4.0-70-generic/build M=/home/coyote/hid-tmff2/hid-tminit modules_install
make[2] : on entre dans le répertoire « /usr/src/linux-headers-5.4.0-70-generic »
  INSTALL /home/coyote/hid-tmff2/hid-tminit/hid-tminit.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.4.0-70-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[2] : on quitte le répertoire « /usr/src/linux-headers-5.4.0-70-generic »
make[1] : on quitte le répertoire « /home/coyote/hid-tmff2/hid-tminit »
make -C /lib/modules/5.4.0-70-generic/build M=/home/coyote/hid-tmff2 modules_install
make[1] : on entre dans le répertoire « /usr/src/linux-headers-5.4.0-70-generic »
  INSTALL /home/coyote/hid-tmff2/hid-tmt300rs.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.4.0-70-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1] : on quitte le répertoire « /usr/src/linux-headers-5.4.0-70-generic »
depmod -A

Kimplul, 2021-04-05:
I don't speak french, but this seems very similar to #8. Did you read through that thread?

lauccode, 2021-04-07:
**It seems ok now, thanks**
make -C hid-tminit install
make[1] : on entre dans le répertoire « /home/coyote/hid-tmff2/hid-tminit »
make -C /lib/modules/5.4.0-70-generic/build M=/home/coyote/hid-tmff2/hid-tminit modules_install
make[2] : on entre dans le répertoire « /usr/src/linux-headers-5.4.0-70-generic »
  INSTALL /home/coyote/hid-tmff2/hid-tminit/hid-tminit.ko
  DEPMOD  5.4.0-70-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[2] : on quitte le répertoire « /usr/src/linux-headers-5.4.0-70-generic »
make[1] : on quitte le répertoire « /home/coyote/hid-tmff2/hid-tminit »
make -C /lib/modules/5.4.0-70-generic/build M=/home/coyote/hid-tmff2 modules_install
make[1] : on entre dans le répertoire « /usr/src/linux-headers-5.4.0-70-generic »
  INSTALL /home/coyote/hid-tmff2/hid-tmt300rs.ko
  DEPMOD  5.4.0-70-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1] : on quitte le répertoire « /usr/src/linux-headers-5.4.0-70-generic »
depmod -A

I can see the module :
**ls -lrta /lib/modules/5.4.0-70-generic/ :**
drwxr-xr-x  2 root root    4096 avril  7 10:25 extra
-rw-r--r--  1 root root  610324 avril  7 10:25 modules.dep
-rw-r--r--  1 root root  853356 avril  7 10:25 modules.dep.bin
-rw-r--r--  1 root root 1392414 avril  7 10:25 modules.alias
-rw-r--r--  1 root root 1369725 avril  7 10:25 modules.alias.bin
-rw-r--r--  1 root root     860 avril  7 10:25 modules.softdep
-rw-r--r--  1 root root  614513 avril  7 10:25 modules.symbols
-rw-r--r--  1 root root  747593 avril  7 10:25 modules.symbols.bin
-rw-r--r--  1 root root   10257 avril  7 10:25 modules.builtin.bin
-rw-r--r--  1 root root   24985 avril  7 10:25 modules.builtin.alias.bin
-rw-r--r--  1 root root     330 avril  7 10:25 modules.devname

**But I don't see detection in oversteers ?**

Kimplul, 2021-04-07:
The modules themselves should be in `/lib/modules/5.4.0-70-generic/extra`.

Usually rebooting will automatically load the drivers, but you can try manually running
```
sudo modprobe hid-tminit
sudo modprobe hid-tmt30rs
``` 

If that doesn't work, please include the dmesg output when you connect the wheel.

lauccode, 2021-04-07:
**I get** 
sudo modprobe hid-tminit 
modprobe: ERROR: could not insert 'hid_tminit': Operation not permitted

**Sorry, what do you mean by** ? 
please include the dmesg output when you connect the wheel.

Kimplul, 2021-04-07:
This might be of use with getting modprobe to work:
https://unix.stackexchange.com/questions/543576/modprobe-fails-with-operation-not-permitted

In short, unsigned code, such as modules compiled on your own, is a bit more tricky to install when certain security features are enabled. One possible solution not mentioned in the thread I linked is disabling UEFI secure boot, but depending on your setup that might not be possible.

>please include the dmesg output when you connect the wheel.

First try to get modprobe working, but in case that fails, unplug the wheel and restart your machine. With the device unplugged, run `sudo dmesg -W` and plug in your wheel. This should give you some output, that could indicate if there is some other issue besides the one with modprobe. Also try running the modprobe commands, `sudo modprobe hid-tminit` and `sudo modprobe hid-tmt300rs`, there might be some more information in the log than the error you got.

lauccode, 2021-04-07:
**I could remove UEFI secure boot.
With sudo dmesg -W , I have** 

[  237.237298] usb 3-4: USB disconnect, device number 7
[  248.719672] usb 3-4: new full-speed USB device number 8 using xhci_hcd
[  248.869013] usb 3-4: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[  248.869015] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  248.869016] usb 3-4: Product: Thrustmaster FFB Wheel
[  248.869017] usb 3-4: Manufacturer: Thrustmaster
[  248.871940] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:044F:B65D.0005/input/input28
[  248.872014] hid-tminit 0003:044F:B65D.0005: input,hidraw3: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:00:14.0-4/input0
[  248.888432] hid-tminit 0003:044F:B65D.0005: Wheel with model id 0x206 is a Thrustmaster T300RS
[  248.888590] hid-tminit 0003:044F:B65D.0005: Success?! The wheel should have been initialized!
[  248.888757] usb 3-4: USB disconnect, device number 8
[  249.551673] usb 3-4: new full-speed USB device number 9 using xhci_hcd
[  249.705224] usb 3-4: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[  249.705226] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  249.705226] usb 3-4: Product: Thrustmaster T300RS Racing wheel
[  249.705227] usb 3-4: Manufacturer: Thrustmaster
[  249.707425] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:044F:B66E.0006/input/input29
[  249.707510] t300rs 0003:044F:B66E.0006: input,hidraw3: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:00:14.0-4/input0
[  249.707646] t300rs 0003:044F:B66E.0006: firmware version 0 is too old, please update.
[  249.707648] t300rs 0003:044F:B66E.0006: note: this has to be done through Windows.
[  249.707648] t300rs 0003:044F:B66E.0006: failed creating force feedback device
[  249.707649] t300rs 0003:044F:B66E.0006: t300rs_init failed
[  249.707652] t300rs: probe of 0003:044F:B66E.0006 failed with error -22

Kimplul, 2021-04-07:
Right, the last bit alleges that the firmware that your wheel has installed is too old for the driver. 

You should try to update the firmware, this has to unfortunately be done through Windows. Install the newest Thrustmaster drivers, update the firmware, and check again if you receive the same error message. If you do, my version check is broken and I need to fix it.

lauccode, 2021-04-07:
I can read on the page of hid-tmff2 :
> Warning: There have been reports that this driver does not work if the wheel's firmware version is any other than v. 31
Should it work with newest Thrustmaster drivers ?
I will try to update it.

Kimplul, 2021-04-07:
At the moment, the newest version is 31. Sorry about the confusing wording.

If Thrustmaster for some reason decides to update their firmware and change the API, I'll probably update the driver accordingly. Unfortunately I'm sort of at Thrustmaster's whim, as writing custom firmware for the wheel is probably a bit above my pay grade, and trying to distribute a specific version of their firmware as a binary is, as far as I'm aware, illegal.

lauccode, 2021-04-07:
**I updated the firmware of the wheel.
It seems OK, I will try later.**
[21745.465748] usb 3-4: USB disconnect, device number 14
[21750.146349] usb 3-4: new full-speed USB device number 15 using xhci_hcd
[21750.296848] usb 3-4: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[21750.296851] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[21750.296853] usb 3-4: Product: Thrustmaster FFB Wheel
[21750.296855] usb 3-4: Manufacturer: Thrustmaster
[21750.301294] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:044F:B65D.000C/input/input38
[21750.301640] hid-tminit 0003:044F:B65D.000C: input,hidraw4: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:00:14.0-4/input0
[21750.320817] hid-tminit 0003:044F:B65D.000C: Wheel with model id 0x206 is a Thrustmaster T300RS
[21750.321067] hid-tminit 0003:044F:B65D.000C: Success?! The wheel should have been initialized!
[21750.321188] usb 3-4: USB disconnect, device number 15
[21750.986345] usb 3-4: new full-speed USB device number 16 using xhci_hcd
[21751.135960] usb 3-4: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[21751.135964] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[21751.135965] usb 3-4: Product: Thrustmaster T300RS Racing wheel
[21751.135967] usb 3-4: Manufacturer: Thrustmaster
[21751.138372] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:044F:B66E.000D/input/input39
[21751.138527] t300rs 0003:044F:B66E.000D: input,hidraw4: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:00:14.0-4/input0
[21751.138738] t300rs 0003:044F:B66E.000D: force feedback for T300RS

lauccode, 2021-04-08:
I tried on other laptop which will have the wheel.
But when I do 
```
sudo modprobe hid-tminit
sudo modprobe hid-tmt30rs
```
I can see nothing. Also if i do 'sudo modprobe hid-<TAB>', I can this those both new module are not detected after a reboot. 
The files *.modules are well in /lib/modules/5.4.0-70-generic/extra.

Could it be because of UEFI secure boot ? (but I have no error message with modprobe)

Kimplul, 2021-04-08:
Just to clarify, `/lib/modules/5.4.0-70-generic/extra` should only contain `hid-tminit.ko` and `hid-tmt300rs.ko`. `*.modules` should be in `/lib/modules/5.4.0-70-generic`. The install script should handle everything automatically.

Modprobe needs to know what modules you've installed, and this is usually done with `sudo depmod -A`, which should be done automatically by `make install`. Is there some error message when you run `sudo make install`?

lauccode, 2021-04-08:
`sudo make install`
**seems OK**
make -C hid-tminit install
make[1] : on entre dans le répertoire « /home/cyclope/hid-tmff2/hid-tminit »
make -C /lib/modules/5.4.0-70-generic/build M=/home/cyclope/hid-tmff2/hid-tminit modules_install
make[2] : on entre dans le répertoire « /usr/src/linux-headers-5.4.0-70-generic »
  INSTALL /home/cyclope/hid-tmff2/hid-tminit/hid-tminit.ko
  DEPMOD  5.4.0-70-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[2] : on quitte le répertoire « /usr/src/linux-headers-5.4.0-70-generic »
make[1] : on quitte le répertoire « /home/cyclope/hid-tmff2/hid-tminit »
make -C /lib/modules/5.4.0-70-generic/build M=/home/cyclope/hid-tmff2 modules_install
make[1] : on entre dans le répertoire « /usr/src/linux-headers-5.4.0-70-generic »
  INSTALL /home/cyclope/hid-tmff2/hid-tmt300rs.ko
  DEPMOD  5.4.0-70-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1] : on quitte le répertoire « /usr/src/linux-headers-5.4.0-70-generic »
depmod -A

And I can see the hid-tminit.ko and hid-tmt300rs.ko updated afer
modules.* in extra, updated too.

But it is as if modules was not loaded

Kimplul, 2021-04-08:
Is there any dmesg output when you run `sudo modprobe hid-tminit` or `sudo modprobe hid-tmt300rs`?

lauccode, 2021-04-08:
Nothing

Kimplul, 2021-04-08:
Maybe a dumb question, but is the wheel connected to the computer? I don't get any output when I run `modprobe` without the wheel connected.

lauccode, 2021-04-08:
**maybe it is OK.**
**When I do 'sudo dmesg -w' :** 
[ 4084.693800] [UFW BLOCK] IN=enp4s0 OUT= MAC=01:00:5e:00:00:fb:30:7c:b2:f1:1d:67:08:00 SRC=192.168.1.1 DST=224.0.0.251 LEN=32 TOS=0x00 PREC=0x80 TTL=1 ID=0 DF PROTO=2 
[ 4089.501024] usb 1-1: new full-speed USB device number 11 using xhci_hcd
[ 4089.651213] usb 1-1: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[ 4089.651219] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 4089.651223] usb 1-1: Product: Thrustmaster FFB Wheel
[ 4089.651226] usb 1-1: Manufacturer: Thrustmaster
[ 4089.654598] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:044F:B65D.000D/input/input41
[ 4089.654936] hid-tminit 0003:044F:B65D.000D: input,hidraw8: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:00:14.0-1/input0
[ 4089.673578] hid-tminit 0003:044F:B65D.000D: Wheel with model id 0x206 is a Thrustmaster T300RS
[ 4089.673709] hid-tminit 0003:044F:B65D.000D: Success?! The wheel should have been initialized!
[ 4089.673885] usb 1-1: USB disconnect, device number 11
[ 4090.337051] usb 1-1: new full-speed USB device number 12 using xhci_hcd
[ 4090.486331] usb 1-1: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[ 4090.486337] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 4090.486340] usb 1-1: Product: Thrustmaster T300RS Racing wheel
[ 4090.486344] usb 1-1: Manufacturer: Thrustmaster
[ 4090.489317] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:044F:B66E.000E/input/input42
[ 4090.489682] t300rs 0003:044F:B66E.000E: input,hidraw8: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:00:14.0-1/input0
[ 4090.489838] t300rs 0003:044F:B66E.000E: force feedback for T300RS

Also, oversteer is detecting the wheel. 

**But nothing with sudo modprobe hid-tminit**

Kimplul, 2021-04-08:
>But nothing with sudo modprobe hid-tminit

Shouldn't matter, from the logs you sent I can see that hid-tminit was loaded and it initialized the wheel correctly. You should be good to go, are there any other issues?

lauccode, 2021-04-08:
It is working in game ! thanks a lot.
Very good job to give the possibility to use the wheel under linux.

Kimplul, 2021-04-08:
Glad it works. If there are any other problems or if you have a suggestion of some kind, just open up another issue.

closed 2021-04-08 by Kimplul