Kimplul/hid-tmff2 #15: Exec format error on modprobe

peterlopen, 2021-04-10:
hi @Kimplul ,

thanks for your driver!

I tried to install and use hid-tmff2 and everything went smoothly until last step:
`
peter@dell:~/Projects/hid-tmff2$ sudo modprobe hid_tmt300rs
modprobe: ERROR: could not insert 'hid_tmt300rs': Exec format error
`
Please, any idea what is wrong?

Thanks a lot.

`
peter@dell:~/Projects/hid-tmff2$ uname -a
Linux dell 5.4.0-70-generic #78-Ubuntu SMP Fri Mar 19 13:29:52 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
peter@dell:~/Projects/hid-tmff2$ ls /lib/modules/$(uname -r)/extra
hid-tminit.ko  hid-tmt300rs.ko
`

Kimplul, 2021-04-10:
Full transparency, I haven't bumped into anything like this myself, so any info I can give is more or less just regurgitated from the internet.

Anycase, from what I can find, there are a number of different causes and fixes, see 

[Here](https://github.com/umlaeute/v4l2loopback/issues/288#issuecomment-617030898) for a compiler issue
[Here](https://bbs.archlinux.org/viewtopic.php?id=243372) for some other/out of date package causing issues
[Here](https://www.linuxquestions.org/questions/slackware-14/%27hdaps%27-exec-format-error-4175493731/) for an arch/platform issue

[Here](https://github.com/abperiasamy/rtl8812AU_8821AU_linux/issues/84#issuecomment-165949940) is also a person that claims running `modprobe` with `--force` sometimes works, but it seems somewhat risky. The kernel usually knows when not to load modules, and its opinions should be respected.

Could you run modprobe with the verbose flag (`modprobe -v`) and see if there is any output in the dmesg buffer? That could help narrow things down.

peterlopen, 2021-04-10:
hi, 

my bad. I compiled with older version of kernel, even before buying just to see how likely it is to work :)
looks good now

thanks again for awesome work!
closing

closed 2021-04-10 by peterlopen

Kimplul, 2021-04-10:
Yeah, that'll do it. Glad you got it working, if you have any other issues or suggestions for improvements, just open up another issue, I'd be happy to help.