Kimplul/hid-tmff2 #16: How to set constant/periodic

peterlopen, 2021-04-15:
Hi @Kimplul ,

more of a question than issue - please, how to set constant/periodic effects?
I can see in a code that it should be supported as there are methods `t300rs_upload_periodic` and `t300rs_upload_constant`, but not sure how to change them. 
Oversteer does not even have those two in force feedback tab.

I tried F1 2019 yesterday and while wheel was recognized and worked in menus it did not reacted at all in game, except same force feedback :( Any hint?

Thanks.

peter

Kimplul, 2021-04-15:
I'm assuming you mean the strength of periodic/constant effects? They're usually set by in-game settings, and the master gain sets the overall force. If you'd like to, you can open up an issue in [oversteer](https://github.com/berarma/oversteer) and see if maybe that's a feature that should be added in.

> like pedals were constantly pressed or something :( Any hint?

Did you try remapping the pedals? Some games, particularly games running under Wine, have some slightly weird defaults. I only have F1 2016 and F1 2018, so I can't tell for sure how F1 2019 behaves, but it should just be as easy as finding `"Accelerate"` in the settings menu and pressing the gas pedal. Repeat for however many other pedals you have.

Do you happen to know how F1 2019 feels on Windows? I've used a tool called ffbwrap to look at what commands games send the wheel, and the F1 series feels somewhat floaty to me, but the wheel is, as far as I can tell, correctly reacting to incoming commands.

peterlopen, 2021-04-15:
Thanks for hits. Will try remapping for sure.
Actually, I was trying wheel first on windows just to get a feel how it should behave (and it is also first time I am using wheel). My impressions for F1 2019 - once I used settings from [here](https://www.racedepartment.com/threads/thrustmaster-t300-settings.170485/) - were really good - car reacted fine and I could also "feel" what is happening.
If there is some test I could do with ffbwrap on windows, just let me know, I can try and send results.
Thank you.

peterlopen, 2021-04-17:
hi @Kimplul ,

you were right, remapping helped. Configuration I used was created on windows and pedals were incorrect. Same applied vice versa - when I used configuration with pedals mapping set in linux on windows, it misbehaved. Strange, I would expect game code to produce same commands and wheel to produce/accept them in same way.
wrt F1 2019 feeling - unfortunately, you are right too - it is floaty in linux - like wheel does not have any "weight", does not push back and even does not return to middle by itself. After few laps I re-booted to windows, difference is such big :(
I will try F1 2017 as that one is linux native, and also few other games (Dirt Rally, Assetto Corsa, Project Cars 2) to see if those will feel better.

Thank you

Kimplul, 2021-04-17:
Hey, sorry for slow reply but I tried looking around for some way to compare the commands that F1 2015 sends on Windows vs Linux under wine/natively, but unfortunately it seems like there simply isn't anything like that available on Windows.

From the ffbwrap output I've noticed that F1 2015 only ever uses one effect, a CONSTANT effect. Trying to change the 'Damper'-value in F1's settings doesn't result in any noticeable change, and there's no new effect or anything seen in the ffbwrap output. I think I should try asking Feral Interactive how and what the damper effect should do, or if there's maybe some value that is read from somewhere that then affects how damper is handled? Sorry for the inconvenience, there's a lot I still don't know about force feedback etc.

About the mappings, I remember tinkering with the default mapping of the wheel at some point, but I don't think I ever got it to where all games recognised the correct axes by default. I might try looking into it again, maybe there's something I missed last time.

peterlopen, 2021-04-17:
I did some more tests, results are not very encouraging :( I would appreciate if you can give me some advice.
F1 2017 - wheel not recognized at all. How did you manage to get F1 2015 to recognize it?
Dirt Rally - options to set FFE in game looks differently than those I found on net. Feel was very strange, wheel was doing what it wanted.
Project Cars - looked promising at start, as I could select exact type of a wheel (t300rs). however pedals were not reacting at all. calibrating did not help - any hint here?

Thanks a lot.

arabek, 2021-04-17:
Which FW version is being reported in the windows driver?

Kimplul, 2021-04-17:
I don't have F1 2017 myself, but I do have F1 2015 and 2018, and they both just worked ootb, minus the weighless wheel of course.

DiRT Rally is quite interesting, as I haven't had any issues with that game myself. That game I've tested on both Windows and Linux, when I was dual-booting, and the FFB felt more or less identical to me. Could you explain what you mean by the options in game look different than on the net? Also, what exactly do you mean by 'the wheel was doing what it wanted'?

Project Cars I haven't tested in a little bit, but I redownloaded it and seems to work fine with ffbwrap. Some proton games have issues with effect directions etc, and ffbwrap allows us to fix them, Project Cars 2 unfortunately is one of the games that don't work ootb. I run Project Cars 2 with
```
ffbwrap --direction-fix --update-fix --offset-fix /dev/input/by-id/usb-Thrustmaster_Thrustmaster_T300RS_Racing_wheel-event-joystick -- %command%
```
in the steam launch options for the game.  I think only the `--direction-fix` option is strictly necessary, but the other ones probably don't hurt.

 About the pedals, couldn't replicate. My pedals were apparently correctly mapped from the start, but it might be that Steam saved my settings from last time, not sure.

@arabek good question, though I hope this isn't the issue. I added in a check for firmware version after the last time that was an issue, but it only checks firmwares lower than 31. If Thrustmaster has released a new version that breaks compatibility I'm going to cry.

berarma, 2021-04-17:
Currently, the ffbwrap workarounds shouldn't be needed. They fixed several issues in Proton/Wine and a bug in the kernel.

Proton 4.11-10 fixed all the issues and I think Linux 5.8 fixed the other one.

The need for these workarounds today could be bacuse of some bug in the driver.

Kimplul, 2021-04-17:
Well, I found this:
https://bugzilla.kernel.org/show_bug.cgi?id=208113

But even so, the direction that Project Cars 2 sets is 32768 -> fixp_sin16 returns 0, propagating forward and makes us end up with the level being set to 0. Is there something I'm missing here?

peterlopen, 2021-04-17:
thanks everyone for answers.

@arabek 
[image]

@Kimplul wrt DiRT Rally - FFE options for me in a game look like this 
[image]
In video I was checking they look like this:
[image]
The wheel was very hard to turn, often it turned in different direction as I was pushing it without any apparent reason. I was afraid I could broke it if I will fight against it :)

I do not have version 2 of Project Cars. Will look into this ffbwrap command, if it can help with pedals. In Oversteer pedals work as expected.

Thanks.

Kimplul, 2021-04-17:
That's interesting, I can't remember seeing that menu myself at any point, but maybe they just changed it? I know DiRT Rally had a pretty massive FFB update at some point.

The firmware version is correct, 31 is the newest and technically speaking only version that my driver supports.

Sorry, I was maybe a bit unclear. ffbwrap is a separate topic from the pedals, I have no idea what's going on with them I'm afraid. As @berarma mentioned, any sufficiently new Wine/Proton version should make ffbwrap unnecessary for anything but debugging drivers.

berarma, 2021-04-17:
I don't remember that FFB screen from DR neither. Are you sure your wheel isn't being detected as a controller?

My G29 is identified as a controller in that game unless I set it to emulate a G27 wheel. There's files in Steam to add support for other wheel models.

You can see that the wheel is recognized as a wheel when the control bindings screens shows wheel icons for the wheel axis.

Kimplul, 2021-04-17:
Good question, I'm not sure myself. When I go to 'Control presets', it shows me 'Thrustmaster T300 RS', but when I activate it it shows for example 'Steer left:  LEFT (LS)', which I suppose means left stick?

This actually leads me to think that F1 might think the wheel is a game controller, and doesn't apply any kind of damping since controllers don't even have any way to implement it. How could I test this out and/or make games read my wheel as a wheel? Is it just messing about with the rdesc?

peterlopen, 2021-04-18:
@Kimplul it bahaves exactly same for me in DR. I can select 'Thrustmaster T300 RS', but for steering I still have 'LS'. and no detailed options in FFE settings. I will check on windows (once I figure re-partitioning to get more space for win).
I actually have PC2, sorry for misinformation. I tried ffbwrap but still no pedals in PC2.

@berarma thanks for oversteer. please, is there some reason while periodic and constant effect can not be set in FF tab?

Based on your comments wrt Proton/wine I am wondering is there some API on windows for FFE? So games do not talk to wheel directly but are using this API? I am thinking whether bug in proton/wine implementation of this API could lead to not detecting pedals in PC2 for me.

Thank you.

berarma, 2021-04-18:
> Good question, I'm not sure myself. When I go to 'Control presets', it shows me 'Thrustmaster T300 RS', but when I activate it it shows for example 'Steer left: LEFT (LS)', which I suppose means left stick?

Yes, that means the game will treat it as a controller. A few games have a setting like "FFB type" or "Device type" where you can select "wheel", but most rely on identification of the device and setting the type internally. The FFB on controllers behaves differently and could explain the floaty feeling.

Even if it's showing the device name it doesn't mean that the game knows that it's a wheel.

> This actually leads me to think that F1 might think the wheel is a game controller, and doesn't apply any kind of damping since controllers don't even have any way to implement it. How could I test this out and/or make games read my wheel as a wheel? Is it just messing about with the rdesc?

First look in the game for a FFB or device type setting. If there isn't, Codemasters games usually have some file where it can be set. See this thread for Dirt Rally: https://steamcommunity.com/app/310560/discussions/1/142261027577140785/#c133258593403735863

You may find similar instructions for other games that don't recognize your wheels.

> @berarma thanks for oversteer. please, is there some reason while periodic and constant effect can not be set in FF tab?

It has to be supported by the driver in the same way that the [new-lg4ff](https://github.com/berarma/new-lg4ff) does it. Oversteer only passes the settings to the driver.

> Based on your comments wrt Proton/wine I am wondering is there some API on windows for FFE? So games do not talk to wheel directly but are using this API? I am thinking whether bug in proton/wine implementation of this API could lead to not detecting pedals in PC2 for me.

Modern games are using the DirectInput API on Windows for handling the wheel including the FFB. Newer version of Proton (and I think Wine) use SDL that in turn uses the kernel API. For detection of wheels and inputs I think games rely on internal information they have. In the end, games see a device that can be of type controller or wheel (used for the FFB action only), axes and buttons. You should be able to remap the pedals in the game as with any other axes.

I have issues with some games when I have the wheel and a controller plugged at the same time. I'd try disconnecting any other controller, even graphics tablets, except the mouse and keyboard.

closed 2021-06-10 by peterlopen

peterlopen, 2021-06-10:
closing, with summary of my testing:

Adding definition for T300RS as a wheel device in games where I could find file inputdevices.json (F1 2018, Dirt Rally) did not help.
Although wheel (including FFE) is working in both Dirt Rally and F1 2019, on windows it feels different/better.

Removing other controllers helped - on windows for AC :) No detection of wheel in AC in Linux.

No resolution for pedals detection in PC2.

Kimplul, 2021-06-12:
Could you specify in which ways the wheel felt different/better on Windows? Was it more floaty, weightless, etc? As previously stated, I know F1 is weightless, I tried contacting the devs but never got a response from them, but I'm curious about DiRT Rally.

Can't test AC as I don't have it, but thanks for the heads up.

Pedals not working is strange, what pedals do you have? I've tested the wheel with T3PA and the ones that my wheel came with on PC2, and they both worked.

fuzunspm, 2022-09-07:
I'm willing to buy T3PA or T-LCM, is it going to work via USB?