Kimplul/hid-tmff2 #17: Firmware updated, still not working

MaestroGlanz, 2021-05-12:
I have updated the firmware with the Thrustmaster tool in Win7, but it is still not working in American Truck Simulator on my Regolith (Ubuntu 20.10).

I might be able to provide some information, if you can tell me, what to look for.

Kimplul, 2021-05-12:
Have you/can you try other games, as well as maybe `fftest`? If it's only American Truck simulator that's not working, it could be the same issue I'm seeing with other games, where the wheel is being recognised as a gamepad, and some or all the FF effects are disabled by the game.

berarma, 2021-05-12:
@Kimplul You could take a look at this SDL function. I think the T300RS hasn't made into the list yet:

https://github.com/libsdl-org/SDL/blob/e1db4b82ecf4b2420840b7b31a666563b2656481/src/joystick/SDL_joystick.c#L2049

Then it depends on games or Wine/Proton using SDL to ask for this information. But I think it's the best place to start.

Games like DR had internally defined profiles that they would apply when the associated device was found, but it would use a gamepad profile when there was none.

MaestroGlanz, 2021-05-12:
@Kimplul

That's the result of the sudo make install command.
`
make -C hid-tminit install
make[1]: Verzeichnis „/home/name/_Computer/Thrustmaster/hid-tmff2/hid-tminit“ wird betreten
make -C /lib/modules/5.11.0-7612-generic/build M=/home/name/_Computer/Thrustmaster/hid-tmff2/hid-tminit modules_install
make[2]: Verzeichnis „/usr/src/linux-headers-5.11.0-7612-generic“ wird betreten
  INSTALL /home/name/_Computer/Thrustmaster/hid-tmff2/hid-tminit/hid-tminit.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.11.0-7612-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[2]: Verzeichnis „/usr/src/linux-headers-5.11.0-7612-generic“ wird verlassen
make[1]: Verzeichnis „/home/name/_Computer/Thrustmaster/hid-tmff2/hid-tminit“ wird verlassen
make -C /lib/modules/5.11.0-7612-generic/build M=/home/name/_Computer/Thrustmaster/hid-tmff2 modules_install
make[1]: Verzeichnis „/usr/src/linux-headers-5.11.0-7612-generic“ wird betreten
  INSTALL /home/name/_Computer/Thrustmaster/hid-tmff2/hid-tmt300rs.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.11.0-7612-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Verzeichnis „/usr/src/linux-headers-5.11.0-7612-generic“ wird verlassen
depmod -A
`
This `/sys/bus/hid/devices/XXXX:044F:B66E.XXX/range` is not found in the `/sys/bus/devices directory`.

The file `/etc/modprobe.d/hid-tmt300rs.conf` is also not available in the modprobe.d directory.

Looks like the installation fails at some point.

Kimplul, 2021-05-12:
The SSL errors don't stop the module from compiling, but might stop the module from loading if your distro has disabled unloading unsigned code. See #8 (maybe also #14?) and particularly the StackOverflow thread linked in that issue:
https://superuser.com/questions/1214116/no-openssl-sign-file-signing-key-pem-leads-to-error-while-loading-kernel-modules/1322832#1322832

>This `/sys/bus/hid/devices/XXXX:044F:B66E.XXX/range`is not found in the `/sys/bus/devices` directory

It should be created automatically when the module is loaded properly, you can run `sudo modprobe hid-tmt300rs` and see if there are any error messages.

>The file /etc/modprobe.d/hid-tmt300rs.conf is also not available in the modprobe.d directory

That file has to be created manually, I should maybe specify that in the README.

Kimplul, 2024-11-03:
Doing a bit of winter cleaning, closing this as the original reporter has apparently deleted their account and there are a couple issues in a similar vein anyway.

closed 2024-11-03 by Kimplul