Kimplul/hid-tmff2 #20: Incompatible gcc/plugin versions

exapsy, 2021-06-12:
When I try to `make` or `dkms-install` it says that there are incompatible gcc/plugin versions, specifically referring to `structleak_plugin.so`. The plugin seems to normally exist inside `/usr/lib/modules/5.10.36-2-MANJARO/build/scripts/gcc-plugins` but there seems to be a version mismatch?

```
DKMS make.log for hid-tmff2-0.8 for kernel 5.10.36-2-MANJARO (x86_64)
Sat Jun 12 01:31:40 PM EEST 2021
make -C hid-tminit 
make[1]: Entering directory '/var/lib/dkms/hid-tmff2/0.8/build/hid-tminit'
make -C /lib/modules/5.10.36-2-MANJARO/build M=/var/lib/dkms/hid-tmff2/0.8/build/hid-tminit modules
make[2]: Entering directory '/usr/lib/modules/5.10.36-2-MANJARO/build'
  CC [M]  /var/lib/dkms/hid-tmff2/0.8/build/hid-tminit/hid-tminit.o
cc1: error: incompatible gcc/plugin versions
cc1: error: failed to initialize plugin ./scripts/gcc-plugins/structleak_plugin.so
make[3]: *** [scripts/Makefile.build:279: /var/lib/dkms/hid-tmff2/0.8/build/hid-tminit/hid-tminit.o] Error 1
make[2]: *** [Makefile:1821: /var/lib/dkms/hid-tmff2/0.8/build/hid-tminit] Error 2
make[2]: Leaving directory '/usr/lib/modules/5.10.36-2-MANJARO/build'
make[1]: *** [Makefile:5: all] Error 2
make[1]: Leaving directory '/var/lib/dkms/hid-tmff2/0.8/build/hid-tminit'
make: *** [Makefile:17: hid-tminit] Error 2
```

exapsy, 2021-06-12:
Okay this was a fast one to solve, but good to keep logs of such issues. The issue is that I recently had a kernel update and haven't restarted the computer yet. I'm not sure about the exact problem specifically (maybe it used the new kernel's modules but it was still saying that it was the old kernel?).

 But a restart solved the isuse!

closed 2021-06-12 by exapsy

Kimplul, 2021-06-12:
Seems likely, a kernel update still sets up the different folders and build toolchains for the newer kernel (as far as I can tell), and as such can lead to issues such as the one you stumbled upon. Thanks for the report though, good to have at hand for future reference.