Kimplul/hid-tmff2 #21: Feedback and encountered problems

Raboebie, 2021-07-07:
Hi,

Off the bat, I would like to that you for your efforts on this. It's great to finally have my wheel working nearly perfectly on Linux :)

I could not find a thread indicating which games have been tested so I made a shortlist of tested games and encountered problems.

1. ACC, works perfectly.
2. AC, works perfectly.
3. AMS2, works perfectly. Had some issues with steam input overriding some inputs so had to flip it on and then off then the pedals worked correctly.
4. F1 2020, this one does not work correctly. I can feel some forces like curbs and bumps but there is no weight to the wheel at all. It also seems to apply cornering forces incorrectly making the experience quite strange. Let me know how I can assist you in debugging this issue. I have not tested any other version of the F1 series but may do that later on.
5. BeanNG works just fine as well. No funnies.

berarma, 2021-07-07:
I think F1 2020 has issues on every wheel, it behaves strangely. It might be related to some bug in Wine/Proton.

Kimplul, 2021-07-07:
Hi, thanks for the report.

F1 is unfortunately a known issue, my current leading theory is that it for some reason checks from some database which wheels it's aware of, and the T300RS (at least the configuration under Linux) isn't on it. As such, the wheel is considered by the game to be a gamepad and disables most effects. There was some discussion about it in #17, and I sent a patch to SDL but I don't think it's been released yet. Sort of unprofessional, but I haven't tested if my patch *actually* works, since I'm working over the summer in another part of the country and don't have the wheel with me.

I suppose it could also be that F1 detects some kind of conflict between its database and the wheel's reported featureset. I haven't added the FF_CUSTOM effect, and if the game tries to use it and fails it might go into a 'failsafe' mode of sorts where only `FF_RUMBLE` is supported. I tried contacting Feral Interactive about the issue but haven't gotten a response.

Or, as @berarma pointed out, a bug in Wine. Anyway, in short, F1 is on my todo-list, but it might take me a while to actually get around to it due to not currently being able to test anything. Sorry about that.

Raboebie, 2021-07-07:
Thanks for the explanation. I appreciate it!

Should I close this issue or leave it open for now?

Kimplul, 2021-07-07:
Leave it open for now, as the issue hasn't been resolved and the other discussions about the topic have mentioned it more as a sidenote.

berarma, 2021-07-07:
Please, take a look at this. https://github.com/berarma/oversteer/issues/67

It might be that the game is sending ffb updates too fast. It seems to be a problem for the G920. The user tried to use a workaround in ffbtools but it didn't work for him.

Raboebie, 2021-07-08:
> Please, take a look at this. berarma/oversteer#67

Thanks!

I gave that a go but ffbwrap does not seem to have any effect. Not sure if it’s logitec specific? Tried it with options like inverse on a few games and it didn’t do anything at all.

Oh just adding to my previous post. BeanNG does work just fine, I just had to invert the FFB for some reason. The game has an option for that luckily.

Raboebie, 2021-07-11:
I compiled and install that patched version of SDL you mentioned [here](https://github.com/Kimplul/hid-tmff2/issues/17).

Is there anything I should do after installing it? Does steam not used a packaged version of SDL? Or would it use the system one if it exists?

Raboebie, 2021-07-11:
Success with that patch btw. I built it in 64bit mode and used LD_PRELOAD when running the game. I now have FFB :dagger: 

e.g. of my steam launch command 

LD_PRELOAD=$LD_PRELOAD:~/git/SDL/build/.libs/libSDL2-2.0.so.0.14.1 %command%

Kimplul, 2021-07-11:
Eyyyy, that's cool. Thanks for testing it out, really appreciate it, and thanks @berarma for pointing out the SDL function.

I recommend still leaving this open for now, I'd like to get an understanding of which games use SDL to detect wheels and features. We also noticed that DiRT rally treats the wheel as a gamepad in the settings, but still uses full FFB, and I think Euro Truck Simulator had some similar issues, and they might all be linked to SDL.

Raboebie, 2021-07-11:
Glad I could help!

Just another note, I had to disable steam controller support on the game properties otherwise the throttle would just be stuck at 100% all the time. After turning that off and remapping the pedals everything just worked. Completed two races now and no funnies :)

Kimplul, 2021-07-18:
Right, I got to mess with the wheel this weekend.

+ The SDL patch works in F1 2018.
+ DiRT Rally is unaffected, though this isn't a huge issue as all effects work fine as far as I can tell.
+ F1 2015 running natively is missing wheel weight, other effects seem fine. Running under Proton the wheel weight is there, just implemented as autocentering. Odd, but I guess it works.
+ Euro Truck Simulator 2: Most effects work fine, only issue I'm seeing is with the 'Force feedback damper' setting.

Sometimes Euro Truck Simulator 2 twitches a bit, apparently it generates a single sine pulse that it likes to sometimes play. For some reason, it also sometimes reverses the direction of the pulse, switches again and repeats a couple times. Occasionally, this is accompanied by a pretty big spike in the effect strength, which results in some twitching. Not really sure what I should do about it, since the spike is "visible" in `ffbwrap` logs, but doesn't feel like the devs would've intended something quite so jarring. DiRT Rally 2 also has some slight twitching, but it uses the constant effect and it feels a bit more like it belongs since you're driving on dirt roads with rocks and stuff instead of smooth asphalt. Incidentally, DiRT Rally 2 doesn't twitch on asphalt, so I guess it's intentional?

berarma, 2021-07-18:
> Sometimes Euro Truck Simulator 2 twitches a bit, apparently it generates a single sine pulse that it likes to sometimes play. For some reason, it also sometimes reverses the direction of the pulse, [...]

Can you post the ffb log lines where the pulses happen?

I did a bit of debugging using ETS2 foe the G29 wheel.

Are you using the demo? I think there were differences between the demo and the game some time ago.

Kimplul, 2021-07-18:
Here's one, they're usually pretty similar. If you'd prefer the whole logfile, I can add it to the repo somewhere.

The 'spike' I was talking about is the magnitude rising to about 2000, in this case 1914 at max, with direction changes between almost every packet. The magnitude at 'idle' is about 200-400, and highest peak I've so far seen was about 2700.

This was captured with the full game.
```
000147332664 > UPLOAD id:0 dir:49152 length:70 delay:0 type:PERIODIC waveform:SINE period:70 magnitude:205 offset:0 phase:0 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000147332667 < 0 id:0
000147332669 > PLAY 0 1
000147332670 < 24
000147340508 > STOP 0
000147340519 < 24
000147340523 > PLAY 0 1
000147340526 < 24
000148680575 > STOP 0
000148680594 < 24
000148680599 > UPLOAD id:0 dir:49152 length:70 delay:0 type:PERIODIC waveform:SINE period:70 magnitude:211 offset:0 phase:0 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000148680601 < 0 id:0
000148680603 > PLAY 0 1
000148680605 < 24
000148687626 > STOP 0
000148687639 < 24
000148687643 > PLAY 0 1
000148687645 < 24
000148735782 > STOP 0
000148735796 < 24
000148735800 > UPLOAD id:0 dir:49152 length:70 delay:0 type:PERIODIC waveform:SINE period:70 magnitude:226 offset:0 phase:0 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000148735803 < 0 id:0
000148735805 > PLAY 0 1
000148735807 < 24
000148743829 > STOP 0
000148743840 < 24
000148743844 > UPLOAD id:0 dir:16384 length:70 delay:0 type:PERIODIC waveform:SINE period:70 magnitude:226 offset:0 phase:0 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000148743847 < 0 id:0
000148743848 > PLAY 0 1
000148743850 < 24
000148752090 > STOP 0
000148752106 < 24
000148752114 > UPLOAD id:0 dir:49152 length:70 delay:0 type:PERIODIC waveform:SINE period:70 magnitude:516 offset:0 phase:0 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000148752121 < 0 id:0
000148752125 > PLAY 0 1
000148752130 < 24
000148759477 > STOP 0
000148759489 < 24
000148759494 > UPLOAD id:0 dir:16384 length:70 delay:0 type:PERIODIC waveform:SINE period:70 magnitude:516 offset:0 phase:0 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000148759497 < 0 id:0
000148759499 > PLAY 0 1
000148759501 < 24
000148775869 > STOP 0
000148775885 < 24
000148775891 > UPLOAD id:0 dir:16384 length:70 delay:0 type:PERIODIC waveform:SINE period:70 magnitude:1050 offset:0 phase:0 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000148775895 < 0 id:0
000148775898 > PLAY 0 1
000148775900 < 24
000148783656 > STOP 0
000148783668 < 24
000148783674 > PLAY 0 1
000148783677 < 24
000148796172 > STOP 0
000148796188 < 24
000148796194 > UPLOAD id:0 dir:16384 length:70 delay:0 type:PERIODIC waveform:SINE period:70 magnitude:1540 offset:0 phase:0 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000148796197 < 0 id:0
000148796200 > PLAY 0 1
000148796203 < 24
000148805153 > STOP 0
000148805163 < 24
000148805167 > UPLOAD id:0 dir:49152 length:70 delay:0 type:PERIODIC waveform:SINE period:70 magnitude:1511 offset:0 phase:0 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000148805169 < 0 id:0
000148805171 > PLAY 0 1
000148805172 < 24
[... 49 lines cut ...]
000148878100 < 0 id:0
000148878102 > PLAY 0 1
000148878104 < 24
000148888304 > STOP 0
000148888314 < 24
000148888318 > UPLOAD id:0 dir:16384 length:70 delay:0 type:PERIODIC waveform:SINE period:70 magnitude:873 offset:0 phase:0 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000148888321 < 0 id:0
000148888323 > PLAY 0 1
000148888325 < 24
000148897413 > STOP 0
000148897425 < 24
000148897431 > UPLOAD id:0 dir:49152 length:70 delay:0 type:PERIODIC waveform:SINE period:70 magnitude:873 offset:0 phase:0 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000148897435 < 0 id:0
000148897439 > PLAY 0 1
000148897442 < 24
000148905325 > STOP 0
000148905336 < 24
000148905341 > UPLOAD id:0 dir:16384 length:70 delay:0 type:PERIODIC waveform:SINE period:70 magnitude:328 offset:0 phase:0 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000148905344 < 0 id:0
000148905346 > PLAY 0 1
000148905348 < 24
000148936769 > STOP 0
000148936788 < 24
000148936795 > UPLOAD id:0 dir:49152 length:70 delay:0 type:PERIODIC waveform:SINE period:70 magnitude:245 offset:0 phase:0 attack_length:0 attack_level:0 fade_length:0 fade_level:0
```

berarma, 2021-07-19:
I remember fixing some corner cases with periodic effects, like maintaining the phase when updating an effect that is playing. I think it's not the case here since the effect is stopped before updating.

I guess those effects simulate the irregular rumble of a truck engine. I don't have the time to look at the code but I hope you don't mind that I try to help by throwing some guesses.

ETS2 seems to send ffb updates as fast as it can. Limiting fps by setting vsync on could change the way ffb works.

Also, in the driver, you could check that:

 - The effect duration, period and phase are respected.
 - The play count is respected.
 - The usb kernel command queue isn't full.

You can use `ffbplay` to play log files and debug them. The files can be edited by hand to play just the effects we want.

Kimplul, 2021-07-19:
I don't mind at all, thank you.

Could you elaborate on what you mean by:
> The effect duration, period and phase are respected.

I noticed that the wheel handles negative periodic magnitude to an extent, but manually phase-shifting the wave 180° resulted in slightly less twitching. And by magnitude I'm referring to the magnitude that's sent to the wheel, so effect magnitude scaled by effect direction. Since the effect direction changes 180° in this case, it essentially just switches the magnitude's sign.

Anyway, this technically doesn't respect the phase as requested by the effect upload, but instead more closely resembles what the effect "*intends*".

> The play count is respected.

I'll probably need to double-check this, although in ETS2's case the play count is just one, followed by a stop command. I wonder if I maybe handle frequent start/stop pairs in a dumb way, as at the moment a stop command doesn't clear the `FF_QUEUE_PLAY-bit, so in theory if the commands are close enough to eachother I could be sending a start and immediately after a stop, due to how my current queue-system is implemented. Don't know if it is at all related, but something that came to mind.

> The usb kernel command queue isn't full.

No such warnings in `dmesg`.

Raboebie, 2021-09-17:
Just noting here.

F1 2021 also works great just like 2020. Still using the same launch command as earlier.

@Kimplul Do you know if that SDL patch has made it mainline yet?

Kimplul, 2021-09-17:
It's present in the 2.16 release. Don't know which version your distro ships with, but seems like the patch is getting out there.

tperka, 2022-03-29:
Hello guys,
Has any of you had any success in playing RaceRoom? I mean the force feedback works but seems to be inverted (after car starts it get pulled to the right and wants to stay there) while the other effects are fine. There is a `inverse FFB` option which does not do anything for me. RFactor 2 works perfect out of the box. I've tried disabling Steam input, sadly no luck :(. Anyone with any experience with RRRE?

Kimplul, 2022-03-30:
Hi @tperka, I opened up a separate issue about RRRE, see #39.

Kimplul, 2024-11-03:
Doing some winter cleaning, the SDL patches are presumably widely distributed by now so I think it's fine to close this.

closed 2024-11-03 by Kimplul