Kimplul/hid-tmff2 #22: Hard crash/reboot when disconnected

DGollings, 2021-08-01:
Pop_OS 21.04

Whenever I unplug the steering wheel my entire pc reboots. Any idea where I should start looking as to a possible cause?

Thanks, ACC is working perfectly :)

Kimplul, 2021-08-01:
Hi, thanks for the report.

When the device is removed, `t300rs_remove` is called, so I would/should maybe start looking for issues there. Please attach your `dmesg` log from last boot with `journalctl --boot=-1`, there should be some kind of stack/call trace with some info.

I'll refrain from theorizing too much about what the issue could be until we get some more data about the situation.

DGollings, 2021-08-02:
It appears to have stopped? Hard to be sure, anyway, captured this log:

<details>
  <summary>Click to expand</summary>

      0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009e01fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009e02000-0x0000000009ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000a000000-0x000000000a1fffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000a200000-0x000000000a20bfff] ACPI NVS
[    0.000000] ```BIOS-e820: [mem 0x00000000db5fe000-0x00000000db78dfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000db78e000-0x00000000db911fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000db912000-0x00000000dbd30fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000dbd31000-0x00000000dcc9cfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dcc9d000-0x00000000deffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000df000000-0x00000000dfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fd000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000181f37ffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 3.2.0 present.
[    0.000000] DMI: Gigabyte Technology Co., Ltd. X570 AORUS ELITE/X570 AORUS ELITE, BIOS F4j 08/02/2019
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 3593.206 MHz processor
[    0.000427] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000429] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000433] last_pfn = 0x181f380 max_arch_pfn = 0x400000000
[    0.000436] MTRR default type: uncachable
[    0.000437] MTRR fixed ranges enabled:
[    0.000438]   00000-9FFFF write-back
[    0.000439]   A0000-BFFFF write-through
[    0.000440]   C0000-FFFFF write-protect
[    0.000440] MTRR variable ranges enabled:
[    0.000441]   0 base 000000000000 mask FFFF80000000 write-back
[    0.000442]   1 base 000080000000 mask FFFFC0000000 write-back
[    0.000443]   2 base 0000C0000000 mask FFFFE0000000 write-back
[    0.000444]   3 base 0000DC1B0000 mask FFFFFFFF0000 uncachable
[    0.000445]   4 disabled
[    0.000446]   5 disabled
[    0.000446]   6 disabled
[    0.000447]   7 disabled
[    0.000447] TOM2: 0000001820000000 aka 98816M
[    0.000986] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.001892] total RAM covered: 3583M
[    0.002006] Found optimal setting for mtrr clean up
[    0.002007]  gran_size: 64K 	chunk_size: 64M 	num_reg: 4  	lose cover RAM: 0G
[    0.002095] e820: update [mem 0xdc1b0000-0xdc1bffff] usable ==> reserved
[    0.002099] e820: update [mem 0xe0000000-0xffffffff] usable ==> reserved
[    0.002103] last_pfn = 0xdf000 max_arch_pfn = 0x400000000
[    0.005488] check: Scanning 1 areas for low memory corruption
[    0.005492] Using GB pages for direct mapping
[    0.005837] RAMDISK: [mem 0x2afc5000-0x317d9fff]
[    0.005841] ACPI: Early table checksum verification disabled
[    0.005845] ACPI: RSDP 0x00000000000F05A0 000024 (v02 ALASKA)
[    0.005848] ACPI: XSDT 0x00000000DBCAC0A0 0000BC (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.005852] ACPI: FACP 0x00000000DBCB2FC0 000114 (v06 ALASKA A M I    01072009 AMI  00010013)
[    0.005856] ACPI: DSDT 0x00000000DBCAC1F0 006DCB (v02 ALASKA A M I    01072009 INTL 20160930)
[    0.005859] ACPI: FACS 0x00000000DBD18E00 000040
[    0.005861] ACPI: APIC 0x00000000DBCB30D8 00015E (v03 ALASKA A M I    01072009 AMI  00010013)
[    0.005863] ACPI: FPDT 0x00000000DBCB3238 000044 (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.005865] ACPI: FIDT 0x00000000DBCB3280 00009C (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.005867] ACPI: SSDT 0x00000000DBCB3320 0000FC (v02 ALASKA A M I    01072009 AMI  01072009)
[    0.005870] ACPI: SSDT 0x00000000DBCB3420 008C98 (v02 ALASKA A M I    00000002 MSFT 04000000)
[    0.005872] ACPI: SSDT 0x00000000DBCBC0B8 0033A6 (v01 ALASKA A M I    00000001 INTL 20160930)
[    0.005874] ACPI: MCFG 0x00000000DBCBF460 00003C (v01 ALASKA A M I    01072009 MSFT 00010013)
[    0.005876] ACPI: SSDT 0x00000000DBCBF4A0 009048 (v01 ALASKA A M I    00000001 INTL 20160930)
[    0.005878] ACPI: HPET 0x00000000DBCC84E8 000038 (v01 ALASKA A M I    01072009 AMI  00000005)
[    0.005880] ACPI: SSDT 0x00000000DBCC8520 000024 (v01 ALASKA A M I    00001000 INTL 20160930)
[    0.005883] ACPI: UEFI 0x00000000DBCC8548 000042 (v01 ALASKA A M I    00000002      01000013)
[    0.005885] ACPI: IVRS 0x00000000DBCC8590 0000D0 (v02 ALASKA A M I    00000001 AMD  00000000)
[... 2810 lines cut ...]
[15323.568228] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[15323.568235] rcu: 	3-...!: (0 ticks this GP) idle=6a8/0/0x0 softirq=1060797/1060797 fqs=2  (false positive?)
[15323.568240] 	(detected by 4, t=15002 jiffies, g=1915525, q=229173)
[15323.568242] Sending NMI from CPU 4 to CPUs 3:
[15323.568252] NMI backtrace for cpu 3 skipped: idling at poll_idle+0x9b/0xb9
[15323.569241] rcu: rcu_sched kthread starved for 15002 jiffies! g1915525 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402 ->cpu=3
[15323.569244] rcu: 	Unless rcu_sched kthread gets sufficient CPU time, OOM is now expected behavior.
[15323.569244] rcu: RCU grace-period kthread stack dump:
[15323.569245] task:rcu_sched       state:I stack:    0 pid:   13 ppid:     2 flags:0x00004000
[15323.569249] Call Trace:
[... cut ...]
[15323.569260]  rcu_gp_fqs_loop+0xe5/0x280
[15323.569263]  rcu_gp_kthread+0x85/0xe0
[15323.569264]  kthread+0x12f/0x150
[15323.569267]  ? rcu_gp_init+0x4d0/0x4d0
[15323.569268]  ? __kthread_bind_mask+0x70/0x70
[15323.569270]  ret_from_fork+0x22/0x30
[15383.575014] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[15383.575020] rcu: 	3-...!: (0 ticks this GP) idle=072/0/0x1 softirq=1060797/1060797 fqs=2 
[15383.575025] 	(detected by 4, t=15002 jiffies, g=1915529, q=232036)
[15383.575027] Sending NMI from CPU 4 to CPUs 3:
[15

DGollings, 2021-08-02:
And another one (truncated)

```
[ 9961.444690] usb 3-3: new full-speed USB device number 8 using xhci_hcd
[ 9961.615493] usb 3-3: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[ 9961.615498] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 9961.615499] usb 3-3: Product: Thrustmaster T300RS Racing wheel
[ 9961.615500] usb 3-3: Manufacturer: Thrustmaster
[ 9961.627735] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.3/usb3/3-3/3-3:1.0/0003:044F:B66E.0019/input/input59
[ 9961.627851] t300rs 0003:044F:B66E.0019: input,hidraw18: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:06:00.3-3/input0
[ 9961.629539] t300rs 0003:044F:B66E.0019: force feedback for T300RS
[10857.464905] show_signal_msg: 19 callbacks suppressed
[10857.464908] vrcompositor[369953]: segfault at 0 ip 00005595ac2a79c6 sp 00007fff3e56c680 error 4 in vrcompositor[5595ac1f2000+452000]
[10857.464916] Code: 8b a3 d8 01 00 00 48 8d 83 c8 01 00 00 48 89 45 c8 49 39 c4 0f 84 da 00 00 00 4c 8d 3d 83 98 29 00 eb 62 90 f0 41 83 45 08 01 <48> 8b 3b 48 8b 07 ff 50 18 89 c2 4c 89 f1 4c 89 fe bf 02 00 00 00
[12019.063846] vrcompositor[392125]: segfault at 0 ip 000056318e8629c6 sp 00007ffc61ac9660 error 4 in vrcompositor[56318e7ad000+452000]
[12019.063854] Code: 8b a3 d8 01 00 00 48 8d 83 c8 01 00 00 48 89 45 c8 49 39 c4 0f 84 da 00 00 00 4c 8d 3d 83 98 29 00 eb 62 90 f0 41 83 45 08 01 <48> 8b 3b 48 8b 07 ff 50 18 89 c2 4c 89 f1 4c 89 fe bf 02 00 00 00
[12740.717002] usb 3-3: USB disconnect, device number 8
[12740.797244] hrtimer: interrupt took 631 ns
[12809.674301] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[12809.674307] rcu: 	3-...!: (0 ticks this GP) idle=9c8/0/0x0 softirq=1060797/1060797 fqs=2  (false positive?)
[12809.674312] 	(detected by 4, t=15002 jiffies, g=1914677, q=226667)
[12809.674313] Sending NMI from CPU 4 to CPUs 3:
[12809.674323] NMI backtrace for cpu 3 skipped: idling at poll_idle+0x9b/0xb9
[12809.675312] rcu: rcu_sched kthread starved for 15002 jiffies! g1914677 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402 ->cpu=3
[12809.675315] rcu: 	Unless rcu_sched kthread gets sufficient CPU time, OOM is now expected behavior.
[12809.675315] rcu: RCU grace-period kthread stack dump:
[12809.675317] task:rcu_sched       state:I stack:    0 pid:   13 ppid:     2 flags:0x00004000
[12809.675319] Call Trace:
[12809.675322]  __schedule+0x23d/0x670
[12809.675326]  schedule+0x4f/0xc0
[12809.675327]  schedule_timeout+0x8a/0x140
[12809.675329]  ? __next_timer_interrupt+0x110/0x110
[12809.675331]  rcu_gp_fqs_loop+0xe5/0x280
[12809.675334]  rcu_gp_kthread+0x85/0xe0
[12809.675336]  kthread+0x12f/0x150
[12809.675338]  ? rcu_gp_init+0x4d0/0x4d0
[12809.675339]  ? __kthread_bind_mask+0x70/0x70
[12809.675341]  ret_from_fork+0x22/0x30
[12869.681102] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[12869.681108] rcu: 	3-...!: (0 ticks this GP) idle=e5c/0/0x0 softirq=1060797/1060797 fqs=2  (false positive?)
[12869.681113] 	(detected by 4, t=15002 jiffies, g=1914681, q=204788)
[12869.681115] Sending NMI from CPU 4 to CPUs 3:
[12869.681125] NMI backtrace for cpu 3 skipped: idling at poll_idle+0x9b/0xb9
[12869.682114] rcu: rcu_sched kthread starved for 15002 jiffies! g1914681 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402 ->cpu=3
[12869.682117] rcu: 	Unless rcu_sched kthread gets sufficient CPU time, OOM is now expected behavior.
[12869.682118] rcu: RCU grace-period kthread stack dump:
[12869.682119] task:rcu_sched       state:I stack:    0 pid:   13 ppid:     2 flags:0x00004000
[12869.682122] Call Trace:
[12869.682125]  __schedule+0x23d/0x670
[12869.682129]  schedule+0x4f/0xc0
[12869.682130]  schedule_timeout+0x8a/0x140
[12869.682132]  ? __next_timer_interrupt+0x110/0x110
[12869.682135]  rcu_gp_fqs_loop+0xe5/0x280
[12869.682137]  rcu_gp_kthread+0x85/0xe0
[12869.682139]  kthread+0x12f/0x150
[12869.682142]  ? rcu_gp_init+0x4d0/0x4d0
[12869.682144]  ? __kthread_bind_mask+0x70/0x70
[12869.682146]  ret_from_fork+0x22/0x30
[12929.687916] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[12929.687922] rcu: 	3-...!: (0 ticks this GP) idle=470/0/0x0 softirq=1060797/1060797 fqs=2  (false positive?)
[12929.687927] 	(detected by 4, t=15002 jiffies, g=1914685, q=179816)
[12929.687929] Sending NMI from CPU 4 to CPUs 3:
[12929.687939] NMI backtrace for cpu 3 skipped: idling at poll_idle+0x9b/0xb9
[12929.688927] rcu: rcu_sched kthread starved for 15002 jiffies! g1914685 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402 ->cpu=3
[12929.688930] rcu: 	Unless rcu_sched kthread gets sufficient CPU time, OOM is now expected behavior.
[12929.688930] rcu: RCU grace-period kthread stack dump:
[12929.688932] task:rcu_sched       state:I stack:    0 pid:   13 ppid:     2 flags:0x00004000
[12929.688935] Call Trace:
[12929.688937]  __schedule+0x23d/0x670
[12929.688941]  schedule+0x4f/0xc0
[... 35 lines cut ...]
[12989.695749] rcu: rcu_sched kthread starved for 15002 jiffies! g1914689 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402 ->cpu=3
[12989.695751] rcu: 	Unless rcu_sched kthread gets sufficient CPU time, OOM is now expected behavior.
[12989.695752] rcu: RCU grace-period kthread stack dump:
[12989.695753] task:rcu_sched       state:I stack:    0 pid:   13 ppid:     2 flags:0x00004000
[12989.695756] Call Trace:
[12989.695759]  __schedule+0x23d/0x670
[12989.695763]  schedule+0x4f/0xc0
[12989.695764]  schedule_timeout+0x8a/0x140
[... cut ...]
[13049.701580] Sending NMI from CPU 4 to CPUs 3:
[13049.701590] NMI backtrace for cpu 3 skipped: idling at poll_idle+0x9b/0xb9
[13049.702580] rcu: rcu_sched kthread starved for 15002 jiffies! g1914693 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402 ->cpu=3
[13049.702583] rcu: 	Unless rcu_sched kthread gets sufficient CPU time, OOM is now expected behavior.
[13049.702584] rcu: RCU grace-period kthread stack dump:
[13049.702585] task:rcu_sched       state:I stack:    0 pid:   13 ppid:     2 flags:0x00004000
```

Kimplul, 2021-08-02:
Weird. First of all, I noticed that `hid-tminit` reports an error with code `-2`, whereas it would typically say something like `Success?! The wheel should have been initialized!`. Seems like the wheel still initializes itself, so maybe it's not related and just an unfortunate coincidence?

I can't see any mention of `t300rs_*` functions anywhere, which is unfortunate.

I do see `hrtimer_` popping up here and there, I wonder if it for some reason tries to reschedule itself even after `t300rs_remove` is finished? Don't really know why this would be, but you could try adding a couple `hid_info(t300rs->hdev, "function name");` lines into `t300rs_remove` and maybe `t300rs_play` and `t300rs_timer`. Those are the only functions that manipulate the timer. `t300rs_timer` can restart the timer, depending on its return value, and `t300rs_play` starts the timer if it's not already running.

One thing I'm also noticing now is that it might be a good idea to wrap the `t300rs_destroy` function body in a pair of `spin_lock_irqsave`/`spin_lock_irqrestore` and maybe set freed pointers to null, in case some other parts of the code is still trying to access them after `destroy` is finished.

Checking if the device is closed before `t300rs_remove` is called might also be a good idea. I believe `t300rs_remove` should be the last call anyone makes to the driver, but I've never actually confirmed this.

This also made me notice that I should probably call `input_ff_destroy` if `input_ff_create` fails, but that's probably not relevant here.

If you have the time to test out some of my ideas, that'd be great, otherwise I'll use this comment as a TODO-list for when I can get around to it.  At the moment it looks like that might be next month at the earliest, sorry for the delay.

DGollings, 2021-08-02:
adding some debug output (hid_info) to a few locations and reinstalling should be easy enough. But for the other ideas I'm lacking quite enough of kernel/driver development knowledge. Could hack about a bit but doubt it would be all that useful. Thanks, I'll let you know if it crashes again with the hid_info's

DGollings, 2021-08-03:
just toggled it on and off again several times without problems, I'll close this issue and reopen if I notice any changes. Thanks though

closed 2021-08-03 by DGollings