Kimplul/hid-tmff2 #24: T2PA pedals (from the T300 RS) are incorrectly detected as clutch and brake

ric2b, 2021-08-20:
Hello,

As the title says, I've tried both Project Cars 2 under Proton and the Oversteer aplication and both detect the wheel correctly but neither one has the gas pedal working.

Oversteer shows the T2PA pedals as clutch and brake instead of brake and gas, PC2 only detects the brake. I've made sure that the wheel is in the correct mode, with the red light and PS3 selected. I've also tried to select the green pedal mode but it didn't make a difference.

Do you have any idea what the issue might be? 

I would be happy to try to debug it myself and open a PR with a fix if you could give me some tips on where to get started :)

Kimplul, 2021-08-20:
Hi, thanks for the report. Just to get the obvious solution out of the way, have you tried remapping the pedals?

The pedals work with the HID subsystem, and the wheel's `rdesc` reports the pedals as axes `X`, `Y` and `Rz`. Some games seem to assume `X` is the gas, while others `Y`. I don't think I've seen anyone assume `Rz` is the gas pedal, though I'm not sure if that's because nobody does or I just haven't seen it yet. Either way, depending on which pedal set is connected to the wheel, either two of the axes or all three are used for pedals. I have a working T3PA set which uses all three axes, and I remember the default pedals being correctly mapped in Oversteer, though they later broke and I can't test with them any more. I've usually used Oversteer as the 'correct' indicator, if it thinks my pedals are correctly mapped then everyone who doesn't agree must be wrong.

In short, not entirely sure why your pedals are mapped incorrectly, could be a firmware or wheel revision thing, or maybe even a global remapping in Linux. If it's a wheel revision thing, the 'best' solution would probably be to figure out how to tell different revisions apart and then modify the `rdesc` accordingly, though that needs some reverse engineering.

berarma, 2021-08-20:
Probaby I'm doing it wrong in Oversteer. It remaps the inputs on modern devices while I think it shouldn't be needed. I just took what seemed to be the common way to map them but it seems I looked at some old code. I'll probably change it so most modern devices will work correctly without the need to remap. I'll review it.

Games are another thing though. They should detect the wheel and remap when needed. Or at least allow for manual remapping. Oversteer doesn't help games recognize the inputs.

ric2b, 2021-08-20:
So I used `evtest`, the wheel shows up as "Thrustmaster Thrustmaster T300RS Racing wheel", the brake is mapped as `ABS_Y` (code 1) and the throttle is `ABS_RZ` (code 5). The wheel rotation is `ABS_X` (code 0).

I'm not used to digging into input devices but here's the content of `/sys/bus/hid/devices/0003:044F:B66E.0007/input/input24/uevent`, might be useful to identify the specific model:

```
PRODUCT=3/44f/b66e/111
NAME="Thrustmaster Thrustmaster T300RS Racing wheel"
PHYS="usb-0000:00:14.0-7/input0"
UNIQ=""
PROP=0
EV=1b
KEY=1fff00000000 0 0 0 0
ABS=30063
MSC=10
MODALIAS=input:b0003v044FpB66Ee0111-e0,1,3,4,k120,121,122,123,124,125,126,127,128,129,12A,12B,12C,ra0,1,5,6,10,11,m4,lsfw
```

I'm not sure how to remap the pedals, I've searched around a bit and I'm not finding what tool I should use for that.

ric2b, 2021-08-20:
Just tried WRECKFEST and it lets me remap the controls. Brake is Y axis and Throttle is Z axis according to it. Wheel is X axis. No FFB though.

Raboebie, 2021-08-20:
@ric2b refer to my thread here which might help getting FFB to work #21

ric2b, 2021-08-20:
@Raboebie Was just looking through the FFB related issues and managed to fix it after repeating the steps here: https://github.com/Kimplul/hid-tmff2/issues/5#issuecomment-695219985

So I guess my initial installation based on the repo README.md instructions had some kind of issue.

I'll close this issue since it looks the mapping is a per-application issue and unrelated to the driver.

Thank you all for being so supportive :)

closed 2021-08-20 by ric2b

Kimplul, 2021-08-20:
Good that you got it working. Sorry about getting the pedal maps (`X`, `Y`...) wrong, should've double-checked.