Kimplul/hid-tmff2 #26: Support on Kernel 5.13 and higher

Raboebie, 2021-09-14:
Hello,

Just a quick question. I updated to Ubuntu 21.10 from .04 and this one ships with kernel 5.13 which includes some basic support for the thrustmaster T300.

I was wondering. How do I blacklist that default driver and use this one? It seems to be conflicting somehow. The weel is detected but no FFB.

Kimplul, 2021-09-14:
5.13 might have the init driver, but that should be it. I suspect you're missing the T300 driver for your kernel version, but have the init driver, which causes the wheel to be detected under the correct name but doesn't provide any FFB support.

Try installing the driver through DKMS if you didn't try that already, it should automatically compile all extra modules when it detects a new kernel version. That being said I'm not entirely sure how well it handles distro updates, I should add it to my todo list. If you already installed it with DKMS, take a look in `dmesg` and check which driver the device gets bound to.

Raboebie, 2021-09-14:
Thanks! I tried that already. It has the init driver yeah.

This is what I see in dmesg

`[  168.184021] usb 1-2.4: USB disconnect, device number 6
[  170.745328] usb 1-2.3: new full-speed USB device number 8 using xhci_hcd
[  171.031422] usb 1-2.3: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[  171.031425] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  171.031426] usb 1-2.3: Product: Thrustmaster FFB Wheel
[  171.031427] usb 1-2.3: Manufacturer: Thrustmaster
[  171.088628] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:01.1/0000:01:00.0/usb1/1-2/1-2.3/1-2.3:1.0/0003:044F:B65D.0009/input/input27
[  171.088875] hid-generic 0003:044F:B65D.0009: input,hidraw6: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:01:00.0-2.3/input0
[  171.095621] hid_tminit: disagrees about version of symbol module_layout`

Kimplul, 2021-09-14:
Right, thanks for the `dmesg` output. The init driver in kernel is named `hid-thrustmaster`, so you could probably try adding `blacklist hid-tminit` into `/etc/modprobe.d/blacklist`. Create the file if it doesn't exist. If that doesn't work, `disagrees about version of symbol module_layout` sounds like the kernel is trying to load a version compiled for some older kernel, though I'm not entirely sure why that would be.

Raboebie, 2021-09-14:
Doing that prevents the wheel from being detected all together.

It's not a major train smash. I'll stay on 5.11 for now until the day comes :)

Booting into 5.11 solves the problem naturally.

Kimplul, 2021-09-14:
Interesting, sounds like 5.13 didn't have the init driver in that case. Can you try manually recompiling the driver, just to be sure?

Raboebie, 2021-09-14:
I have. Tried with dkms and doing a manual build and install. Neither seems to revive it for me.

Kimplul, 2021-09-14:
Alright, thanks for the heads up. I'll look into it on my VM.

Kimplul, 2021-09-14:
I was able to replicate it, although for some reason my VM was able to load the init driver just fine but complained about disagreements when trying to load the T300 driver. Anyhow, I got it working by running

```
sudo dkms remove hid-tmff2/0.8
sudo dkms build hid-tmff2/0.8
sudo dkms install hid-tmff2/0.8
```

I don't really understand why it started working, maybe the distro update somehow messed up DKMS? Check if it works for you.

Raboebie, 2021-09-15:
Hmm I also don't know what to tell you now as updating the system and reinstalling the whole driver solved the problem.

There must have been something strange going on after the dist upgrade. I think we can close this now? Issues must have been my side.

Kimplul, 2021-09-15:
Good thing you got it working. If someone finds issues in my DKMS setup that could've caused this, feel free to open up an issue, but for now let's assume it wasn't my fault.

closed 2021-09-15 by Kimplul

Raboebie, 2021-09-15:
Sorry just posting here for clarity on what happened. I think :dagger: 

When I updated I didn't clean the build directory before installing again so I think it installed a version that was still linked to the older kernel headers :)