Kimplul/hid-tmff2 #28: Ferrari F1 Wheel support

ReazerDev, 2021-10-03:
Hi, I've got a T300rs and the (old) Ferrari F1 Wheel attachment.
When using the normal PlayStation attachment, everything's working fine, but whenever I use my F1 Wheel, it's not getting picked up by the driver.
I'm willing to test/help anything you need! :)

Kimplul, 2021-10-03:
Hi, thanks for the report. First of all, could you clarify what you mean by
> not getting picked up by the driver.

Do you mean that the buttons won't work, or that the device doesn't show up at all in `lsusb`? Or maybe something else?

ReazerDev, 2021-10-03:
It shows up as "Thrustmaster FFB Wheel" in lsusb. But in Oversteer it doesn't show op at all

Kimplul, 2021-10-03:
Thanks, that sounds like the attachment causes the wheel to report itself under a different ID than the PlayStation attachment. @scarburato has a table of known IDs over at https://github.com/scarburato/hid-tminit

Could you post `dmesg` output when you connect the wheel? There should be some output from `hid-tminit` in there with the wheel model ID.

ReazerDev, 2021-10-03:
`Unknown wheel's model id 0x203, unable to proceed further with wheel init`

This one?

Kimplul, 2021-10-03:
Yep, thanks. I'll try adding that ID to hid-tminit tomorrow, I'm not home at the moment. If you don't want to wait that long you can try adding `{0x0203, 0x0005, "Thrustmaster T300RS (F1 attachment)"},` to `hid-tminit.h`, around line 48.

Although this does make me think that the lower byte represents possible wheel attachments and the higher byte is the base. So maybe the init model should be changed to only check the base? Not sure.

ReazerDev, 2021-10-03:
How do I build hid-tminit?

ReazerDev, 2021-10-03:
And yeah, I guess the second byte is the base

ReazerDev, 2021-10-03:
Even after adding the new line, I still get unknown model id in dmesg, but interestingly my mouse stops working

Kimplul, 2021-10-03:
The mouse thing is weird, not sure what that's about. Did you run `rmmod hid-tminit` and `modprobe hid-tminit` to refresh the module loaded into the kernel? 
Also, forgot to mention initially but `th_wheels_infos_length` should be changed to 6 after adding the line I mentioned earlier.

ReazerDev, 2021-10-03:
I just rebooted my pc, as I didn't know how to reload the module. And yes, I forgot to change the length to 6🤦🏼‍♂️ I'll try tomorrow

Kimplul, 2021-10-04:
Hi, I forked a version of `hid-tminit` that only looks at the wheel base ID instead of base + attachment. Could you check if it initializes your wheel?

https://github.com/Kimplul/hid-tminit

Should be just `make && make install` to build and install. If you've installed `hid-tmff2` with DKMS, you'll need to temporarily remove the module with `dkms remove hid-tmff2/0.8`. Also if your kernel is newer than `5.14`, you might need to run `rmmod hid-thrustmaster` before installing `hid-tminit`.

ReazerDev, 2021-10-04:
Ah yes, the problem is that I'm on 5.13 and had to do `rmmod hid-thrustmaster`.
It's getting picked up as `Thrustmaster Thrustmaster T300RS Racing wheel` now.

I've got one more question, you can select an 'Advanced Mode' on Windows, which maps a few buttons differently. (There are two joysticks on the F1 Wheel and if in "normal" mode, it treats them as one, if you select advanced, it treats one as a joystick and the other as individual buttons).

I probably have to listen to the USB traffic using Wireshark, select advanced mode on windows and see what happens, right?
Since I've never done anything like that, could you send me a small guide on how I would do it?
(I can send you my Discord if that'd be easier)

ReazerDev, 2021-10-04:
[image]
Oh btw. I just noticed that there is an error

Kimplul, 2021-10-04:
> Oh btw. I just noticed that there is an error

That's unfortunately expected. The wheel breaks the USB spec by completely killing off the connection as soon as it initializes, and that results in an error code. Me and @scarburato discussed whether we should suppress the error, but decided against it, as we couldn't figure out which error codes were 'good' and which were 'bad'.

> I probably have to listen to the USB traffic using Wireshark, select advanced mode on windows and see what happens, right?

Yep, pretty much. I wrote up a quick guide on how I did my captures here:
https://github.com/Kimplul/hid-tmff2/issues/6#issuecomment-720086629

If that's not detailed enough, I'd be glad to hop on to a discord call (or chat, whichever you prefer) and try to answer questions. There's a number of tidbits of information in past issues for this project, but (https://github.com/Kimplul/hid-tmff2/issues/25) is probably the most useful thread at the moment. I'm sure there are also a number of blog posts etc. about USB packet capture, but none that I can see that would directly touch on Thrustmaster devices.

ReazerDev, 2021-10-04:
Nice, thank you! I'll try around a little and if I get stuck, I'll post a comment here and we can hop on a call👍

Kimplul, 2021-10-15:
Should be doable from within native Windows with Wireshark, but the setup is a bit different from what I described earlier. The wireshark wiki has a brief section on it:
https://wiki.wireshark.org/CaptureSetup/USB#Windows

I'm not sure if the USB packet formatting is different depending on which platform is used, but the actual content of the packets should still be the same.

ReazerDev, 2021-10-16:
[t300_advanced_toggling.zip](https://github.com/Kimplul/hid-tmff2/files/7358053/t300_advanced_toggling.zip)
Hi, I hope that this is correct^^
I've included two files, one when I switch from the normal mode to the advanced, and the other is from advanced to normal.

I don't know anything about USB, but I think the switching happens between or around the two URB_FUNCTION_ABORT_PIPE?

Kimplul, 2021-10-16:
> I don't know anything about USB, but I think the switching happens between or around the two URB_FUNCTION_ABORT_PIPE?

I think it might actually happen earlier. When switching from normal to advanced, it seems that the wheel's USB device ID changes from `0xb66e` to `0xb66f`. I suspect that packets 41 in `normal_to_advanced.pcapng` and 39 in `advanced_to_normal.pcapng` are what actually causes the switch to happen, sort of similar to how `hid-tminit` causes the switch from init to this driver.

Should be relatively simple to add in a flag of some kind that causes the driver to send out the mode switch command. I have exams coming up in a couple of days so I'll focus on them for the time being and come back to this issue next week. You're of course free to grok around the source code, just be sure to do it in a VM so you don't accidentally brick your OS.

ReazerDev, 2021-10-16:
I'd try myself, but I don't know anything about drivers and I have no clue what happens in the code haha.

Good luck ony our exams!

Kimplul, 2024-11-03:
Doing some winter cleaning, this was taken care of literal years ago by now but I just forgot to close the issue, so doing it now.

closed 2024-11-03 by Kimplul