Kimplul/hid-tmff2 #30: T3PA + TH8A addons on T300RS base in PS4 mode.

arabek, 2021-11-30:
I know that the wheel base is supposed to be used in ps3 mode, while running on PC, but…

When running so, the TH8A addon needs to be connected to the PC via a dedicated usb cable. 

I wonder if it could be reworked, so that the wheel will be recognised in PS4 mode with all attachements. I could do all the necessary testing. Just point me in the right direction.

Kimplul, 2021-11-30:
I suspect it might not be impossible, though I haven't really studied how the wheel identifies/behaves itself when connected in PS4 mode. If we're lucky it could just be a driver-side change, for example the Windows driver refuses to serve a wheel in PS4 mode but everything else works the same way. On the other hand, we might get unlucky and the firmware has some kind of check to see what kind of host it's connected to, or the PS4 mode behaves completely differently from the PS3 mode and would require a rewrite or maybe even a separate driver module.

If we're unlucky, we would probably need to capture packets between a PS4 and the wheel itself, which would require a physical USB sniffer. They do exist, but are generally pretty expensive. Case and point
https://www.totalphase.com/products/beagle-usb480/

In any case, I'll try snooping around and see if we might be lucky and report back. Could take a while, I have a pretty busy schedule this week, sorry in advance.

Kimplul, 2021-11-30:
Quickly checked out what happens when the wheel is connected in PS4 mode, the wheels shows up with product ID `0xb66d` as opposed to `0xb65d`. When I tried adding this PID to `hid-tminit`, the wheel goes into a kind of bootloop, it disconnects itself but doesn't change ID. Out of curiosity I tried skipping the `hid-tminit` module and attached the device directly to the `hid-tmt300rs` driver and surprisingly, I can play FFB effects just fine. I'm pretty sure the `rdesc` will need some modifications for different attachments to be properly recognised, but more promising start than I was expecting.

arabek, 2021-11-30:
That's great news! Lemme know if you need some grabs of the T3PA and TH8A addons specifically.

Kimplul, 2021-12-01:
Okay, the buttons from the wheel worked just fine but pedals and wheel rotation didn't. Seems that the wheel uses a 54 byte input section marked as `Vendor defined 1` usage page, which means that Linux doesn't know how to interpret the data it gets. I modified the `rdesc` to get input based on what I saw with my own wheel, but I suspect that the different addons might have different regions that they use and which need to be added to the `rdesc`.

I pushed a version to the `ps4` branch that has (for me, at least) working FFB, wheel rotation and pedals. What I still need is to know if/where the addons are mapped. You can probably figure this out by running something like `cat '/sys/debug/kernel/hid/XXX:044F:B66D.XXX/events'`, which should output the raw data as it is captured from the device. You should see a bunch of `GenericDesktop.0000 = 0` along with some other things. With the addons connected, try different inputs and note down which values start changing, and if you can, which value corresponds to which position.

Other than that, there still seems to be one oddity: The wheel doesn't autocenter itself when not being used, as in the PS3 mode. Do you know if it does this on a real PS4 as well?

PS. I might still have the axes of the pedals mixed up, I always get them confused.

arabek, 2021-12-02:
I'll test it over the weekend and het back to you. Cheers!

arabek, 2022-01-17:
I've managed to test this branch over the weekend, and pedals+wheel (including clutch)+ffb works as expected. I've yet to test the events for the shifter addon, but for now this looks good and could be merged as is.

arabek, 2022-01-17:
So the `GenericDesktop.0000 =` value changes to `1,2,4,8,16,32,64,128` respectively for the stick position gears 1-8 (where 8 is reverse basically). Is that enough of an information to get things going?

Kimplul, 2022-01-19:
> So the GenericDesktop.0000 = value changes to 1,2,4,8,16,32,64,128

Good start, seems like it's a fairly typical case of 'bit N on/off' for each gear.  I would still need to know the index to where this value is, a dump of what you're seeing would probably be enough. `GenericDesktop.0000` isn't just one value, it's more of a buffer, and from what I saw with my wheel the pedals were mapped to some index inside this buffer.

arabek, 2022-01-26:
Ok, i think i got it. Should be pretty obvious at this point. And i'm guessing 00 is when the shifter is in neutral, thus, reported anytime a transition between positions 1-8 happened.

```
report (size 64) (numbered) =  01 80 80 80 80 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a5 2f ff ff ff ff ff ff 01 ff ff 00 00 00 00 00 00 00 00 00 00
report (size 64) (numbered) =  01 80 80 80 80 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a5 2f ff ff ff ff ff ff 00 ff ff 00 00 00 00 00 00 00 00 00 00
report (size 64) (numbered) =  01 80 80 80 80 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a5 2f ff ff ff ff ff ff 02 ff ff 00 00 00 00 00 00 00 00 00 00
report (size 64) (numbered) =  01 80 80 80 80 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a5 2f ff ff ff ff ff ff 00 ff ff 00 00 00 00 00 00 00 00 00 00
report (size 64) (numbered) =  01 80 80 80 80 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a5 2f ff ff ff ff ff ff 04 ff ff 00 00 00 00 00 00 00 00 00 00
report (size 64) (numbered) =  01 80 80 80 80 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a5 2f ff ff ff ff ff ff 00 ff ff 00 00 00 00 00 00 00 00 00 00
report (size 64) (numbered) =  01 80 80 80 80 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a5 2f ff ff ff ff ff ff 08 ff ff 00 00 00 00 00 00 00 00 00 00
report (size 64) (numbered) =  01 80 80 80 80 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a5 2f ff ff ff ff ff ff 00 ff ff 00 00 00 00 00 00 00 00 00 00
report (size 64) (numbered) =  01 80 80 80 80 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a5 2f ff ff ff ff ff ff 10 ff ff 00 00 00 00 00 00 00 00 00 00
report (size 64) (numbered) =  01 80 80 80 80 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a5 2f ff ff ff ff ff ff 00 ff ff 00 00 00 00 00 00 00 00 00 00
report (size 64) (numbered) =  01 80 80 80 80 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a5 2f ff ff ff ff ff ff 20 ff ff 00 00 00 00 00 00 00 00 00 00
report (size 64) (numbered) =  01 80 80 80 80 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a5 2f ff ff ff ff ff ff 00 ff ff 00 00 00 00 00 00 00 00 00 00
report (size 64) (numbered) =  01 80 80 80 80 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a5 2f ff ff ff ff ff ff 40 ff ff 00 00 00 00 00 00 00 00 00 00
report (size 64) (numbered) =  01 80 80 80 80 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a5 2f ff ff ff ff ff ff 00 ff ff 00 00 00 00 00 00 00 00 00 00
report (size 64) (numbered) =  01 80 80 80 80 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a5 2f ff ff ff ff ff ff 80 ff ff 00 00 00 00 00 00 00 00 00 00
report (size 64) (numbered) =  01 80 80 80 80 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a5 2f ff ff ff ff ff ff 00 ff ff 00 00 00 00 00 00 00 00 00 00
```

Kimplul, 2022-01-27:
Alright, I modified the `rdesc` of the PS4 mode, top of `ps4` at the moment, 4503f35397f24623ecb286dbea6685a9be67b7d9. The general idea is that there's now eight extra 'buttons', buttons 15-23, with each button being 'on' representing a gear. See if it works, maybe try running `jstest-gtk` and check that buttons light up when you shift into gear.

I'm not entirely sure how games expect shifters to present themselves, so if some games have different default preset mappings, feel free to give me an update (assuming the changes work at all, that is).

arabek, 2022-01-28:
Using js-test all the positions are being recognized as button presses, so at least that's working ok, but…

Seems that windows games in Proton expect this button mapping:
gear:
1: button 14
2: button 12
3: button 11
4: button 13
5: button 15
6: button 17

Also, i noticed that the deadzone on the wheel axis, is quite high, even when setting it to 0 with:
```# /usr/bin/evdev-joystick --evdev /dev/input/by-id/usb-Thrustmaster_Thrustmaster_Racing_Wheel_FFB-event-joystick --deadzone 0
Event device file: /dev/input/by-id/usb-Thrustmaster_Thrustmaster_Racing_Wheel_FFB-event-joystick
New dead zone value: 0
Trying to set all axes deadzone to: 0
Absolute axis 0x00 (0) (X Axis) Setting deadzone value to : 0
(value: 32722, min: 0, max: 65535, flatness: 0 (=0.00%), fuzz: 255)
Absolute axis 0x01 (1) (Y Axis) Setting deadzone value to : 0
(value: 65421, min: 0, max: 65535, flatness: 0 (=0.00%), fuzz: 255)
Absolute axis 0x02 (2) (Z Axis) Setting deadzone value to : 0
(value: 65535, min: 0, max: 65535, flatness: 0 (=0.00%), fuzz: 255)
Absolute axis 0x05 (5) (Z Rate Axis) Setting deadzone value to : 0
(value: 65535, min: 0, max: 65535, flatness: 0 (=0.00%), fuzz: 255)
Absolute axis 0x10 (16) (Hat zero, x axis) Setting deadzone value to : 0
(value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
Absolute axis 0x11 (17) (Hat zero, y axis) Setting deadzone value to : 0
(value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
Absolute axis 0x28 (40) (Miscellaneous) Setting deadzone value to : 0
(value: 128, min: 0, max: 255, flatness: 0 (=0.00%), fuzz: 0)
```

Kimplul, 2022-01-28:
Huh, pretty interesting. Buttons below 14 are already used by buttons on the steering wheel iself, not entirely sure what this means. Does the default mapping overlap at all? And did you get the deadzone fixed?

arabek, 2022-01-28:
Yeah, the deadzone seems to be misreported in jstest (text/console version, don't have the gtk one) - works fine in game. Ignore.

Should i try and make a list of how each button is being reported in game? I can test with Beam.NG, Assetto Corsa and ETS2/ATS. For obvious reasons, only ETS2/ATS can be run both in proton and linux native mode. I could prepare a comprehensive list for that.

Kimplul, 2022-01-28:
> Should i try and make a list of how each button is being reported in game? 

That would be really helpful, yes please.

Kimplul, 2024-11-03:
Doing some winter cleaning, since we got the shifter to a usable state I think it's fine to close this. @arabek if you're still up for that list, feel free to open up a new issue :)

closed 2024-11-03 by Kimplul