Kimplul/hid-tmff2 #32: T300RS not  recognized on Steam/KartKraft

jade2020, 2021-12-30:
Hi,

I'm using 5.14.18-1-MANJARO and have recently installed KartKraft on Steam. My system specs are:
Intel Core i7-10875H
graphics: NVIDIA TU106M [GeForce RTX 2060 Mobile]
nvidia driver: 495.44
RAM: 32GB

My wheel is a Thrustmaster T300RS and is recognized as such when I do lsusb:

`Bus 001 Device 015: ID 044f:b66e ThrustMaster, Inc. Thrustmaster T300RS Racing wheel`

However, the wheel is only recognised as 'generic gamepad' in Steam and I cannot configure the wheel in KartKraft either.

I followed the instructions for installing the linux kernel module but I got errors:

```
[john@ja-clevo hid-tmff2]$ **make**
make -C hid-tminit KDIR="/lib/modules/5.14.18-1-MANJARO/build" 
make[1]: Entering directory '/home/john/GIT/hid-tmff2/hid-tminit'
make -C /lib/modules/5.14.18-1-MANJARO/build M=/home/john/GIT/hid-tmff2/hid-tminit modules
make[2]: Entering directory '/home/john/GIT/hid-tmff2/hid-tminit'
make[2]: *** /lib/modules/5.14.18-1-MANJARO/build: No such file or directory.  Stop.
make[2]: Leaving directory '/home/john/GIT/hid-tmff2/hid-tminit'
make[1]: *** [Makefile:5: all] Error 2
make[1]: Leaving directory '/home/john/GIT/hid-tmff2/hid-tminit'
make: *** [Makefile:17: hid-tminit] Error 2

[john@ja-clevo hid-tmff2]$ **sudo make install**
[sudo] password for john: 
make -C hid-tminit KDIR="/lib/modules/5.14.18-1-MANJARO/build" install
make[1]: Entering directory '/home/john/GIT/hid-tmff2/hid-tminit'
make -C /lib/modules/5.14.18-1-MANJARO/build M=/home/john/GIT/hid-tmff2/hid-tminit modules_install
make[2]: Entering directory '/home/john/GIT/hid-tmff2/hid-tminit'
make[2]: *** /lib/modules/5.14.18-1-MANJARO/build: No such file or directory.  Stop.
make[2]: Leaving directory '/home/john/GIT/hid-tmff2/hid-tminit'
make[1]: *** [Makefile:7: install] Error 2
make[1]: Leaving directory '/home/john/GIT/hid-tmff2/hid-tminit'
make: *** [Makefile:17: hid-tminit] Error 2
Under lib/modules, I don't have a a build directory:

john@ja-clevo hid-tmff2]$ **ls /lib/modules/5.14.18-1-MANJARO/**
extramodules/              modules.alias              modules.builtin.alias.bin  modules.dep                modules.order              modules.symbols.bin        kernel/                    modules.alias.bin          modules.builtin.bin        modules.dep.bin            modules.softdep            pkgbase                    kernelbase                 modules.builtin            modules.builtin.modinfo    modules.devname            modules.symbols            vmlinuz

```
Is there anything else I can do to try and install the drivers or to get the wheel to be correctly recognized?

Thanks,

Kimplul, 2021-12-30:
The errors you're getting seem to point towards not having the kernel headers installed. On arch-based systems it should be a package called `linux-headers`, try installing it and then running `make` again.

jade2020, 2022-01-01:
Thanks!

I tried to install linux-headers but it turns out my kernel 5.14.18 was EOL! I updated my kernel to 5.15.7-1, installed linux-headers and ran make. All appeared to instal though I did get some SSL errors -  

```
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
```
I'll try the wheel and see if it works...

jade2020, 2022-01-02:
I've tried but unfortunately, the wheel is still not recognised in either KartKraft or Automobilista.

Kimplul, 2022-01-02:
Right. First of all, SSL errors are to be expected if you're not trying to sign your own drivers, which I most people aren't.

Second, just to make absolutely sure, you ran `sudo make install` after make, and you have your wheel set to PS3 mode? (small slider on the lower lefthand side of the wheel)

Is the wheel recognized in other games, or software like Oversteer?

jade2020, 2022-01-02:
I only have 2 games: Kartkraft and Automobilista 2. The wheel is not recognised in either. I definitely ran `sudo make install` after make.

In PS3 mode, the device shows as

`Bus 001 Device 015: ID 044f:b66e ThrustMaster, Inc. Thrustmaster T300RS Racing wheel`

In PS4 mode, it shows as

`Bus 001 Device 008: ID 044f:b66d ThrustMaster, Inc. Thrustmaster Racing Wheel FFB`

I've tried both PS3 and PS4 modes but the wheel is still not recognised. Interestingly, in PS4 mode, Automobilista crashes as soon as I launch it (that's repeatable).

jade2020, 2022-01-02:
I think it's working now!!!

I think what's made the difference is selecting PS3 mode and then rebooting the machine with the USB still plugged in. The wheel is now recognised in KartKraft. FFB seems to be working as well. I got a nice judder when I went onto the grass! I'll try Automobilista2 and give an update.

Thanks for all your help!

jade2020, 2022-01-03:
Still crashing in Automobilista2. I'll just use it with KartKraft.

Happy for this issue to be closed.

closed 2022-01-03 by jade2020

Kimplul, 2022-01-03:
Glad you got it working, at least for KartKraft. Thanks for opening up this report, made me realize I should probably add in some more info for people who've never compiled kernel modules before, there have been some number of similar issues opened.

I don't have Automobilista 2 myself, but if you get it working feel free to give me an update. I don't see any particular reason why it shouldn't work with this driver, you could maybe try with different Proton versions but other than that I don't really know what the issue might be, sorry.

jade2020, 2022-01-04:
No worries. Thanks for taking the time to write the module in the
first place!