Kimplul/hid-tmff2 #33: Use pedals separately

hodasemi, 2022-01-15:
Hi,

I got an G29 wheel base and wanted to use the T-LCM pedals separately via USB. Is there a way to do it?

Kimplul, 2022-01-15:
Hi, don't have the pedals myself so I don't know for sure, but if they're halfway sensibly made they should be able to report their axes etc. through standard HID protocols. Did you already try just plugging them in and seeing what happens?

hodasemi, 2022-01-15:
Yea, I tried it. 
Oversteer doesn't report any input. I tried it in F1 2021, there is also no input recorded.

Kimplul, 2022-01-15:
Do the pedals show up as a joystick? If you run `jstest-gtk`, does it show anything?

hodasemi, 2022-01-15:
Yes, the pedals show up. Input is also recorded correctly.

Kimplul, 2022-01-15:
In that case I suspect that the games don't know how to handle input from two separate sources. Not sure how Thrustmaster does it on Windows, maybe they somehow pipe the input from the pedals to the wheel's driver?

hodasemi, 2022-01-15:
So I would close this issue since the device is actually recognized by the system.

closed 2022-01-15 by hodasemi

berarma, 2022-01-15:
Let me chime in. There's currently an issue in SDL that doesn't report devices with only axes as joysticks because it guesses they're accelerometers.

You could create a virtual device that added a fake button and it would work. There's an open issue in Proton's Github related to this.

hodasemi, 2022-01-15:
How do I create such a virtual device and connect the necessary parts to my actual pedal? I wouldn't call me a beginner, but I never toyed with devices at all.

Kimplul, 2022-01-19:
I think this is the issue berarma mentioned:
https://github.com/ValveSoftware/Proton/issues/5126

In the thread, berarma also linked a file that shows how to tell SDL to treat the device as a joystick:
https://github.com/libsdl-org/SDL/blob/d4f2f01580454deef8ac43d8939ebc907d4ad759/README-linux.txt

berarma, 2022-01-19:
Yes, we followed the conversation in an [Oversteer issue](https://github.com/berarma/oversteer/issues/89#issuecomment-1014374008) and forgot to update the thread here.