Kimplul/hid-tmff2 #35: Driver stopped working

kuboslawik, 2022-01-26:
First of all, thanks for this great driver, thanks to it i was able to enjoy my whell on linux for some time. Unfortunately it stopped working yesterday. I installed all my updates yesterday (5.15.16-1-MANJARO), and for sure I used the wheel same day, but i can't recall if it was before or after udpates.

All in all, now it comes up as 'Bus 001 Device 008: ID 044f:b65d  ThrustMaster, Inc. Thrustmaster FFB Wheel', like when i didn't have drivers installed. I reinstalled oversteer (also removed ./home configs) and drivers. Then I tried to manually load drivers (your driver name is hid_tmff?) but after this whell comes up again as 'ThrustMaster, Inc. Thrustmaster FFB Wheel'

Output of lsmod:
hid_tminit             16384  0
hid_thrustmaster       16384  0
hid_tmff               16384  0
ff_memless             20480  1 hid_tmff
joydev                 28672  1

It works with limited functionality as joydev wheel only and doesn't show up in Oversteer.

Driver installation:
[2022-01-26T17:51:07+0100] [ALPM] transaction started
[2022-01-26T17:51:07+0100] [ALPM] installed dkms (3.0.3-1)
[2022-01-26T17:51:07+0100] [ALPM] transaction completed
[2022-01-26T17:51:07+0100] [ALPM] running '30-systemd-update.hook'...
[2022-01-26T17:51:12+0100] [ALPM] transaction started
[2022-01-26T17:51:12+0100] [ALPM] installed hid-tmff2-dkms (0.8-1)
[2022-01-26T17:51:12+0100] [ALPM] transaction completed
[2022-01-26T17:51:12+0100] [ALPM] running '30-systemd-update.hook'...
[2022-01-26T17:51:12+0100] [ALPM] running '70-dkms-install.hook'...
[2022-01-26T17:51:12+0100] [ALPM-SCRIPTLET] ==> dkms install --no-depmod hid-tmff2/0.8 -k 5.14.21-2-MANJARO
[2022-01-26T17:51:15+0100] [ALPM-SCRIPTLET] ==> dkms install --no-depmod hid-tmff2/0.8 -k 5.15.16-1-MANJARO
[2022-01-26T17:51:19+0100] [ALPM-SCRIPTLET] ==> depmod 5.14.21-2-MANJARO
[2022-01-26T17:51:24+0100] [ALPM-SCRIPTLET] ==> depmod 5.15.16-1-MANJARO

Kimplul, 2022-01-26:
> 'Bus 001 Device 008: ID 044f:b65d ThrustMaster, Inc. Thrustmaster FFB Wheel'

This is the device's ID before it is initialized, so it sounds like `hid-tminit` or `hid-thrustmaster` fails to detect it. `hid-tminit` is the name of the external module that comes with this driver, `hid-thrustmaster` is the module that is built into the kernel. The 'actual' driver for the T300RS is `hid-tmt300rs`.

> your driver name is hid_tmff?

`hid_tmff` is a driver for older Thrustmaster wheels, built into the kernel, and I don't have anything to do with it. I know, the naming is a bit of a mess, sorry. The idea was to eventually have the newer wheels under `hid-tmff2` but (at least so far) only the T300RS is implemented under this umbrella. 

Anycase, could you post output of `dmsg` when you plug your wheel in? It might give us a better view of what's going on.

kuboslawik, 2022-01-26:
In `/lib/modules/5.15.16-1-MANJARO/kernel/drivers/hid` only hid-thrustmaster and hid-tmff are present. 

Here is dmesg after reboot:

   25.664690] usb 1-1.3: new full-speed USB device number 8 using xhci_hcd
[   25.874787] usb 1-1.3: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[   25.874798] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   25.874802] usb 1-1.3: Product: Thrustmaster FFB Wheel
[   25.874804] usb 1-1.3: Manufacturer: Thrustmaster
[   26.051552] audit: type=1131 audit(1643219330.229:119): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=user@969 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   26.058918] audit: type=1131 audit(1643219330.236:120): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=user-runtime-dir@969 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   26.471285] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:02.1/0000:16:00.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:044F:B65D.000A/input/input35
[   26.471546] hid-thrustmaster 0003:044F:B65D.000A: input,hidraw9: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:16:00.0-1.3/input0
[   26.493422] hid_tminit: loading out-of-tree module taints kernel.
[   26.493480] hid_tminit: module verification failed: signature and/or required key missing - tainting kernel
[   26.494598] hid-thrustmaster 0003:044F:B65D.000A: Unknown wheel's model id 0x209, unable to proceed further with wheel init
[   33.306824] audit: type=1131 audit(1643219337.482:121): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=systemd-timedated comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   33.331315] audit: type=1334 audit(1643219337.509:122): prog-id=0 op=UNLOAD
[   33.331332] audit: type=1334 audit(1643219337.509:123): prog-id=0 op=UNLOAD
[   33.331337] audit: type=1334 audit(1643219337.509:124): prog-id=0 op=UNLOAD
[   36.037776] audit: type=1131 audit(1643219341.412:125): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=systemd-hostnamed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   36.211535] audit: type=1334 audit(1643219341.589:126): prog-id=0 op=UNLOAD
[   36.211554] audit: type=1334 audit(1643219341.589:127): prog-id=0 op=UNLOAD
[   36.211559] audit: type=1334 audit(1643219341.589:128): prog-id=0 op=UNLOAD
[   38.630886] audit: type=1100 audit(1643219344.005:129): pid=1772 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:authentication grantors=pam_faillock,pam_permit,pam_faillock acct="kuba" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[   38.631129] audit: type=1101 audit(1643219344.005:130): pid=1772 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="kuba" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[   38.632118] audit: type=1110 audit(1643219344.009:131): pid=1772 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[   38.633559] audit: type=1105 audit(1643219344.009:132): pid=1772 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_open grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'

Kimplul, 2022-01-26:
Alright, I think I know what's going on. The wheel model that `hid-thrustmaster` reads is actually model + attachment, i.e. which steering wheel is connected to the base. `hid-tminit` separates the value into two and works 'correctly', and I suspect that before you used `hid-tminit`, whereas this recent update switched to using `hid-thrustmaster`, causing the initialization to fail.

Quick fix should be to add a file `/etc/modprobe.d/hid_thrustmaster.conf` with the content `blacklist hid_thrustmaster` to disallow the kernel from using the internal module, proper fix would be me sending in the patch to the kernel module as well.

kuboslawik, 2022-01-26:
Blacklisting this module fixed the issue.

Kimplul, 2022-01-26:
Alright, glad we got it working, thanks for the report.

closed 2022-01-26 by Kimplul

topjor, 2022-05-14:
Having this same issue, might be useful to add it to the readme so people having this issue know to use the workaround until it is patched upstream

Kimplul, 2022-05-18:
Good idea. I did raise the issue of upstreaming over at https://github.com/scarburato/hid-tminit/issues/11 quite a while ago with no response, I guess I should upstream it myself at some point soon.

scarburato, 2022-05-20:
Uh sorry I completely forgot 😅