Kimplul/hid-tmff2 #36: No FFB for Thrustmaster T248

Draluy, 2022-03-26:
# Compatibility Report
- Name of the game with compatibility issues: Automobilista 2 and Assetto Corsa Competizione
- Steam AppID of the game: 1066890 and 805550

## System Information
- GPU: AMD AMD Radeon RX 5700 XT
- Driver/LLVM version:  (NAVI10, DRM 3.41.0, 5.13.0-37-generic, LLVM 12.0.1)
- Kernel version:  5.13.0-37-generic
- Link to full system information report as [Gist](https://gist.github.com/):  https://gist.github.com/Draluy/9fba014c2bcef2fcb8e89b01051c53be
- Proton version: Experimental or 6.3-8
I am using a Thrustmaster T248.

## Symptoms 

No force feedback in both games, even if the controller is recognized correctly. I can play the game, but there is a uniform resistance of the wheel, with no proper force feedback felt.

## Reproduction

Launch any of the games in proton experimental or 6.3-8, and feel no feedback. I am using a Thrustmaster T248.

Please let me know how I can help to add support for it, I'm not tech savvy enough to implement it on my own and do a pull request :(

Kimplul, 2022-03-26:
Just a sanity check, are you using the `t248` branch?

I'm aware of some games not activating FFB for the wheel (mainly Codemaster games) and I'm still investigating, but `fftest` and so on do detect the device just fine on my machine.

Draluy, 2022-03-26:
Hi, no I wasnt. I switched branches, and now it seems to work!
Thanks!

Kimplul, 2022-03-26:
Cheers. Some notes:

T248 support is still what I consider 'experimental', there shouldn't be any instabilities (but if you encounter any, please open up an issue) but as previously stated, not all games seem to fully work. It seems likely that these issues are external to the driver, but I'd like to make sure. Also, Proton versions older than 7 seem to not recognize the clutch, try to run games with the newest available Proton version when possible.

At some point I'll merge the `t248` branch into `master`, I can ping this thread when that happens. I'll write some more documentation about the wheels and by then Oversteer should have support for the wheel ootb, I just need to send some patches around. Also, there could be some issues with F1 games and the T248, as they use a list of wheels inside SDL2 to determine whether FFB should be activated. I've patched the list with the T248, but it'll take a little while for the patch to reach distros. If you want to, you can try compiling newest SDL from source and use LD_PRELOAD to use your compiled version instead of the system defaults.

Anycase, if you find any issues besides the possible ones I've mentioned here, feel free to open up an issue, they're greatly appreciated.

Draluy, 2022-03-26:
Hi, you are right about the clutch not being recognized. But, with proton experimental, the wheel behaves strangely: there is a lot less resistance when turning it, and there are no road effects or other feedback. FFB doesnt work for me in experimenta not 7.0-1

Kimplul, 2022-03-27:
Which games are behaving strangely? You opened up an issue about Automobilista 2, but do you mean that all games running with newer Proton are behaving the same way or just some games?

Draluy, 2022-03-28:
Yes, sry that wasn't clear.

Kimplul, 2022-03-30:
Sorry, I can't replicate. Assetto Corsa Competizione has fully working FFB with Proton 7.0 for me. Have you tried playing with the settings in game?

Raboebie, 2022-03-30:
Hi, just want to jump in here. On kernel 5.13 and newer you need to make sure the driver is compiled against the correct kernel headers. I had this exact issue. This is because the wheel is detected by the built-in driver but it does not support FFB (or does not have to FFB component yet ). @Kimplul sorry I can remember the exact driver names.

The driver on this repo does support ACC, can confirm this from my side.

Draluy, 2022-03-30:
> Sorry, I can't replicate. Assetto Corsa Competizione has fully working FFB with Proton 7.0 for me. Have you tried playing with the settings in game?

YEs, no luck with that. Maybe @Raboebie  is on to something tho, will have to try.

Draluy, 2022-04-03:
Hi again,

Today i fired Assetto Corsa Competizione with proton 6.3-8, as its the only one that makes that game work with FFB. But, no FFB. So,  I reinstalled the driver, restarted my pc, and the FFB started working again. The driver is compiled against the correct kernel headers.

I can use some effects using fftest (but not all), and the wheel is not detected on oversteer.
Should I reopen the ticket, create a new one?
What logs could I provide?

Draluy, 2022-04-03:
Also, no detection with proton 7.

Draluy, 2022-04-09:
Hi,

My t248 is a PS version.

Here is the output of dmesg -w when I plug the wheel:

```
[ 4784.300131] input: Thrustmaster Thrustmaster Racing Wheel FFB as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:06:00.1/usb1/1-6/1-6:1.0/0003:044F:B696.000A/input/input28

[ 4784.300215] hid-generic 0003:044F:B696.000A: input,hidraw9: USB HID v1.11 Joystick [Thrustmaster Thrustmaster Racing Wheel FFB] on usb-0000:06:00.1-6/input0

[ 4784.419795] input: Thrustmaster Thrustmaster Racing Wheel FFB as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:06:00.1/usb1/1-6/1-6:1.0/0003:044F:B696.000A/input/input29

[ 4784.419864] tmff2 0003:044F:B696.000A: input,hidraw9: USB HID v1.11 Joystick [Thrustmaster Thrustmaster Racing Wheel FFB] on usb-0000:06:00.1-6/input0

[ 4784.448001] tmff2 0003:044F:B696.000A: force feedback for T248
```

Kimplul, 2022-04-09:
Alright, the correct driver is picked up. Have you updated the wheel's firmware? This unfortunately has to be done through Windows with Thrustmaster's tools.

Draluy, 2022-04-09:
No I havent. I'll try with a VM.

Draluy, 2022-04-11:
I updated the wheel , but the results are the same. This looks similar to this issue: https://github.com/ValveSoftware/Proton/issues/5582

Draluy, 2022-04-11:
@Kimplul Good news. After updating the wheel i tried again with proton experimental, and it works on Assetto Corsa Competizione! So I dont know if the new firmware or a new version of the proton experimental did the trick, but the driver is not the problem. :)

closed 2022-04-11 by Draluy

Kimplul, 2022-04-11:
Glad to hear it, always a bit unsatisfying when things 'just start working', but still better than not working at all.