Kimplul/hid-tmff2 #37: Crash of T248 with Automobilista2

Draluy, 2022-03-27:
Hi,

Before installing the driver, I could play automobilista 2, albeit with no FFB.
Now, with the T248 plugged, the game crashes on startup. If I disconnect the wheel, start the game, and reconnect the wheel, the game crashes the moment the wheel is connected. This happens with proton experimental, 7.0 and 6.3.

How can I give you proper info to debug this? Like logs or  smthing.

Kimplul, 2022-03-27:
First and foremost would probably be to check what output `dmesg -w` gives when the game crashes. Altough I have to say it seems somewhat unstable even without the T248, crashed on me when I went to have a look, which might indicate that this is an issue with Proton.

Kimplul, 2022-03-30:
Gave it a go, Automobilista crashes with Proton 7.0 and newer, but 6.3-8 started up nicely. Same behaviour with T300RS, I can see some reports on ProtonDB that Logitech wheels function correctly. Nothing shows up in `dmesg` but from the Proton log I can see that a library called `tm_api_lib_x64.lib` is loaded and from what I can tell immediately causes a backtrace. Seems like some update to Wine/Proton broke this library that guessing from the name is probably used to handle Thrustmaster wheels.

No clue what exactly broke though, the trace is pretty impenetrable to me.

Draluy, 2022-04-16:
Automobilista2 crashes with 6.3, 7.0 and experimental with the wheel attached. Will try to see if I can get any help on the proton repo.

Raboebie, 2022-04-24:
Would like to add on here. I have the same problem with the T300RS. When the wheel is plugged in the game crashes on startup. Unplugged it starts up normally.

Raboebie, 2022-04-24:
@Draluy I found a workaround for now. I installed PC2 just to check if that same lib is referenced and it was not. So I renamed tm_api_lib_x64.lib to tm_api_lib_x64.lib.old launched the game and it worked like a charm :)

Draluy, 2022-04-24:
@Raboebie Thanks, that solved the crashing problem. Now have to figure out why my t248 doesnt have any rumble on that game :)

Raboebie, 2022-04-24:
So this is most likely related to SDL2. @Kimplul would it be required to add the device ID's of the 248 to SDL to make it work?

I know I had this problem in some games not too long ago.

Kimplul, 2022-04-24:
I patched the T248 into SDL a little while ago, but it hasn't been incorporated into a release yet. You could try compiling SDL from master (https://github.com/libsdl-org/SDL)

albfan, 2024-04-02:
For the record: 

- https://github.com/libsdl-org/SDL/pull/4378
- https://github.com/libsdl-org/SDL/pull/5444

Kimplul, 2024-11-03:
Doing some winter cleaning, the patches are probably widely distributed by now and I presume this issue can be closed.

closed 2024-11-03 by Kimplul