Kimplul/hid-tmff2 #38: DiRT Rally 2.0 with T248

Kimplul, 2022-03-30:
DiRT Rally 2.0's FFB doesn't work ootb, but I got it working by adding the line
```
  <device id="{B696044F-0000-0000-0000-504944564944}" name="tm_t248" priority="100" type="wheel" />
``` 
into the games' local file `input/devices/device_defines.xml` and creating a file `input/actionmaps/tm_t248.xml` with the content
```
<?xml version="1.0" encoding="utf-8"?>
<action_map name="tm_t248" device_name="tm_t248" library="lib_direct_input" version="19">
  <axis_defaults>
    <axis name="di_z_axis_rotation">
      <action deadzone="0" name="driving.accelerate" />
      <action deadzone="0" name="menu.navigate.scroll.down" />
      <action deadzone="0" name="replay.forward" />
    </axis>
    <axis name="di_y_axis">
      <action deadzone="0" name="driving.brake" />
      <action deadzone="0" name="menu.navigate.scroll.up" />
      <action deadzone="0" name="replay.rewind" />
    </axis>
    <axis name="di_slider_0">
      <action deadzone="0.2" saturation="0.6" name="driving.clutch" />
    </axis>
    <axis name="di_x_axis">
      <action deadzone="0" name="driving.steer.left" />
      <action deadzone="0" name="driving.steer.right" />
      <action deadzone="0" name="replay.forward" />
      <action deadzone="0" name="replay.rewind" />
    </axis>
  </axis_defaults>
  <group name="driving">
    <action name="accelerate">
      <axis name="di_z_axis_rotation" type="negative" />
    </action>
    <action name="brake">
      <axis name="di_y_axis" type="negative" />
    </action>
    <action name="change_view">
      <axis name="di_button_3" />
    </action>
    <action name="clutch">
      <axis name="di_slider_0" type="negative" />
    </action>
    <action name="handbrake">
      <axis name="di_button_8" />
    </action>
    <action name="headlights">
      <axis name="di_button_4" />
    </action>
    <action name="pause">
      <axis name="di_button_5" />
    </action>
    <action name="reset_vehicle">
      <axis name="di_dpad_0_down" />
    </action>
    <action name="roadside_repair">
      <axis name="di_dpad_0_up" />
    </action>
    <action name="wipers">
      <axis name="di_button_2" />
    </action>
    <group name="gear">
      <group name="sequential">
        <action name="up">
          <axis name="di_button_1" />
        </action>
        <action name="down">
          <axis name="di_button_0" />
        </action>
      </group>
    </group>
    <group name="look">
      <action name="back">
        <axis name="di_button_9" />
      </action>
      <group name="shared">
        <action name="right">
          <axis name="di_dpad_0_right" />
        </action>
        <action name="left">
          <axis name="di_dpad_0_left" />
        </action>
      </group>
    </group>
    <group name="steer">
      <action name="left">
        <axis name="di_x_axis" type="lower" />
      </action>
      <action name="right">
        <axis name="di_x_axis" type="upper" />
      </action>
    </group>
  </group>
  <group name="menu">
    <action name="start">
      <axis name="di_button_5" />
    </action>
    <action name="option">
      <axis name="di_button_11" />
    </action>
    <group name="navigate">
      <action name="back">
        <axis name="di_button_8" />
      </action>
      <action name="down">
        <axis name="di_dpad_0_down" />
      </action>
      <action name="left">
        <axis name="di_dpad_0_left" />
      </action>
      <action name="right">
        <axis name="di_dpad_0_right" />
      </action>
      <action name="select">
        <axis name="di_button_4" />
      </action>
      <action name="up">
        <axis name="di_dpad_0_up" />
      </action>
      <action name="cancel_keybinding">
        <axis name="di_button_12" />
      </action>
      <group name="tab">
        <group name="primary">
          <action name="left">
            <axis name="di_button_0" />
          </action>
          <action name="right">
            <axis name="di_button_1" />
          </action>
        </group>
        <group name="secondary">
          <action name="left">
            <axis name="di_y_axis" type="negative" />
          </action>
          <action name="right">
            <axis name="di_z_axis_rotation" type="negative" />
          </action>
        </group>
      </group>
    </group>
    <group name="context">
      <action name="1">
        <axis name="di_button_9" />
      </action>
      <action name="2">
        <axis name="di_button_2" />
      </action>
    </group>
  </group>
  <group name="replay">
    <action name="exit">
      <axis name="di_button_8" />
    </action>
    <action name="forward">
      <axis name="di_x_axis" type="upper" />
      <axis name="di_z_axis_rotation" type="negative" />
    </action>
    <action name="rewind">
      <axis name="di_x_axis" type="lower" />
      <axis name="di_y_axis" type="negative" />
    </action>
    <action name="playback_step_up">
      <axis name="di_dpad_0_up" />
    </action>
    <action name="playback_step_down">
      <axis name="di_dpad_0_down" />
    </action>
    <action name="pause">
      <axis name="di_button_4" />
    </action>
    <group name="toggle">
      <action name="ui">
        <axis name="di_button_3" />
      </action>
    </group>
    <group name="camera">
      <action name="previous">
        <axis name="di_button_0" />
      </action>
      <action name="next">
        <axis name="di_button_1" />
      </action>
    </group>
  </group>
  <group name="network">
    <group name="toggle">
      <action name="voice">
        <axis name="di_button_11" />
      </action>
    </group>
  </group>
</action_map>
```

I suspect something similar should be possible with DiRT 4, but no success so far.

Kimplul, 2022-09-11:
DiRT 4 seems to assume some set pedal mappings, and for example uses some pedal axes to navigate certain menus. However, this mapping seems to assume inverted input, for example when the gas pedal is completely up the game interprets that to be 100 % gas, and certain menus just scroll infiinitely down. This can be countered to an extent by depressing the gas pedal, though there's still some jitter where the selected element jumps between two adjacent choices.

Blop32, 2022-10-15:
Same menu scrolling issue in DiRT 4 with the T300 RS GT here.
In contrary to the T248, DiRT4 does know the T300 RS with proton (no way to get it runnning in native). It is already included in the `device_defines.xml` and has a its own `tm_t300_rs.xml` in `/input/actionmaps/`.
Didn't manage to permanently remove the scrolling though. Neither through editing the `tm_t300_rs.xml` nor through the in game settings (all menu navigation entries are greyed out). Anyway its the clutch signal that triggers the menu scrolling. Pushing the clutch half way and maintaining it will halt the scrolling. Don't see any jittering here if the clutch is maintained in position. Then manual scrolling with the mouse is possible and adjusting key bindings is possible. Just maintain the clutch half pressed and reassign all key bindings as needed. Annoying but workable... Got it up and running like that.
FFB also works with the T300 and can be tweaked.

Kimplul, 2022-10-16:
Thanks for the report, appreciate it.

> Just a thought: Maybe it is related to the fact that the normal T300 RS does not have a clutch pedal. Only the T300 RS GT version has one?

Could be, though you can buy pedal sets with three pedals. I had the T3PA set, now I'm using the set that came with the T248, but in both cases it was the gas pedal that caused the scrolling. I don't have the original pedals to test with unfortunately.