Kimplul/hid-tmff2 #39: Race Room Racing Experience

Kimplul, 2022-03-30:


ddb1024, 2022-04-03:
I tried RRRE also with the T300.
And see the behavior mentioned by tperka (https://github.com/Kimplul/hid-tmff2/issues/21#issuecomment-1082474483) 
> after car starts it get pulled to the right and wants to stay there

Tried different Proton Versions, Settings... With ge-6.19-2 there isn't this problem but also no FFB at all.

Kimplul, 2022-04-08:
Okay, I got the game to run and seems like the issue seems to be that the game uploads a periodic effect with magnitude 0, i.e. essentially just a fancy constant effect. The direction of the force is handled with the `offset` parameter, although curiously I can only see negative values. I'm not sure but this seems like a Proton bug to me, as I can replicate this behaviour on Windows by applying effects with the same parameters as Proton tries to feed my driver. This coupled with some autocentering probably explains the 'sticking to the right', although I'm not entirely sure what is meant by 'inverted FFB' as all FFB just sort of stops after a couple seconds for me.

tperka, 2022-06-02:
Thanks for providing useful info @Kimplul. Will try to create an issue on proton-ge mentioning issues you found. You can find some details about 'inverted FFB' here: https://forum.sector3studios.com/index.php?threads/cant-drive-in-a-straight-line.7674/

Generally, it seems like it could work if instead of pulling to one side it would pull to the center as expected

Raboebie, 2022-07-29:
Perhaps #46  could help you out here? Would be curious

fuzunspm, 2022-08-25:
> Okay, I got the game to run and seems like the issue seems to be that the game uploads a periodic effect with magnitude 0, i.e. essentially just a fancy constant effect. The direction of the force [...]

I have the same sticking to the right issue

Kimplul, 2022-09-10:
Partial update: https://github.com/Kimplul/hid-tmff2/issues/47 led me to find out I had a bug in how I handled durations of effects. I've patched the issue and

> all FFB just sort of stops after a couple seconds for me.

seems to also have been fixed by the patch. The game still pulls heavily to the right, but my previous observation doesn't really seem to apply anymore, as testing RRRE again today it looks like the offset is kept at a constant `-1`. More research is needed.

Kimplul, 2022-09-10:
Also, does anyone know how you're supposed to interpret the in-game FFB meter? While driving it shows just a constant red line at the top of the graph, but while stopped it drops down to midlevel with some noise added to it, guessing engine rumble.

If the line show how much force should be applied left/right, then it seems like the game itself is calculating the constant pull to the right, but that doesn't really make sense. Haven't tried the game on Windows, so I don't really know how the graph *should* look like.

Kimplul, 2022-09-10:
For reference, here's what happens when I try to drive, keeping the wheel more or less straight:

[image]

Here's what happens when I stop and let go of the wheel:
 
[image]

And here's a video from someone presumably running the game on Windows where you can see a normal FFB meter:
https://www.youtube.com/watch?v=s3tuw4OdOo8

The FFB seems to be at least partially determined by the wheel's position, but of course I don't know the internals of the game. Could be that there's a bug in the Proton subsystem that the game tries to use, leading to nonsencial FFB calculations that are then passed on to the driver.

tperka, 2022-09-26:
As mentioned [here](https://github.com/ValveSoftware/Proton/issues/1321#issuecomment-1257279521) by @berarma, the game seems to be working with Logitech wheels. Could you please provide me with some tutorial on how to set up debug environment? I'm a programmer and I think I could try to help in my free time, I'm really keen to make it work in RRRE as it's my goto sim besides rFactor 2. @Kimplul

Kimplul, 2022-09-27:
Debugging effects that are sent to the driver can be done through `ffbwrap` from [here](https://github.com/berarma/ffbtools). Just add something like this in the Steam launch options:
```
$PATH_TO_FFBWRAP/ffbwrap --logger=/tmp/rrre /dev/input/by-id/usb-Thrustmaster_Thrustmaster_T300RS_Racing_wheel-event-joystick  -- %command%
```

You should see a file with all effects sent to the driver in a file starting with `/tmp/rrre`, followed by a timestamp.

Debugging the driver is probably best done through `hid_info` or `pr_info` or similar calls, you *can* run the kernel under `gdb` but it's a pretty complicated process and I didn't find it to be of much use when doing module-level debugging. Then you just open up `dmesg -w` and read whatever output shows up.

Debugging Proton is new to me as well, I know there are a number of debug flags you can give through `WINEDBG`, and apparently there's a `winedbg` program.

I'm not sure what a good starting point would be, as the effects that are sent to the device seem to make sense to me (as in, values that are sent create the effects that they should). The in-game representation of the FFB seems to go haywire, so there might be some functionality that `tm_api_lib` provides that the driver doesn't implement correctly or at all, or whose implementation runs into some unimplemented/buggy part of Wine, don't know.

What was sort of curious last time I had a look at `ffbwrap` logs was that the effects weren't at a 90 degree angle, rather something like 90.3 (approximately, uint16_t: 16384 would be 90 but I read something like 16589, need to double check at some point) which I don't really understand. Could be related.

If you want to try to upload effects to the wheel yourself, try https://github.com/Kimplul/ffbsdl. It uses the SDL wrapper around Linux's native `ioctl` mechanism so it might not be 100% accurate, but probably a good starting point. I haven't actually looked at the source so I don't know where or if there are any discrepancies between the SDL and native interpretetation of effects.

fuzunspm, 2023-04-01:
any progress about this, it's still sticking to the right

Kimplul, 2023-04-05:
@fuzunspm 

Not really, at least as far as I'm aware. I tried double checking my previous findings but wasn't able to launch a race, regardless of which proton version I used. Might be a temporary server issue. I'll maybe try again later, but from your comment it seems that https://github.com/Kimplul/hid-tmff2/issues/39#issuecomment-1242765056 is still about as much as I know about the issue.

Vivien-lelouette, 2023-09-28:
I can confirm that the issue is still here :cry:. Does anybody managed to make the T300RS work with RRE ?

MicHaeL-MonStaR, 2024-06-16:
Unfortunately the same seems to happen with the T248, in the middle of June with the current latest Proton, as well as ProtonGE (not that it needs to be Proton causing it, but just clarifying). - When I started the game initially, when in-game (on-track or in-race), the wheel would just turn in a weird pattern, but also it would not even respond to turning the wheel and the car would be steering to the right constantly. - Then I decided to just restart the game and see what happens. And while the behavior changed, it seemed to be like what is described here, that it would just pull to the right constantly and also do a short pattern, though it WOULD respond to turning the wheel.
Using the "reversed" setting also didn't do anything. - I could however simply use the wheel and pedals without FFB at all, but of course that doesn't help much. So it's very much the FFB being the issue, not the inputs.

Someone's FFB-guide, who probably used it through Windows, describes that this game recognizes the T248 as the TS-PC base and that supposedly this is intentional for this wheel to have compatibility with older games. - And while it does show up that way, it doesn't really seem to help.

Not sure if this adds anything, but there's also a weird behavior where, even if you choose/activate one of the controller-profiles/presets in the settings, and then change bindings, for some reason it will just make yet another profile without choosing to make a duplicate yourself. - I'm not sure if that would happen with using any other controller (like keyboard or gamepad), but it's just weird, almost as if it sees it as a new instance of the same controller every time you re-bind anything.

I didn't think it would be such an issue with this game, as most of them seem to work in the end. - I wanted to give RR a try today, but I guess back to rFactor for now.

kiiro94, 2024-11-26:
Same happens to me with a T300, the wheel pulls to the right.

FrFloyd, 2025-06-21:
Hello, same problem here with a T300 ferrari alcantara edition. my wheel goes violently to the maximum right, is shaken while remaining at this level when I start a race. gaming is impossible. is there a known solution ? I post a thread here : https://forum.kw-studios.com/index.php?threads/linux-mint-22-1-xia-thrusmaster-t300-is-forcing-at-right.20533/ thanks.

FrFloyd, 2025-06-22:
:(
I put a lot of money in raceroom so I'll prefered it works fine :(
Automobilista 2 cost a lot of money....there's a demo that I'll try.
I will retest raceroom with GE-proton 10.4 to see if it works better....
then I'll try AC.

FrFloyd, 2025-06-22:
yes I recently go on linux and i had r3e on windows.
i'll follow the price of the base of ams2 ;)

MmAaXx500, 2025-07-22:
### TLDR

The good news: I got it working
The bad news: Only to some extent and requires wine/proton patching

---

### What's working?

I had the same experience as people in this issue: The wheel is sticking to one side on start and every movement of the wheel causes strong jerks.

With patches or dll injection, I was able to make it work. It does not try to pull itself out of my hand and seems to behave the same as on Windows. To have similar experience, you have to change the max wheel range to match the car's and limit the global force strength to ~40% in oversteer. On Windows, this is done by using hardware specific commands (more on that later).

### What's not working?

For some unknown reason, the sine effect (used as the fancy constant force) stops after about a minute.

### How can I try it out?

In case you just want to try it out, the preferred and probably the easiest method is by using my precompiled [release from my proton-ge-custom fork](https://github.com/MmAaXx500/proton-ge-custom/releases) the [install instructions are in the readme](https://github.com/MmAaXx500/proton-ge-custom/tree/tm-rrre-patches?tab=readme-ov-file#native).
The patch is living on the [`tm-rrre-patches` branch](https://github.com/MmAaXx500/proton-ge-custom/tree/tm-rrre-patches) if you want to [compile it yourself](https://github.com/MmAaXx500/proton-ge-custom/tree/tm-rrre-patches?tab=readme-ov-file#building).

The other method is more suitable to someone who wants to dig deeper in the issue (or other issues). To have some reliable way to compare Windows' and Wine's behavior, I made [DInput Detour](https://github.com/MmAaXx500/dinput-detour) that logs most of the interesting DirectInput methods and its parameters. DInput Detour uses dll injection to intercept and potentially modify the calls. It was useful to me to quickly iterate between the various ideas without recompiling proton. By default, it just logs the calls, so I made a branch [`rrre-escape`](https://github.com/MmAaXx500/dinput-detour/tree/rrre-escape) that emulates a behavior that is more acceptable to RRRE. The usage instructions are in the readme.

### What was the issue?

The DirecInput API provides a way to execute hardware specific commands. These are the [`IDirectInputDevice8::Escape`](https://learn.microsoft.com/en-us/previous-versions/windows/desktop/ee417891(v=vs.85)) and [`IDirectInputEffect::Escape`](https://learn.microsoft.com/en-us/previous-versions/windows/desktop/ee417941(v=vs.85)) methods. Using these commands, you can execute various things like query max/current wheel range, or change the wheel's range or force strength.

Because Wine doesn't have this call implemented (especially the `IDirectInputDevice8` one) and just returns "success" (`DI_OK`) RRRE assumed wrong wheel parameters and caused incorrect calculation somewhere inside the physics engine. Fortunately, returning "not supported" (`DIERR_UNSUPPORTED`) solved the main issue. Of course, using that solution not solves that the wheel's range is not changed to the car's and the force strengths are not changed automagically either, that's why it needs to be done manually.

Because this comment is quite long already and discussing these hardware specific commands is another can of ~~worms~~ features I opened a new issue about it over there: _issue pending_

### Next steps?

Well, I have no idea what is the current flow of the Wine/Proton patches and where should I submit them. So, if you happen to know, please share with me.

But before any patch is sent anywhere, I would like to first fix up RRRE to be in a playable state to know that the patches are correct, so testers are welcome.

MmAaXx500, 2025-07-23:
> For some unknown reason, the sine effect (used as the fancy constant force) stops after about a minute.

Some more details on that topic. The effects sent from the game _seems_ to be fine. When this driver (hid-tmff2) receives the periodic effect to be created, it has the duration of `0` and that's sent as `0xffff` to the wheel. That's ok.
However, when an update is received to the same periodic effect it has the duration of `0xffff` and that's sent as `0xfffe` and that is roughly 65 seconds.

Currently, I don't know where is the issue in the `wine -> sdl -> hid-tmff2` chain.

Kimplul, 2025-07-24:
Hi @MmAaXx500, thanks for looking into this issue, looks like you've made some great progress. I'll definitely try out your patched Wine, but I probably won't have time for that for at least a week or so unfortunately.

In the meantime, I wanted to ask regarding:

> However, when an update is received to the same periodic effect it has the duration of 0xffff and that's sent as 0xfffe and that is roughly 65 seconds.

Can Wine dump the [DIEFFECT](https://learn.microsoft.com/en-us/previous-versions/windows/desktop/ee416616(v=vs.85)) structure during a `SetParameters`? If not, I guess you can monkeypatch your Wine fork to do so. Could give us an indication if the `0xffff` is due to `dwDuration` actually requesting a (long, but still) finite length effect, or if there's maybe some code path that's a bit buggy in Wine.

FrFloyd, 2025-11-28:
https://github.com/Kimplul/hid-tmff2/issues/39#issuecomment-3104689106

hello @MmAaXx500 , sorry for my english.

I would have liked to test your patch, but I'm new to Linux and I admit I don't understand anything on your instructions for installing it. :(

and if it doesn't work, I don't know how uninstall the patch...

I installed Steam with Logiteque of my linux mint zara
I install proton ge custom 10.25 with protonup-qt (https://github.com/DavidoTek/ProtonUp-Qt) and enable it in steam

i download your "zip" then now.....don't know what i must do.

FrFloyd, 2025-11-28:
yes ! now I understand. :)
I will test it when I'll have time.

THANKS !

edit : I have directory "compatibilitytools.d"

[image]

FrFloyd, 2025-11-28:
raceroom load if I select DXVK (don't know what is it and why it works).

and your patch works !

my wheel don't come alone to the right now !!!! thanks !

must find how configure oversteer and the ffb in the game now. lol

FrFloyd, 2025-11-28:
`It seems I forgot to rename 😅, whoops`
lol :)
`Yes, the DXVK version is the one for Linux. (DXVK is a DirectX-Vulkan translation layer to be used under Wine/Proton)`
ok. :)
last time I load the game (most month before now), it works without choosing DXVK
`Glad you got it working!`
👍 
`I just checked ProtonDB to see if the game is currently broken, and what I found is another confirmation that the patched proton is working. I need to figure out which one is the appropriate upstream to send the patch to.`
> <img alt="Image" width="1072" height="257" src="https://private-user-images.githubusercontent.com/36025825/520264006-f9ddf7ae-b370-467c-8c9e-43ae2b65c19c.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjQzNTQyNjYsIm5iZiI6MTc2NDM1Mzk2NiwicGF0aCI6Ii8zNjAyNTgyNS81MjAyNjQwMDYtZjlkZGY3YWUtYjM3MC00NjdjLThjOWUtNDNhZTJiNjVjMTljLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTExMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMTI4VDE4MTkyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE4NDBkMzdlZGE4MmFmYTYyN2I5ZTQyMGYyNDM0OWYwYWRlNDY2MTFmOWU5OTlkN2E1ODA5Y2E2ZDU2YmRiOTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.YIK-e6w4DIhAYvRjM7WhUIj6hOSr7AkKrQWGEhJ72ck">

possible to implement it directly in the custom proton...10.26 ?  😅

FrFloyd, 2025-11-29:
that's it....it works, so it'll be "cool" :)

fuzunspm, 2025-11-29:
@FrFloyd 
> edit : what the option they all have ??
> 
> `mangohud %command% -webHudUrl=https://sealhud.github.io/` `MANGOHUD=1 gamemoderun %command%` `DXVK_FRAME_RATE=145 MANGOHUD=1 gamemoderun %command%`

WebHudUrl: Enables a custom overlay. Requires dash.exe (run after game launch like: `STEAM_COMPAT_DATA_PATH="/YOUR_STEAM_LIBRARY_PATH/SteamLibrary/steamapps/compatdata/211500" ~/.steam/root/compatibilitytools.d/rrre-escape-v1/proton runinprefix ~/Downloads/dash/dash.exe`). Note: You must disable the native game UI settings for the overlay to work correctly, More info: [https://github.com/sealhud/sealhud.github.io](https://github.com/sealhud/sealhud.github.io)

and yes game is working perfectly with @MmAaXx500 patch, that protondb report is mine

MmAaXx500, 2026-01-06:
I tested ETS2 and BeamNG with my patched proton and there were no issues. I was unable to get SnowRunner running, so I was unable to test it.

However, when I tried to test RRRE I noticed that FFB works even without my patch. I tested with Proton-GE 10-10, that is the base version of my patched Proton, and with Proton Experimental. Checking the game changelog, it seems that somehow the issue was noticed. Two entries from the [4th December changelog](https://store.steampowered.com/news/app/211500/view/514102575975367489):
> - Updated to latest Thrustmaster SDK [...]
> - Force Feedback - Forces are now consistent no matter the steering range selected in car setup. Fixes FFB becoming really strong and compressed with low values of steering range.

So until someone reports that they need my patch to have working FFB in RRRE I will not start the upstreaming as it is pointless right now. I'm not aware of any other game that needs this patch.

---

@cazzoo Sorry, I completely missed your edits because GitHub does not send emails about edits. Thanks for the tests.

> I noticed that FFB in RRRE does work up until I get to game settings, then, forces drops totally. I have to restart the game to get forces working back, restarting the session doesn't help.

I don't have this issue, what wheel are you using? (I'm using T300) If you think it is a proton issue, you can test with other proton versions now that the game itself seems to be fixed.

fuzunspm, 2026-01-06:
> > I tried with latest GE and experimental proton but I still need your version of proton to get RRRE working
> 
> Interesting, maybe I messed up something in a way that made it work for me. Just to be sure, are you also using RRRE v0.9.7.46?

I don't remember version number but it's always up to date on stable channel

MmAaXx500, 2026-01-14:
In order to find out that what fixed it for me, I'm trying to downgrade RRRE to previous versions, but so far all versions I tried are refusing to start because of some server error. Will continue to try more versions

MmAaXx500, 2026-02-09:
Now the fix is merged into Wine, it will be part of the 11.3 release.