Kimplul/hid-tmff2 #42: large dead zone in the center

sayapinsergej, 2022-06-22:
I got a very large dead zone in the center. I created a file /etc/udev/rules.d/99-joydev.rules. What value should be changed to --deadzone 0?

Kimplul, 2022-06-22:
`--deadzone 0` should apply to all axes, if there's still a deadzone it's likely something in game. To answer your question though, the `X` axis is the wheel.

sayapinsergej, 2022-06-24:
In the oversteer program, the steering wheel rotation is displayed correctly. I have jstest-gtk installed, and it shows a large dead zone in the center

Kimplul, 2022-06-24:
Yep, Oversteer uses the `evdev` subsystem, which is the subsystem whose deadzone is set to zero by the udev rule. `jstest-gtk` uses the older `joydev` subsystem, which is at this point obsolete and I'm not aware of any games that use it.

You can read up on the older interface and how to handle its deadzones here:
https://wiki.archlinux.org/title/Gamepad#Joystick_API_deadzones

Rudicito, 2022-08-21:
Hi,
I have the same problem, a large dead zone in the center and the correct rotation in the oversteer program, but the thing is that is happen in wine with lutris even with thrustmaster driver install and thethe dead zone set to 0 with the registry key. I can see it in the wine control panel and it's the same thing in the game. It's the same thing on steam. I don't understand how other persons can play games with a thrustmaster wheel with wine. I'm on Nobara 36. I don't think that Wine and steam using joydev subsystem. Can somebody help me. Thanks in advance.

Kimplul, 2022-08-21:
Wine uses SDL2, which uses the `evdev` subsystem, you're correct.

Could you specify which wheel you're using and in which mode it is, and which games have this dead zone? All Steam games?

Rudicito, 2022-08-21:
Hi, Thank you for responding.
I'm using the t300rs gt edition in ps3 normal mode. The only game that works for me is RaceRoom Racing Experience on steam. I managed to go into the control panel of the steam proton wine of the game and I see that the wheel rotation is correct. My question is where I have to look to find a config file or something like that, to be able to do the same in the wine-ge-custom in lutris [https://github.com/GloriousEggroll/wine-ge-custom](url). The game I mainly want to play in lutris with my wheel is Richard Burns Rally. Thanks in advice.

Kimplul, 2022-08-21:
> I managed to go into the control panel of the steam proton wine of the game and I see that the wheel rotation is correct.

Do you mean that Proton's control panel shows the wheel rotation without a deadzone only in RRRE? If that's the case, it seems like there is some configuration parameter or something messing up your game prefixes, but I don't know what that would be or why only RRRE is stepping around the issue.

> My question is where I have to look to find a config file or something like that, to be able to do the same in the wine-ge-custom in lutris

Generally speaking there are two deadzones, one provided by the subsystem, in Wine's case that's `evdev` and then there's the game's own deadzone. If the game uses some Windows internal deadzone Wine can provide it, which is where the Wine deadzone registry key comes in, but it seems to be pretty rarely needed.

Other than those things, I'm not aware of any config files for deadzones, sorry.

14c4b06b824ec593239362517f538b29, 2022-09-28:
I had the same issue with the dead center playing on Lutris. Disabling Luris Runtime solved that issue for me. This can be done for each game setting by selecting a game > Configure > System options > Enable the toggle (Disable Lutris Runtime).

sayapinsergej, 2022-11-17:
Dear Kimplus hello! How can you make your driver work not only in udev, but also with joydev? P.S. this is very necessary!

sayapinsergej, 2022-11-17:
I can't get the T300 to work properly with GIMX in any way

Kimplul, 2022-11-17:
> How can you make your driver work not only in udev, but also with joydev?

If you're asking about the deadzone, this command seems to remove the deadzone completely:

```
jscal -s 6,1,1,32767,32767,16384,16384,1,3,448,574,1394469,1394469,1,3,448,574,1394469,1394469,1,3,448,574,1394469,1394469,1,0,0,0,536870912,536870912,1,0,0,0,536870912,536870912 /dev/input/js0
```

Bit of a mouthful, but essentially `jscal` has to translate the wheel's internal value representation to something the `joydev` subsystem expects. Modifying the output of `jscal -p /dev/input/js0` gives us the command above.

Don't try to manually calibrate the wheel with `jstest-gtk`, it seems to assume the driver does this automatically and caps the wheel's rotation to only left turns. In theory I could add in a linear filter for the X-axis in the driver itself, but seeing that the `joydev` subsystem is deprecated I don't really see a need for it. The `evdev` subsystem works.

Otherwise, if you mean that the wheel doesn't work at all with `joydev`, something else is broken.

> P.S. this is very necessary!

Why, if you don't mind me asking?

> I can't get the T300 to work properly with GIMX in any way

From a quick look around the GIMX website, it looks like you would need a config file for the T300. I couldn't find one on their forums, have you tried making one yourself?

sayapinsergej, 2022-11-17:
I made the configuration file, the steering wheel is working, there is a force feedback, the problem is only in a larger dead zone, about 90 degrees

Kimplul, 2022-11-17:
I took a look around `gimx-config`, seems there is a tab `Joystick corrections` with broken-line correction parameters. This implies they're the same as what `jscal` uses, which in the command I sent earlier would be `32767,32767,16384,16384`, though I don't know if they're intended to be identical. The gimx documentation doesn't mention this tab for some reason.

sayapinsergej, 2022-11-17:
Unbelievable! It works! Is it necessary to save it somehow? After all, every time the steering wheel is connected, it can be js 0, and the next time it is js 1?

Kimplul, 2022-11-17:
Could you clarify if the `jscal` command works or the `Joystick corrections` tab in gimx?

Either way, the `jscal` command should be applied every time on reboot, yes. Essentially just set up a udev rule. Instructions here:
https://wiki.archlinux.org/title/Gamepad#Joystick_API_deadzones

As a sidenote, I'm somewhat surprised if `gimx` actually uses the `joydev` subsystem, as it seems to query `/dev/input/event` files as well. I don't understand why it would want to use the deprecated subsystem over `evdev`.

sayapinsergej, 2022-11-17:
It is your jscal team that works!

sayapinsergej, 2022-11-17:
I will try to create a rule using your link. P.S. Thank you so much! You've helped me a lot. You are very cool!

sayapinsergej, 2022-11-18:
Kimplul, need help, I can't create a udev rule
udevadm info -q property --property ID_VENDOR_ID --value /dev/input/jsX
I substitute my values
udevadm info -q property --property 044f --value /dev/input/js0
And I get this: info: unrecognized option '--property'
What am I doing wrong?

Kimplul, 2022-11-18:
> I substitute my values
> udevadm info -q property --property 044f --value /dev/input/js0

You're not supposed to put the output of the previous command there, rather use  `ID_VENDOR_ID` and `ID_MODEL_ID` as they are in the examples. Anycase, sorry, I could've posted the file content directly last time but I was in a hurry to class, here's an example that seems to work:

`/etc/udev/rules.d/85-jscal-custom-calibration.rules`
```
SUBSYSTEM=="input", ATTRS{idVendor}=="044f", ATTRS{idProduct}=="b66e", RUN+="/usr/bin/jscal -s 6,1,1,32767,32767,16384,16384,1,3,448,574,1394469,1394469,1,3,448,574,1394469,1394469,1,3,448,574,1394469,1394469,1,0,0,0,536870912,536870912,1,0,0,0,536870912,536870912 /dev/input/js%n"
```

The exact name of the file is largely irrelevant, if you already have a udev file for the wheel you can add it in there, after the evdev rule. So, for example currently I have

`/etc/udev/rules.d/99-joydev.rules`
```
SUBSYSTEM=="input", ATTRS{idVendor}=="044f", ATTRS{idProduct}=="b66e", RUN+="/usr/bin/evdev-joystick --evdev %E{DEVNAME} --deadzone 0"
SUBSYSTEM=="input", ATTRS{idVendor}=="044f", ATTRS{idProduct}=="b66e", RUN+="/usr/bin/jscal -s 6,1,1,32767,32767,16384,16384,1,3,448,574,1394469,1394469,1,3,448,574,1394469,1394469,1,3,448,574,1394469,1394469,1,0,0,0,536870912,536870912,1,0,0,0,536870912,536870912 /dev/input/js%n"
```

which removes the deadzone in both `evdev` and `joydev`. I suppose I could add this to the README.

sayapinsergej, 2022-11-18:
Everything worked out! Thank you very much! 🤝

Kimplul, 2024-11-03:
Doing some winter cleaning, closing this old issue as completed.

closed 2024-11-03 by Kimplul