Kimplul/hid-tmff2 #47: FFB not working (Fedora 36 dkms-install)

Wunderharke, 2022-09-07:
FFB is not working for me, and I am kind of lost how to solve this. Can someone please explain what possibly can go wrong? FFB isn't working for F1 2021 and Assetto Corsa Competizione.

Wunderharke, 2022-09-07:
btw firmware is on 31.

Kimplul, 2022-09-07:
Hi, could you specify if other games work or if F1 and AC are the only ones not working?

Wunderharke, 2022-09-07:
It's a fresh install and I only tries those. But I could try other games as well. What game usually works well?

Kimplul, 2022-09-07:
If no games you've tried work, it's likely the driver isn't properly loaded or you have some weird version of the wheels that doesn't play well with the driver as it is written.

For the first option, try running `dmesg -w` and plugging your wheel in. If `hid-tmff2` shows up without an error staus, it should be properly loaded. 

I mainly play DiRT rally, works on my machine ™.

Wunderharke, 2022-09-07:
This is the output of `dmesg -w`
[image]

I will try DiRT Rally tomorrow.

Kimplul, 2022-09-07:
Yep, driver is loaded and seems to be initialized correctly as far as I can tell. Curious to see if DiRT works.

Oh, and did you install the Windows driver inside the games' Wine prefixes? Apparently some games rely on metadata the driver installer provides, which could affect how the game behaves. I'm not sure about which games exactly, as this is still a relatively new discovery and I haven't looked into it too deeply but there's some great info courtesy of @raboebie in https://github.com/Kimplul/hid-tmff2/issues/46

Wunderharke, 2022-09-08:
Yes I installed the driver for both F1 and ACC, but it is still not working. It might be relevant to mention that I installed Steam as flatpak, maybe this could make problems with the driver?

detrophy, 2022-09-10:
The same issue with the wheel, but on Manjaro 21.3.7 "Ruah". The fftest tool is generating FFB.

Kimplul, 2022-09-10:
Which proton version are you using?

detrophy, 2022-09-10:
I've set the proton version to the latest available. It's the beta version. Could try to switch to the 7.X version?

Kimplul, 2022-09-10:
Seems like the issue was that I didn't test 0e47624ebbed48e922fe27170364aebf79217e7b properly. Turns out that I had a bug in how I was updating the duration of different effects, and that commit exposed it, but not so that I would've noticed.

Sorry, try out the `master` branch, F1 at least works for me. For some reason AC and ACC refuse to start, I'll try out some different proton versions.

Wunderharke, 2022-09-10:
F1 is working fine for me, ACC launches as well. AC is not installed right now unfortunately.

detrophy, 2022-09-10:
I've used the default Proton version, but tried with experimental. Also updated the version to the last one in the master branch.
FFTEST will somewhat work, but only if I set the correct device (event23, not 0).
The wheel itself is working, but no FFB.
Is there anything I could provide to you that will help pinpoint the issue?

Kimplul, 2022-09-10:
@detrophy 
> FFTEST will somewhat work, but only if I set the correct device (event23, not 0).

Could you clarify what you mean by this? Typically you would use something like
```
fftest /dev/input/by-id/usb-Thrustmaster_Thrustmaster_T300RS_Racing_wheel-event-joystick
```
> The wheel itself is working, but no FFB.

I'm guessing you mean that games accept input from the wheel, i.e. you can drive around in games but there's no FFB?

> Is there anything I could provide to you that will help pinpoint the issue?

Sure, first of all which games have you tested? Did you try rebooting after installing the newer version? Which wheel are you using, and in which mode (if applicable)?

detrophy, 2022-09-10:
> @detrophy
> 
> > FFTEST will somewhat work, but only if I set the correct device (event23, not 0).
> 
> Could you clarify what you mean by this? Typically you would use something like

There is some feedback, but not all. Oversteer can center the wheel, as an example.

> ```
> fftest /dev/input/by-id/usb-Thrustmaster_Thrustmaster_T300RS_Racing_wheel-event-joystick
> ```
fftest /dev/input/event23

Maybe it's just the default fftest goes to, but it is defaulting to event0 instead of 23

> > The wheel itself is working, but no FFB.
> 
> I'm guessing you mean that games accept input from the wheel, i.e. you can drive around in games but there's no FFB?

This is correct. Everything works fine. All input. Only FFB that is not working correctly. I also tried installing the TM drivers in the steam game prefix (/home/detrophy/.local/share/Steam/steamapps/compatdata/805550/pfx/) but this didn't change change anything.

> > Is there anything I could provide to you that will help pinpoint the issue?
> 
> Sure, first of all which games have you tested? Did you try rebooting after installing the newer version? Which wheel are you using, and in which mode (if applicable)?

Currently only ACC, as this is the only game I game with the wheel.
Yes, I did a reboot, multiple times between steps when necessary.

I'm using the T300RS GT in the PS3 mode, not inverted.

Kimplul, 2022-09-11:
Pretty weird, ACC works for me with Proton 7.0-4, and if `fftest` works then the driver seems to be loaded. I also upgraded my wheel's firmware to 34 but works just fine. Do you have any other games to try the wheel with? Also, I'm assuming you have FFB turned on in ACC? There should be a small FFB gauge in the bottom right of the screen, does it show anything while you're driving?

detrophy, 2022-09-11:
To check if there is no mistake from my side or by a misconfiguration, I've uninstalled steam, oversteer and tmff2 and removed all modules and the dkms installation. I will try to reinstall tmff2 later on without dkms installation.

Wunderharke, 2022-09-14:
Are you both talking about Assetto Corsa Competizione? I have the feeling you are talking past each other. @Kimplul should Assetto Corsa 1 be working again by now?

Wunderharke, 2022-09-14:
ACC, Steam Flatpak, Proton Experimental, Firmware 31, tmff2 is working e2dd7bdc3a22ac8b8551db4154fa36fcc631b402 on my machine (fedora 36).

Kimplul, 2022-09-14:
> Are you both talking about Assetto Corsa Competizione?

By ACC I'm referring to Assetto Corsa Competizione, yes.

> should Assetto Corsa 1 be working again by now?

Don't know, can't get it to launch. The guides on ProtonDB haven't worked for me, so I can't test the game. In theory it should, though.

detrophy, 2022-09-21:
So, I've done some more testing on my side. Checking double I'm not the one with a defect configuration here.

I've found out that the game is using the wheel, even when the steam controller support is set to off and setting it to "default" will then enable FFB.

Can someone confirm this is the same with other systems and I'm not running into a bug?

[image]

Wunderharke, 2022-09-25:
I am unfortunately quite busy at the moment. I will try to confirm this on thursday.

Kimplul, 2022-09-25:
Can't confirm, FFB works for me with both default and disabled per-game controller settings. I also tried turning on generic controller configuration but FFB still works. Strange.

peter-vasut, 2023-01-15:
@Wunderharke I see you've installed the Windows driver. When I did this (the installation failed, on windows when you click trough the setup, it closes and the installer with progressbar gets open, this didn't happen on Linux for me). After this the wheel wasn't even recognized by Dirt Rally 2 or Steam. Removing correct folder from `~/.steam/steam/steamapps/compatdata` fixed this for me. Maybe the windows driver creates some random issues for you too, did you try to run things without it?

Wunderharke, 2023-01-16:
After reinstalling the games they worked with and without installing the driver somehow. 

I think this issue is solved. I am available for some testing if issues return.

closed 2023-01-16 by Wunderharke