Kimplul/hid-tmff2 #48: Thrustmaster TX Support

Austin-2, 2022-10-20:
Hello,

I was looking for support for a Thrustmaster TX wheel. Mine is the Ferrari 458 wheel, but it is technically a Thrustmaster TX.

https://support.thrustmaster.com/en/product/txracingwheel-en/

From my understanding, this wheel is the same as the T300RS, but for Xbox/PC instead of PS4/PC. The drivers on Windows use the same files.

I was able to get the wheel working with tmdriv, but it did not have FFB.

I would love if there would be support for my wheel on Linux!

thanks!

Kimplul, 2022-10-20:
Hi, I could try adding the wheel ID to my driver, if we're lucky it might work without much tweaking. Although I wouldn't say it's bound to work, as the T500 also uses the same driver installer files but doesn't work the same way as a T300. Not sure about the base, looks very similar but I guess we'll see.

I see you've found the `hid-tminit` repo. Apparently the TX requires a two-step boot process, at least based on https://gitlab.com/her0/tmdrv/-/blob/master/tmdrv_devices/thrustmaster_tx.py. Shouldn't be an issue, but probably requires modifying the driver a bit more than just adding a wheel value as all other wheels are single stage.

Okay, let's try to get the PC mode working. I added the TX to the `hid-tminit` module accept list according to what I found in [tmdrv](https://gitlab.com/her0/tmdrv/-/blob/master/tmdrv_devices/thrustmaster_tx.py). Please install my fork: https://github.com/Kimplul/hid-tminit/tree/master

With the fork installed, please show the `dmesg` output when plugging in the wheel. There should be an error about an unknown ID, which should be one of the values we're missing. Once we have this, we should be able to move on to tackle the second boot stage.

Austin-2, 2022-10-21:
> Okay, let's try to get the PC mode working. I added the TX to the hid-tminit module accept list according to what I found in [tmdrv](https://gitlab.com/her0/tmdrv/-/blob/master/tmdrv_devices/thrustmaster_tx.py). Please install my fork: https://github.com/Kimplul/hid-tminit/tree/master

First off, thanks for looking into this for me! I am a bit of a linux noob (trying to set this up on my steamdeck).

I cloned your fork into a folder, went into that folder and ran `make` and it returned this error:

> error: implicit declaration of function ‘hid_is_usb’ [-Werror=implicit-function-declaration]
  300 |         if (!hid_is_usb(hdev))
      |              ^~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:272: /home/deck/Downloads/hid-tminit/hid-tminit.o] Error 1
make[1]: *** [Makefile:1847: /home/deck/Downloads/hid-tminit] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.13.0-valve21.3-1-neptune/build'
make: *** [Makefile:5: all] Error 2

Not sure if it is a configuration issue on my end, or something with the file. I am getting the sense that the veriable needs to be declared before it is used, but I am unsure what to declare it as.

Any help is appreciated.

Kimplul, 2022-10-21:
> Not sure if it is a configuration issue on my end, or something with the file. I am getting the sense that the veriable needs to be declared before it is used, but I am unsure what to declare it as.

Should be provided for you in `linux/hid.h`: https://elixir.bootlin.com/linux/v6.1-rc1/source/include/linux/hid.h#L867
Apparently the steam deck headers are slightly different, or there's some config issues. If that is the only error, you could try removing the if statement, it's mostly for paranoia, and we can try to figure out the root cause later.

Btw, I'm not familiar with the steam deck, how are you compiling things? Is there a kernel header package and have you installed it?

berarma, 2022-10-21:
I think this inline wasn't introduced until some later 5.x version. Probably Steam Deck uses a kernel that doesn't have this inline yet.

Kimplul, 2022-10-21:
Thanks for the tip, 5.15 apparently. Solves that mystery.

Austin-2, 2022-10-21:
@Kimplul 

It was able to make and install after removing the if. Wheel is showing in dmesg but not recognized in games just yet.

dmesg didn't work at first so I ran what I found here: https://unix.stackexchange.com/questions/390184/dmesg-read-kernel-buffer-failed-permission-denied
```

% sudo sysctl kernel.dmesg_restrict=0

kernel.dmesg_restrict = 0

```

My `dmesg` is as follows:

> [  256.681760] usb 1-1.3: new full-speed USB device number 4 using xhci_hcd
[  256.810799] usb 1-1.3: New USB device found, idVendor=044f, idProduct=b664, bcdDevice= 1.01
[  256.810812] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  256.810816] usb 1-1.3: Product: Thrustmaster TX GIP Racing Wheel
[  256.810819] usb 1-1.3: Manufacturer: Thrustmaster
[  256.810822] usb 1-1.3: SerialNumber: 0000000000000000
(deck@steamdeck hid-tminit)$

Kimplul, 2022-10-21:
I suspect `sudo dmesg` would also work but in any case, the `hid-tminit` driver isn't picking up the wheel for some reason. Could you try `sudo modprobe hid-tminit` and replugging the wheel?

Austin-2, 2022-10-21:
> [  852.382679] cs35l41 spi-VLV1776:00: DSP1: Firmware version: 3
> [  852.382688] cs35l41 spi-VLV1776:00: DSP1: cs35l41-dsp1-spk-prot.wmfw: Fri 02 Apr 2021 21:03:50 W. Europe Daylight Time
> [  852.446128] cs35l41 spi-VLV1776:00: DSP1: Firmware: 400a4 vendor: 0x2 v0.33.0, 2 algorithms
> [  852.446325] cs35l41 spi-VLV1776:00: DSP1: 0: ID cd v29.53.0 XM@94 YM@e
> [  852.446345] cs35l41 spi-VLV1776:00: DSP1: 1: ID f20b v0.0.1 XM@170 YM@0
> [  852.458279] cs35l41 spi-VLV1776:01: DSP1: Firmware version: 3
> [  852.458288] cs35l41 spi-VLV1776:01: DSP1: cs35l41-dsp1-spk-prot.wmfw: Fri 02 Apr 2021 21:03:50 W. Europe Daylight Time
> [  852.521241] cs35l41 spi-VLV1776:01: DSP1: Firmware: 400a4 vendor: 0x2 v0.33.0, 2 algorithms
> [  852.521412] cs35l41 spi-VLV1776:01: DSP1: 0: ID cd v29.53.0 XM@94 YM@e
> [  852.521420] cs35l41 spi-VLV1776:01: DSP1: 1: ID f20b v0.0.1 XM@170 YM@0
> [  853.105809] cs35l41 spi-VLV1776:00: DSP1: Firmware version: 3
> [  853.105820] cs35l41 spi-VLV1776:00: DSP1: cs35l41-dsp1-spk-prot.wmfw: Fri 02 Apr 2021 21:03:50 W. Europe Daylight Time
> [  853.170199] cs35l41 spi-VLV1776:00: DSP1: Firmware: 400a4 vendor: 0x2 v0.33.0, 2 algorithms
> [  853.170345] cs35l41 spi-VLV1776:00: DSP1: 0: ID cd v29.53.0 XM@94 YM@e
> [  853.170352] cs35l41 spi-VLV1776:00: DSP1: 1: ID f20b v0.0.1 XM@170 YM@0
> [  853.183271] cs35l41 spi-VLV1776:01: DSP1: Firmware version: 3
> [  853.183280] cs35l41 spi-VLV1776:01: DSP1: cs35l41-dsp1-spk-prot.wmfw: Fri 02 Apr 2021 21:03:50 W. Europe Daylight Time
> [  853.246664] cs35l41 spi-VLV1776:01: DSP1: Firmware: 400a4 vendor: 0x2 v0.33.0, 2 algorithms
> [  853.246825] cs35l41 spi-VLV1776:01: DSP1: 0: ID cd v29.53.0 XM@94 YM@e
> [  853.246832] cs35l41 spi-VLV1776:01: DSP1: 1: ID f20b v0.0.1 XM@170 YM@0
> [  858.539212] input: Microsoft X-Box 360 pad 0 as /devices/virtual/input/input33
> [  880.266571] hid_tminit: loading out-of-tree module taints kernel.
> [  880.266654] hid_tminit: module verification failed: signature and/or required key missing - tainting kernel
> [  888.822475] usb 1-1.3: new full-speed USB device number 6 using xhci_hcd
> [  888.950433] usb 1-1.3: New USB device found, idVendor=044f, idProduct=b664, bcdDevice= 1.01
> [  888.950442] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [  888.950444] usb 1-1.3: Product: Thrustmaster TX GIP Racing Wheel
> [  888.950446] usb 1-1.3: Manufacturer: Thrustmaster
> [  888.950448] usb 1-1.3: SerialNumber: 0000000000000000
>

Added everything that seemed kinda relevant.

Kimplul, 2022-10-21:
Just checking: did you also remove `return -EINVAL;` after the if statement?

Austin-2, 2022-10-21:
> Just checking: did you also remove `return -EINVAL;` after the if statement?

Yep, commented out the statement like this:

[image]

Kimplul, 2022-10-21:
Generally speaking the SSL error should be treated more as a warning, and I can't see anything in the pasted output about the module being rejected, so I'm not entirely sure what's going on. Signing yourself is still probably a good idea.

I did notice that the `dmesg` output reads `hid-tminit`, whereas the current driver should show up as `hid-thrustmaster`, Wonder if that's still the old version?

Could you paste the output of
```
sudo rmmod hid-tminit
sudo modprobe hid-thrustmaster
sude dmesg
```

Preferably unplugging the wheel before running the commands?

Austin-2, 2022-10-21:
> sudo rmmod hid-tminit
> sudo modprobe hid-thrustmaster
> sude dmesg

[dmesg.txt](https://github.com/Kimplul/hid-tmff2/files/9841245/dmesg.txt)

the output of this was insanely large. Saved to a file, hope its not a pain to look through :smile:

Kimplul, 2022-10-21:
Sorry, should've been more clear, didn't need the whole output, just the bit that's generated when the wheel is plugged in.

Austin-2, 2022-10-21:
> Sorry, should've been more clear, didn't need the whole output, just the bit that's generated when the wheel is plugged in.

Do you need me to generate a new message? I made that dmesg when my wheel was unplugged. I can another one with the wheel plugged in or out if you need.

Kimplul, 2022-10-21:
Yeah, what I meant was that the two first commands would hopefully load the newer module and the `dmesg` to be like https://github.com/Kimplul/hid-tmff2/issues/48#issuecomment-1287086087 but hopefully with different output.

Austin-2, 2022-10-21:
> Yeah, what I meant was that the two first commands would hopefully load the newer module and the `dmesg` to be like [#48 (comment)](https://github.com/Kimplul/hid-tmff2/issues/48#issuecomment-1287086087) but hopefully with different output.

Okay, so just to confirm, I am a bit confused on what would help you more:

- Remove existing tminit
  - `sudo rmmod hid-tminit`
  - `sudo modprobe hid-thrustmaster`
  - plug in wheel
  - `sudo dmesg`
- Remove existing tminit and redo the install.
  -   `sudo rmmod hid-tminit`
  - `sudo modprobe hid-thrustmaster`
  - make
  - sudo make install
  - plug in wheel
  - `sudo dmesg`

What would you advise?

Kimplul, 2022-10-21:
> So we are looking for
> 
> [ 880.266571] hid_tminit: loading out-of-tree module > taints kernel.
> [ 880.266654] hid_tminit: module verification failed: signature and/or required key missing - tainting kernel
> 
> to instead say something about hid-thrustmaster?

Yep, pretty much. The confusing bit here is that the module's filename is `hid-tminit`, but the `.name` property in the driver is `hid-thrustmaster`. This is apparently to keep up to the in-kernel counterpart.

> Okay, so just to confirm, I am a bit confused on what would help you more:
>
>     Remove existing tminit
>        sudo rmmod hid-tminit
>        sudo modprobe hid-thrustmaster
>        plug in wheel [...]

The second one. Possibly also run `sudo depmod -A` before plugging in the wheel.

By the by, you can also run `sudo dmesg -w` first and then plug in the wheel, makes it a bit easier to see what exactly is new and what isn't. Same goes for `rmmod` and `modprobe`, so in one terminal `sudo modprobe -w` and in another `sudo modprobe ...` but do whatever is easier for you.

Sorry again for the confusing requests.

Austin-2, 2022-10-21:
> > So we are looking for
> > [ 880.266571] hid_tminit: loading out-of-tree module > taints kernel.
> > [ 880.266654] hid_tminit: module verification failed: signature and/or required key missing - tainting kernel
> > to instead say something about hid-thrustmaster?
> 
> Yep, pretty much. The confusing bit here is that the module's filename is `hid-tminit`, but the `.name` property in the driver is `hid-thrustmaster`. This is apparently to keep up to the in-kernel counterpart.
> 
> > Okay, so just to confirm, I am a bit confused on what would help you more:
> > ```
> > Remove existing tminit
> >    sudo rmmod hid-tminit
> >    sudo modprobe hid-thrustmaster
> >    plug in wheel
> >    sudo dmesg
> > Remove existing tminit and redo the install.
> >     sudo rmmod hid-tminit
> >     sudo modprobe hid-thrustmaster
> >     make
> >     sudo make install
> >     plug in wheel
> >     sudo dmesg
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > What would you advise?
> 
> The second one. Possibly also run `sudo depmod -A` before plugging in the wheel.
> 
> By the by, you can also run `sudo dmesg -w` first and then plug in the wheel, makes it a bit easier to see what exactly is new and what isn't. Same goes for `rmmod` and `modprobe`, so in one terminal `sudo dmesg -w` and in another `sudo modprobe ...` but do whatever is easier for you.
> 
> Sorry again for the confusing requests.

Lol do not be sorry, you are helping out here which I really appreciate :smile: But yeah, i was a bit confused as to why the driver would be hid-thrustmaster when the file is tminit, but i understand now.

I ran the second set of commands. I edited my last reply a ton to update info but I did run the second set of commands again, and hid-tminit is no longer showing up.

> [ 1276.560471] usb 1-1.2: new full-speed USB device number 9 using xhci_hcd
> [ 1276.690403] usb 1-1.2: New USB device found, idVendor=044f, idProduct=b664, bcdDevice= 1.01
> [ 1276.690415] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [ 1276.690419] usb 1-1.2: Product: Thrustmaster TX GIP Racing Wheel
> [ 1276.690450] usb 1-1.2: Manufacturer: Thrustmaster
> [ 1276.690453] usb 1-1.2: SerialNumber: 0000000000000000
>

I confirmed that hid-thrustmaster exists as a module and it is loaded via `modprobe` but still not showing up when I plug in the wheel. It is loaded, as I can see it when i run `lsmod`, but it says 0 devices are using it.

isopix, 2022-10-21:
I love Plasma too.

But pasting screenshots instead of code diffs ('git diff $file.c' does
wonders) is ridiculous at best.

Kimplul, 2022-10-21:
> EDIT: On a side note, I have tried rebooting after all this, and I don't see hid-thrustmaster loaded anymore, until I manually load it with modprobe hid-thrustmaster.

Yeah, typically the kernel only loads the drivers when it needs to, and in this case the kernel for some reason doesn't associate the driver with the wheel, though I don't really understand why. The `thrustmaster_devices` array contains the USB vendor and product IDs that the driver would like to be probed for, after which `thrustmaster_probe()` should be called, but I don't see it happening in your `dmesg` output for some reason. Manually loading the driver should really never be necessary, but it can be useful for forcing it out or in to the kernel when testing out changes.

> Also, I ran sudo depmod -A and nothing happened. It looks like it generates a make file? do I need to run that make file?

I believe it updates the kernel's internal cache of available modules, it doesn't generate any files.

@isopix 
> But pasting screenshots instead of code diffs ('git diff $file.c' does
wonders) is ridiculous at best.

Bit rude, though I agree with the sentiment.

Austin-2, 2022-10-21:
> Yeah, typically the kernel only loads the drivers when it needs to, and in this case the kernel for some reason doesn't associate the driver with the wheel, though I don't really understand why. The [...]

Any ideas on where to go from here? Would the answer lie in updating hid-tminit to accept my device idea be the key here? Or really any other reasonable next steps

Kimplul, 2022-10-21:
I suspect the `tminit/thrustmaster` thing might be blocking us, I went ahead and created a `dev` branch and renamed everything to `hid-tminit-dev`:
https://github.com/Kimplul/hid-tminit/tree/dev

Please try it out, so

+ `make`
+ `sudo make install`
+ `sudo depmod -A`
+ `sudo modprobe hid-tminit-dev`
+ plug in wheel
+ `dmesg`

We should see `hid-tminit-dev` and at least know the newest code is being loaded.

Austin-2, 2022-10-21:
No luck with that one.

> [13302.158376] usb 1-1.2: new full-speed USB device number 10 using xhci_hcd
> [13302.286933] usb 1-1.2: New USB device found, idVendor=044f, idProduct=b664, bcdDevice= 1.01
> [13302.286946] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [13302.286950] usb 1-1.2: Product: Thrustmaster TX GIP Racing Wheel
> [13302.286953] usb 1-1.2: Manufacturer: Thrustmaster
> [13302.286956] usb 1-1.2: SerialNumber: 0000000000000000
> [13387.200410] usb 1-1.2: USB disconnect, device number 10
> [13395.084779] usb 1-1.2: new full-speed USB device number 11 using xhci_hcd
> [13395.213947] usb 1-1.2: New USB device found, idVendor=044f, idProduct=b664, bcdDevice= 1.01
> [13395.213958] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [13395.213962] usb 1-1.2: Product: Thrustmaster TX GIP Racing Wheel
> [13395.213965] usb 1-1.2: Manufacturer: Thrustmaster
> [13395.213967] usb 1-1.2: SerialNumber: 0000000000000000
>

No mention of hid-tminit-dev in that. Something odd I noticed, not sure if it will lead us anywhere, but `listing the modules returns

> hid_tminit_dev         16384  0

Should it be `hid_tminit_dev` or `hid-tminit-dev`?

Kimplul, 2022-10-22:
> Should it be hid_tminit_dev or hid-tminit-dev?

The underscores are expected, they're just for show I suppose. The commands work with dashes.

> Also, part of me is wondering if !hid_is_usb was necessary, is there any way I can declare that not inline?

If you want to, you could just copy the definition from here:
https://elixir.bootlin.com/linux/v6.1-rc1/source/include/linux/hid.h#L867

Though I doubt it'll do much, as I'm guessing you're connecting the wheel via USB and the line exists to fail if the device is not a USB device.

Did you get the module signing to work? My understanding is that unsigned modules should be prevented from loading, but I guess it could be possible that they're prevented from being probed as well.

Also, I wonder if the Steam Deck is making things difficult. Do you have some other Linux machine or virtual machine you could test?

Austin-2, 2022-10-22:
> The underscores are expected, they're just for show I suppose. The commands work with dashes.
>
>
> If you want to, you could just copy the definition from here: [...]

Yes it is a USB device. I can try this out. I assume I just copy over that definition somewhere before of that if statement and remove the exclamation from the original script.

> Did you get the module signing to work? My understanding is that unsigned modules should be prevented from loading, but I guess it could be possible that they're prevented from being probed as well.

I didn’t try to sign the modules at all. Your understanding is what I have as well. The modules did show as loaded, and tminit did show the message in `dmesg` and I believe it at least showing is just a warning that it is loaded, but not signed.

> Also, I wonder if the Steam Deck is making things difficult. Do you have some other Linux machine or virtual machine you could test?

If the hid_is_usb doesn’t work, i can try this hopefully today or more likely tomorrow. Might just go with Ubuntu as that is what I am familiar with. Just gotta grab a usb drive big enough to run it off of.

Austin-2, 2022-10-23:
> If you want to, you could just copy the definition from here: https://elixir.bootlin.com/linux/v6.1-rc1/source/include/linux/hid.h#L867
>
> Though I doubt it'll do much, as I'm guessing you're [...]

Got the module to make and install, but had to manually load by inserting this definition. Since it has been a couple of days I forgot if this is new, but I am seeing `hid-tminit-dev` in `dmesg`

> [   63.779627] hid_tminit_dev: loading out-of-tree module taints kernel.
> [   63.779721] hid_tminit_dev: module verification failed: signature and/or required key missing - tainting kernel
> [   78.664771] usb 1-1.2: new full-speed USB device number 4 using xhci_hcd
> [   78.793641] usb 1-1.2: New USB device found, idVendor=044f, idProduct=b664, bcdDevice= 1.01
> [   78.793653] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [   78.793656] usb 1-1.2: Product: Thrustmaster TX GIP Racing Wheel
> [   78.793659] usb 1-1.2: Manufacturer: Thrustmaster
> [   78.793661] usb 1-1.2: SerialNumber: 0000000000000000

 
> Did you get the module signing to work? My understanding is that unsigned modules should be prevented from loading, but I guess it could be possible that they're prevented from being probed as well.

Haven't figured out how to do this quite yet.

 
> Also, I wonder if the Steam Deck is making things difficult. Do you have some other Linux machine or virtual machine you could test?

Just tried on Ubuntu, and no differences vs. what I am seeing on the Steamdeck in regards to the `dmesg`.

Kimplul, 2022-10-23:
> Just tried on Ubuntu, and no differences vs. what I am seeing on the Steamdeck in regards to the dmesg.

This is arguably the most telling. Based on the `dmesg` output, I'm starting to suspect that the wheel (or at least the initial stage) isn't reporting itself as a HID device, rather a raw USB device. `hid-tminit` is a HID driver, and that might explain why it doesn't show up.

I don't know if it's possible to make the driver accept raw USB IDs or if we'd need a separate driver for it. I'll do some research and report back.

Austin-2, 2022-10-23:
> This is arguably the most telling. Based on the `dmesg` output, I'm starting to suspect that the wheel (or at least the initial stage) isn't reporting itself as a HID device, rather a raw USB device. `hid-tminit` is a HID driver, and that might explain why it doesn't show up.

Do you know if there is anyway I can confirm this for you? Either on windows or Linux?

Kimplul, 2022-10-23:
Not really, no. My plan is to maybe write a couple skeleton drivers to check if this theory holds water. Typically a HID device would get picked up by `hid-generic`, which doesn't seem to be happening in this case, but I don't know if there are corner cases where this doesn't apply.

In other words, uncharted territory for me.

berarma, 2022-10-23:
The Logitech G920 for XBOX/PC needs a `usb_modeswitch` command to put it in PC mode. The command is currently included in the default UDEV rules. Maybe this wheel is something like this?

Kimplul, 2022-10-23:
Good point. `tmdrv` could be used as a stop-gap to kick the wheel into its final form, and then from there see if the T300 driver works for the TX.

Kimplul, 2022-10-31:
Hello again, sorry for the wait. Assuming my theory about `hid` vs `usb` is correct, there doesn't seem to be a (non-hacky, at least) way to add raw USB devices to a hid driver. This means that `hid-tminit` would need to be rewritten for the TX to be initialized the typical way. As such, I think @berarma's idea about `usb_modeswitch` might be the better option for now, if we get it working. Something like
```
usb_modeswitch -v 044f -p b664 -M 418300010000
```

might work for the first init stage, based on [this](https://gitlab.com/her0/tmdrv/-/blob/master/tmdrv_devices/thrustmaster_tx.py).

Short-term, I'd like to see if the TX actually is just a T300 in disguise. I suggest @Austin-2 you try initializing the wheel with `tmdrv`. I added the TX's "final" PC mode ID to my driver over in the `tx` branch: https://github.com/Kimplul/hid-tmff2/tree/tx

If `tmdrv` initializes the wheel like I expect it to, the driver should be able to pick it up fully initialized and treat it as if it was a T300, with hopefully (at least partially) working FFB.

Austin-2, 2022-11-01:
> Hello again, sorry for the wait. Assuming my theory about `hid` vs `usb` is correct, there doesn't seem to be a (non-hacky, at least) way to add raw USB devices to a hid driver. This means that [...]

Hey,

Thanks for looking into this, but I am a little confused.

Do you want me to do anything with 

`usb_modeswitch -v 044f -p b664 -M 418300010000`

Or just make a driver with your branch at https://github.com/Kimplul/hid-tmff2/tree/tx?

Austin-2, 2022-11-01:
I installed the driver from that new branch I was getting errors of no makefile, but I was able to install via dkms. After installing I had to run `sudo modprobe hid-tmff-new` and got this in `dmesg`

```
[  623.899112] usb 1-1.3: new full-speed USB device number 5 using xhci_hcd
[  624.028512] usb 1-1.3: New USB device found, idVendor=044f, idProduct=b664, b                                          cdDevice= 1.01
[  624.028524] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber                                          =3
[  624.028527] usb 1-1.3: Product: Thrustmaster TX GIP Racing Wheel
[  624.028530] usb 1-1.3: Manufacturer: Thrustmaster
[  624.028533] usb 1-1.3: SerialNumber: 0000000000000000
[  773.643807] hid_tmff_new: loading out-of-tree module taints kernel.
[  773.643905] hid_tmff_new: module verification failed: signature and/or requir                                          ed key missing - tainting kernel
[  778.465781] [drm] Failed to add display topology, DTM TA is not initialized.
[  779.597287] usb 1-1.3: USB disconnect, device number 5
[  784.155904] usb 1-1.3: new full-speed USB device number 6 using xhci_hcd
[  784.285767] usb 1-1.3: New USB device found, idVendor=044f, idProduct=b664, b                                          cdDevice= 1.01
[  784.285779] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber                                          =3
[  784.285782] usb 1-1.3: Product: Thrustmaster TX GIP Racing Wheel
[  784.285785] usb 1-1.3: Manufacturer: Thrustmaster
[  784.285788] usb 1-1.3: SerialNumber: 0000000000000000

```

Kimplul, 2022-11-01:
> I installed the driver from that new branch I was getting errors of no makefile,

I did find a typo which causes the compilation to fail, sorry about that, but what do you mean no makefile? `dkms` still uses the makefile, so I don't really think the driver you're loading is the one with the TX stuff. Please try compiling manually and paste any errors into your comment.

> After installing tmdrv, this is my dmesg [...]

Looks about right, I can see `hid-generic` so the wheel should be able to be picked up once the stuff mentioned earlier is sorted out.

> I am unable to run usb_modeswitch on my Steamdeck. If this is the right direction, I can try the following on my Ubuntu install.

What exactly do you mean? Is it not installed by default, or did you get an error message? Is it not available through `pacman`?

Kimplul, 2022-11-02:
>  Not sure how to compile manually, unless you mean make instead of using dkms. When doing that, I got this error:
>
>     make -C hid-tminit KDIR="/lib/modules/5.13.0-valve21.3-1-neptune/build"
>     make[1]: Entering directory '/home/deck/Downloads/hid-tmff2/hid-tminit'
>     make[1]: *** No targets specified and no makefile found. Stop.
>     make[1]: Leaving directory '/home/deck/Downloads/hid-tmff2/hid-tminit'
>     make: *** [Makefile:16: hid-tminit] Error 2

Oh, right, you're missing the git submodule. You just need to run
```
git submodule init
git submodule update
```

and then the compilation should succeed. Usually the submodule is added automatically when `--recurse-submodules` is added to `git clone`, guess you just missed it. After that, run `make` and `sudo make install`.

> I didn't know I was able to do that. I was able to install it is sudo pacman -S usb_modeswitch.

Okay, good to know.

> Just for clarification, what is the next plan we are attempting? I am unsure how the usb_modeswitch falls into the equation.
>
>    1. Install the new tmff2 driver
>    2. Initiate the wheel with tmdrv
>    3. Use usb_modeswitch
>    4. ???

Right now, 1. Mostly because I want to see if the wheel actually is just a T300 in disguise. If it is, then great, we can do some bughinting and see if there are any obvious differences. If it doesn't, trying to figure out how to boot a non-working device seems like wasted effort and I'd prefer to get the wheel working first, then streamlining the init process.

You already showed that 2 works more or less how I expect it to, but this step is temporary and eventually it'd be nice to have the wheel automatically initialized without having to install third-party tools besides this and the init driver (which are bundled together in this repo, the errors you reported was for the `hid-tminit` driver missing).

The TX init sequence differs from the other wheels that I'm aware of in that it doesn't show up as a HID device, which would either require rewriting `hid-tminit` to a USB driver instead of a HID driver. `usb_modeswitch` would've maybe been a preferable option to this, at least it'd require less effort and it was preinstalled on my machine at least. However, if it's not as standard as I thought it was, that might throw a wrench in the cogs. Not sure whether rewriting `hid-tminit` or just adding a note in the README about requiring *some* third party tools is better.

Austin-2, 2022-11-02:
I was able to make the driver. I wasn't sure what this driver name was, but I did a `modprobe` on the following: `hid-thrustmaster`, `hid-tmff-new`, `hid-tminit`. Let me know if I should do anything else, the `dmesg` has no new things to my eye.

> [  592.662030] hid_tminit: loading out-of-tree module taints kernel.
> [  592.662120] hid_tminit: module verification failed: signature and/or required key missing - tainting kernel
> [  863.657458] usb 1-1.2: new full-speed USB device number 4 using xhci_hcd
> [  863.780307] usb 1-1.2: New USB device found, idVendor=044f, idProduct=b664, bcdDevice= 1.01
> [  863.780318] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [  863.780321] usb 1-1.2: Product: Thrustmaster TX GIP Racing Wheel
> [  863.780324] usb 1-1.2: Manufacturer: Thrustmaster
> [  863.780327] usb 1-1.2: SerialNumber: 0000000000000000
> [  864.720984] usb 1-1.2: USB disconnect, device number 4
> 
> 
>

Kimplul, 2022-11-02:
You need to initialize the wheel first with `tmdrv`, only after that can the driver pick the wheel up. `hid-tmff-new` is the name of the FFB driver, by the way.

Austin-2, 2022-11-02:
Installed and modprobed `hid-tmff-new`

> [ 1193.332362] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-1/1             -1.2/1-1.2:1.0/0003:044F:B65D.0009/input/input33
> [ 1193.332771] hid-generic 0003:044F:B65D.0009: input,hidraw6: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB W             heel] on usb-0000:04:00.3-1.2/input0
> [ 1193.679393] usb 1-1.2: USB disconnect, device number 5
> [ 1194.140127] usb 1-1.2: new full-speed USB device number 6 using xhci_hcd
> [ 1194.266041] usb 1-1.2: New USB device found, idVendor=044f, idProduct=b669, bcdDevice= 1.00
> [ 1194.266053] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [ 1194.266056] usb 1-1.2: Product: Thrustmaster TX Racing Wheel
> [ 1194.266059] usb 1-1.2: Manufacturer: Thrustmaster
> [ 1194.356408] input: Thrustmaster Thrustmaster TX Racing Wheel as /devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1             /1-1/1-1.2/1-1.2:1.0/0003:044F:B669.000A/input/input34
> [ 1194.356703] hid-generic 0003:044F:B669.000A: input,hidraw6: USB HID v1.11 Joystick [Thrustmaster Thrustmaster TX R             acing Wheel] on usb-0000:04:00.3-1.2/input0
> [ 1269.455940] usb 1-1.2: USB disconnect, device number 6
>

Kimplul, 2022-11-04:
Not sure if this is related:
https://unix.stackexchange.com/questions/12005/how-to-use-linux-kernel-driver-bind-unbind-interface-for-usb-hid-devices

I thought `hid-generic` would automatically look for more specific drivers and that the issue described above wasn't a thing anymore, but in this case it seems like it might still be. I'm reasonably certain the driver has the correct `VendorID`/`ProductID` values, so I don't really know why the driver isn't being probed.

Other things that came to mind:

+ Did you run `sudo depmod -A` after `sudo make install`?
+ Did you try the driver on sometihing other than the SteamDeck?

penzoiders, 2022-11-15:
Hi, was looking around for Thrustmaster TX Linux support and ended up here.
I'm using Chimera OS (Arch based) and I have a TX which is not recognized, I'm no Linux newbie but never went into coding drivers. If there's anything I can help with I'll be happy to contribute.

Can you tell me what it's needed to test/gather data for the current research stage?

also: will it be useful if I can share an ssh access to a machine to test the hardware on (create a temp ssh keypair and send the public here, I'll provide a public endpoint to ssh into it)

Kimplul, 2022-11-15:
Hi @penzoiders , the main issue right now seems to be that the driver for some reason either isn't being probed or rejects the wheel when presented with it. I'm unsure why this is, as the wheel's vendor / product ids should be correct in the driver and I can't see any attempts by this driver to pick up the wheel in the provided `dmesg` logs.

I would appreciate it if you checked if the `tx` branch shows the same behavior for you as it does for @Austin-2, at least it would confirm that the behavior is something I've messed up rather than just the Steam Deck being weird or something. The steps to take are outlined higher up in the thread: Install this driver, initialize the wheel with `tmdrv`, see what happens.

penzoiders, 2022-11-17:
OK, I'll try on my Manjaro desktop and give you feedback.
I see that there're AUR packages for your repo too.. but I guess I have to take the `tx`branch manually isn't it?

Kimplul, 2022-11-17:
Yep, the package uses the main branch.

bytes-ls, 2022-12-07:
Using TX branch, at first dmesg told me to update the wheels firmware, which made sense because I had never updated it so I went and did that using windows.

After doing that, when I run tmdrv, the lights on the wheel start blinking, then they go out, dmesg says the thrustmaster is in bootloader mode, and the wheel becomes unusable until I do a rescue-mode reflash on windows again.

`[ 1809.860914] usb 1-9: new full-speed USB device number 17 using xhci_hcd
[ 1810.002210] usb 1-9: New USB device found, idVendor=044f, idProduct=b669, bcdDevice= 1.00
[ 1810.002212] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1810.002213] usb 1-9: Product: Thrustmaster TX Racing Wheel
[ 1810.002214] usb 1-9: Manufacturer: Thrustmaster
[ 1810.004111] input: Thrustmaster Thrustmaster TX Racing Wheel as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/0003:044F:B669.0007/input/input29
[ 1810.004419] hid-generic 0003:044F:B669.0007: input,hidraw5: USB HID v1.11 Joystick [Thrustmaster Thrustmaster TX Racing Wheel] on usb-0000:00:14.0-9/input0
[ 1810.061188] input: Thrustmaster Thrustmaster TX Racing Wheel as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/0003:044F:B669.0007/input/input30
[ 1810.061510] tmff2 0003:044F:B669.0007: input,hidraw5: USB HID v1.11 Joystick [Thrustmaster Thrustmaster TX Racing Wheel] on usb-0000:00:14.0-9/input0
[ 1810.061823] tmff2 0003:044F:B669.0007: force feedback for T300RS
[ 1810.071468] usb 1-9: USB disconnect, device number 17
[ 1811.160963] usb 1-9: new full-speed USB device number 18 using xhci_hcd
[ 1811.301857] usb 1-9: New USB device found, idVendor=044f, idProduct=b663, bcdDevice= 0.01
[ 1811.301870] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1811.301876] usb 1-9: Product: Thrustmaster TX Bootloader
[ 1811.301880] usb 1-9: Manufacturer: Thrustmaster`

Kimplul, 2022-12-07:
Thanks for the info @bytes-ls, from the `dmesg` output I can see that the `tmff2` driver is being loaded as I would expect, so maybe the Steam deck kernel has some extra security enabled for module loading? Not sure.

Anycase, more pressing is that the driver apparently causes the wheel to brick itself, and as such the wheel unfortunately isn't a drop-in replacement for the T300 as far as this driver is concerned. They may still be really similar, but we would need to do some digging and slight reverse engineering. I've outlined the steps in the [wiki](https://github.com/Kimplul/hid-tmff2/wiki#how-to-add-in-support-for-a-new-t-series-wheel), if anyone is interested.

D0V4HKIIN, 2023-01-15:
I might give it a try with my wheel on my pc and see if i can get anything working.

Kimplul, 2023-08-23:
Hello again, @davidedmundson recently got FFB for the TX working and submitted a patch. I merged it into `master` in https://github.com/Kimplul/hid-tmff2/commit/19c4f683e1e9ecada52c2731ea1a55bc294bd808, would be great if you could give it a go and report back. Currently `hid-tminit` is still missing support for the TX, so you will have to initialize the wheel with `tmrdv`.

SebiAi, 2023-10-06:
Works for me in *Euro Truck Simulator 2* and *Project CARS 2* with force feedback. I don't have a comparison to Windows therefore I don't know how accurate the force feedback is.

Here is the dmesg output (with added comments):
```
--- Initial plug in ---
[21104.820867] usb 1-9: new full-speed USB device number 44 using xhci_hcd
[21104.974276] usb 1-9: New USB device found, idVendor=044f, idProduct=b664, bcdDevice= 1.01
[21104.974290] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[21104.974295] usb 1-9: Product: Thrustmaster TX GIP Racing Wheel
[21104.974300] usb 1-9: Manufacturer: Thrustmaster
[21104.974304] usb 1-9: SerialNumber: 0000DAE35565C194
--- Execution of tmdrv script ---
[21118.677612] usb 1-9: USB disconnect, device number 44
[21119.312630] usb 1-9: new full-speed USB device number 45 using xhci_hcd
[21119.461702] usb 1-9: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[21119.461704] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[21119.461705] usb 1-9: Product: Thrustmaster FFB Wheel
[21119.461706] usb 1-9: Manufacturer: Thrustmaster
[21119.463064] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/0003:044F:B65D.001A/input/input55
[21119.520769] hid-thrustmaster 0003:044F:B65D.001A: input,hidraw6: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:00:14.0-9/input0
[21119.540473] hid-thrustmaster 0003:044F:B65D.001A: Unknown wheel's model id 0x104, unable to proceed further with wheel init
[21119.688998] usb 1-9: USB disconnect, device number 45
[21120.324587] usb 1-9: new full-speed USB device number 46 using xhci_hcd
[21120.473731] usb 1-9: New USB device found, idVendor=044f, idProduct=b669, bcdDevice= 1.00
[21120.473734] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[21120.473735] usb 1-9: Product: Thrustmaster TX Racing Wheel
[21120.473736] usb 1-9: Manufacturer: Thrustmaster
[21120.475079] input: Thrustmaster Thrustmaster TX Racing Wheel as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/0003:044F:B669.001B/input/input56
[21120.475192] hid-generic 0003:044F:B669.001B: input,hidraw6: USB HID v1.11 Joystick [Thrustmaster Thrustmaster TX Racing Wheel] on usb-0000:00:14.0-9/input0
[21120.581551] input: Thrustmaster Thrustmaster TX Racing Wheel as /devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/0003:044F:B669.001B/input/input57
[21120.640725] tmff2 0003:044F:B669.001B: input,hidraw6: USB HID v1.11 Joystick [Thrustmaster Thrustmaster TX Racing Wheel] on usb-0000:00:14.0-9/input0
[21120.668456] tmff2 0003:044F:B669.001B: force feedback for TX
```

The only thing I did not manage to do is to set up a proper `udev` rule for the `tmdrv` script, I just ran it as root - maybe someone can help me with that (`/etc/udev/rules.d/50-thrustmastertx.rules`):
```
SUBSYSTEMS=="usb", ATTRS{idVendor}=="044f", ATTRS{idProduct}=="b664,b65d,b669", GROUP="users", MODE="0666"
```
Yes, I did reload the `udev` rules (`sudo udevadm control --reload`) and reinserted the Thrustmaster TX.

---

System info:
> OS: Linux Mint 21.1 x86_64
> Kernel: 5.15.0-86-generic
> DE: Cinnamon 5.6.8
> WM: Mutter (Muffin)

franckyvrac, 2023-12-08:
Hi all,
is this works for Thrustmaster TMX ?

allylikesu, 2024-12-06:
Hello, I want to thank everybody in this thread and everybody involved in this project. After switching to Linux, I thought my expensive wheel setup would never be used again. After many months of periodically trying to get my TX working, I finally was able to get it working -- WITH force feedback -- in ETS2 for the first time last night. 

Although there is one problem, the steering deadzone. I tried the udev rules listed in the readme, and even successfully ran the jscal and evdev-joystick commands in my own shell, but the deadzone did not change at all. (And yes, my steering deadzone in ETS2 was set to 0%.)

pastaq, 2025-05-14:
I'm working on a kernel driver for initializing the the TMX. I figured since the TX also uses the same USB protocol I'd add support for it as well, but I need a tester for the TX wheel. Preferably someone who can build the kernel from source and install it without too much fuss and is fairly patient with iterating through code.

lowercasedots, 2025-11-01:
I'm seeing mentions of force feedback working on the TX in this thread, but I'm not sure if I'm missing something or if CachyOS has made some changes which somehow interefere with wheels.

I've installed this with the AUR package, installed the optional udev rules mentioned on the readme, and when I plug the wheel in I see it detected in dmesg. The wheel also does the initial calibration where it turns all the way left and right before centering itself.
I then go to the terminal to initialize the wheel with tmdrv as mentioned above: ./tmdrv.py -d thrustmaster_tx
The wheel then does another round of calibration and things seem okay - Automobilista 2 detects the wheel and I can set bindings, and Raceroom picks up a Thrustmaster TX profile without any fuss.

The issue is when I try to turn the wheel to say, calibrate the steering range, I get resistance as if the wheel is trying to re-center itself constantly. And this is the only "force" I get - once in game there's no actual FF response from the game. Has anyone else run into anything similar?

Kimplul, 2025-11-02:
Hi @lowercasedots, the symptoms you're describing (`fftest`'s "function not implemented" and only having autocentering) sound like what happens when the wheel is uninitialized and no driver is installed, but from the `dmesg` log it seems like the driver is getting loaded and picks up the wheel properly, which I don't think I've seen before. Seems like the issue is somewhere between the driver and FFB users (`fftest`, automobilista, etc.), and likely not related to hardware. Just to check, was the `dmesg` output collected at the same time that `fftest` was run? If it wasn't, double-check that you're not forgetting to run `tmdrv` every time you connect the wheel or something along those lines. I don't have a TX myself so I don't really know what quirks it might have, sorry.

lowercasedots, 2025-11-03:
Oh wow, well. There's one oversight then. Running fftest with `dev/input/by-id/usb-Thrustmaster_Thrustmaster_TX_Racing_Wheel-event-joystick` as the device does load the FFB options, although choosing any of them just.. doesn't actually do anyhting.

```
Device /dev/input/by-id/usb-Thrustmaster_Thrustmaster_TX_Racing_Wheel-event-joystick opened
Features:
  * Absolute axes: X, Y, Z, RZ, Hat 0 X, Hat 0 Y, 
    [27 00 03 00 00 00 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: Constant, Periodic, Ramp, Spring, Friction, Damper, Rumble, Inertia, Gain, Autocenter, 
    Force feedback periodic effects: Square, Triangle, Sine, Saw up, Saw down, 
    [00 00 00 00 00 00 00 00 00 00 FF 1F 03 00 00 00 ]
  * Number of simultaneous effects: 16

Setting master gain to 75% ... OK
Uploading effect #0 (Periodic sinusoidal) ... OK (id 0)
Uploading effect #1 (Constant) ... OK (id 1)
Uploading effect #2 (Spring) ... OK (id 2)
Uploading effect #3 (Damper) ... OK (id 3)
Uploading effect #4 (Strong rumble, with heavy motor) ... OK (id 4)
Uploading effect #5 (Weak rumble, with light motor) ... OK (id 5)
Enter effect number, -1 to exit
4
Now Playing: Strong Rumble
Enter effect number, -1 to exit
```

I'll try Speed Dreams - I just gave it one shot and it crashed while loading into Practice, so I'll have to look into it when I have more time this afternoon. However it was detecting input from the wheel for setting up bindings for steering/throttle/etc, but I didn't see where to configure FFB settings during my quick glance-through.

patternseek, 2025-11-03:
This looks extremely similar to my experience recently on Nixos with the TX ("leather edition", I don't know if theres any variance in the TX wheels).

I'll have another go at setting it up and collect my process and results in case they shine any additional light on the issue.

mjpawlowski, 2026-10-09:
re: https://github.com/Kimplul/hid-tmff2/issues/48#issuecomment-4952375615

Another TX data point backing up @oydoy firmware finding, plus what's going on underneath.

Setup: TX, Fedora 44 (kernel 7.2.x), hid-tmff2 `d890a93` via DKMS, tmdrv run from a udev rule.

**Firmware 56:** the driver binds and logs `force feedback for TX`, but the wheel ignores *everything* sent after init, not only effects. Writing `range` (270/540/900) doesn't move the end stops (they stay at about ±180°), and no `fftest` effect does anything. usbmon shows every interrupt OUT completing with status 0, so the packets arrive and the firmware drops them. That's also why zeroing gain/`FF_AUTOCENTER` made no difference.

**Firmware 58.06** (updated with Thrustmaster's Windows tool), same machine, same hid-tmff2 build: `range` works (the end stops move, and the axis spans the full 0–65535 at 270, 540 and 900) and all `fftest` effects work. The driver's USB traffic is byte-identical to the fw 56 capture, so the only change is the firmware.

So if writing `range` doesn't move the end stops, the firmware is too old. For the README: 56 doesn't work and 58.06 does. I haven't tested anything in between.

One caveat on the centering pull: on 58.06 the wheel still has a strong self-centering spring after init. That's the firmware's default autocenter, which hid-tmff2 leaves alone until something sets `FF_AUTOCENTER` (games and Oversteer usually do). Writing `FF_AUTOCENTER` (e.g. 10%) to the event device removes it, and the setting stays after the device is closed. If you still feel the pull after updating, that's the cause, not the old bug. Might be worth a default at init or a line in the README.

I also captured the official Windows driver (2.11.57.0, fw 58.06) with USBPcap: init, range, Test Forces, gain/autocenter. Range, gain and autocenter packets match hid-tmff2's. Windows also sends a vendor control request `41 48 0040` at init and sets bit 0x80 on the effect upload opcodes (`ea/eb/e4/e9`, play `89 01`), but 58.06 doesn't need either. Windows scales range as `range * 0xffff / 1080` (900° → `0xd555` vs `0xd2f0`), about 1% off.

@lowercasedots @patternseek your symptoms (effects upload OK but nothing happens, only the centering pull) look like this. Check the firmware version in Thrustmaster Control Panel if you have access to a Windows PC.

patternseek, 2026-10-09:
Yep. I actually spent a week of evenings with Claude trying to get my wheel working and got a fully working driver for this version working... Then realised all I needed was a firmware upgrade. I forget the versions but I was above the one stated in the readme but not fully up to date. So yes likely the same issue. The protocol that worked with my firmware version was totally different to the one in the driver. Very strange.

Kimplul, 2026-10-12:
README should be clarified, the specified firmware version really only applies to the T300, as that model was (at least up until now, hah) the only one with known incompatibilities. Each model has its own firmware version, so they're not directly comparable (so for example version 58 on TX could be older than 34.1 on T300, for all I know).

I'll add a note that 58.06 is the currently known oldest known working firmware.