Kimplul/hid-tmff2 #50: T500RS detection

Kimplul, 2022-10-29:
Hi @hoover67, I took the liberty to open up an issue for you in reference to #49 , hope you can at least comment on issues.

Just so we're clear, this driver is just the basis for the T500 driver. I don't own a T500 myself, and @cazzoo kindly offered to try reverse-engineering the wheel, but he hasn't had much time to do so. As such, the FFB won't work even if the wheel is detected properly, sorry.

If you'd like to help with the reverse-engineering, there's a short description of what has to be done in the [wiki](https://github.com/Kimplul/hid-tmff2/wiki#how-to-add-in-support-for-a-new-t-series-wheel).

Anycase, the actual issue you reported is technically in https://github.com/scarburato/hid-tminit. Thrustmaster wheels have a pretty dumb thing where they require the computer to first initialize them (such as telling the wheel which model it is, wtf?), and the `hid-tminit` driver is specifically for this as multiple wheels use more or less the same init sequence. The `hid-tmff2` driver then takes over and provides FFB for the initialized wheel.

From the log it seems that the error is `-EPIPE`, which according to [this](https://www.kernel.org/doc/html/latest/driver-api/usb/error-codes.html#error-codes-returned-by-in-urb-status-or-in-iso-frame-desc-n-status-for-iso) might be related to hardware issues. I remember contacting Thrustmaster about something similar and apparently it is recommended to use USB3 ports and try different ports. Not sure if it's relevant here, but probably worth a shot if you didn't already try it.

@cazzoo Have you run into anything similar with your wheel?

cazzoo, 2022-10-29:
I don't remember facing this issue, no. 
Just to add more clarifications on what @Kimplul said, I started to "try" to reverse engineer the T500 RS driver but I had to put that work on pause for personal reasons. 

What I can suggest to you @MmAaXx500 is to try to get my fork and try to move on with it. You can open PR/MR onto my project. I'd be glad to review / include so we can then propose the driver to this repo. 

If you already have good knowledge, smart mind and a lot of spare time but no experience yet in USB packet sniffing, we can probably schedule a sharing session to hand my investigations over to you. 
Let me know 

Cheers

Kimplul, 2023-03-24:
No response in a while so I'm closing this and the related pull request. I would suggest further discussion related to the T500 to be held in #18 for now.

@cazzoo Do you mind if i ping #18 with a request for volunteers to continue with the reverse engineering, and maybe add a notice about it in the README? The topic's been a bit stale for a while now, thought I might gage interest.

closed 2023-03-24 by Kimplul

cazzoo, 2023-03-24:
In fully fine with this. You can call for volunteers. 
Fun fact: I just played around with chat GPT to help decoding hex dumps I produced a long time ago. While it's far from being perfect, it can surely help to decode the captures and ease the development of the driver. 
That said, I was far from my personal setup and still I don't have time to spend working on the driver but definitely i'm fine to review contributions on my fork