Kimplul/hid-tmff2 #51: Remove/Reinstall Driver

sayapinsergej, 2022-11-08:
Hi! How to remove/reinstall the driver?

Kimplul, 2022-11-08:
Reinstallation is basically just running `make && sudo make install`, removing the driver can be done with `rm /usr/lib/modules/$(uname -r)/extra/hid-tmff2.ko`. This is assuming you installed the driver manually, `dkms` works with the standard `dkms remove ...`.

Since this is a kernel module, you might want to reinstall your machine for any changes to take effect. If you're not patient enough, something like `sudo rmmod hid-tmff-new && sudo modprobe hid-tmff-new` should force the kernel to reload the driver.

MicHaeL-MonStaR, 2024-05-24:
> Reinstallation is basically just running `make && sudo make install`...

I'm trying to do this, as I suspect the driver broke somehow, but it just returns: `make: *** No targets specified and no makefile found.  Stop.`

I'm guessing I need to designate a target, but I don't know how that's done.

Kimplul, 2024-05-24:
Sounds like you're in the wrong directory, check that you're in the top directory to where you cloned the repository. Running `ls` should show (among other things) a file called `Makefile`, which is what `make` reads to know how to compile the driver.

MicHaeL-MonStaR, 2024-05-24:
I didn't navigate to any directory as I don't know about that. - I also don't know where I would have cloned the repository. I've only followed installation-instructions about half a year ago and so it would be whatever it is by default.

It shows a hid-tmff2 folder in Home. Is that where I should run it from (or run the Terminal from in the first place)?

Kimplul, 2024-05-24:
> It shows a hid-tmff2 folder in Home. Is that where I should run it from (or run the Terminal from in the first place)?

Yep, sounds like that would be it, just go `cd hid-tmff2` in a terminal and continue with the installation. In general, I would recommend cloning repositories somewhere other than your home directory, but as long as it works I guess.

> I suspect the driver broke somehow,

I suspect you updated your system, and it pulled in a new version of Linux. Whenever this happens, the driver has to be compiled again for the newer kernel. This is done automatically by DKMS, and I would recommend you try that install method Otherwise you'll have to periodically run `make && make install` which can get a bit annoying.

MicHaeL-MonStaR, 2024-05-24:
Yes, Pop!_OS updates a lot of stuff regularly and I rarely change anything manually. I haven't in recent days, but somewhere between just a few days ago and today something changed and the wheel stopped responding to Overdrive and game-feedback (though it was still detected). So it's hard to tell what all changes with "automatic" updates.

I was looking at the installation-methods but didn't even remember which one I used, so apparently it would be the manual one. - I there a way to switch to the DKMS-method? Or is it just a matter of doing it that way and it will overwrite it?

For now it seems reinstalling worked, although it did indicate some things about different versions indeed.

Kimplul, 2024-05-24:
> I there a way to switch to the DKMS-method? Or is it just a matter of doing it that way and it will overwrite it?

Just perform a DKMS install as normal. You might find in `dmesg` that Linux complains about duplicated drivers if you install with both methods, but that will sort itself out the next time you update Linux, no need for any kind of manual intervention.

closed 2024-05-24 by Kimplul