Kimplul/hid-tmff2 #52: T300 installation

CD1728, 2022-11-30:
Hello,
I'm looking to get my T300 + th8 shifter working.
I am a beginner with Linux and I can't manage to do the installation correctly.
My t300 is recognized but no FFB!
I am under Ubuntu
Sorry for the beginner questions!
After entering the command to download the file, or should I enter "make" and then "sudo make install"?
The seized "DKMS" is it to be made in more?

Thanks for your help

Kimplul, 2022-11-30:
Hi.

> After entering the command to download the file, or should I enter "make" and then "sudo make install"?

Sorry, I'm not entirely sure what you're asking, but I'm guessing you mean you ran `make` and `sudo make install` and if there's anything else you should do? No, the kernel module should be installed and plugging in your wheel should load the driver automatically.

> The seized "DKMS" is it to be made in more?

`DKMS` is a tool that will automatically recompile the driver when you upgrade your kernel, so assuming you're asking whether you should use it: Probably.

As for no FFB, could you post the output of  `sudo dmesg -w` when you plug the wheel in after having installed the driver?

So, in steps

1. Install the driver
2. Run `sudo dmesg -w`
3. Plug the wheel in
4. Post what happened in the `dmesg` log.

To test FFB, you could try running `fftest /dev/input/by-id/usb-Thrustmaster_Thrustmaster_T300RS_Racing_wheel-event-joystick` and following the on-screen instructions.

CD1728, 2022-11-30:
I have installed GCC
When I enter the hid-tmff2/hid-tminit directory, I still get this error message for the make command

christian@chrisPC:~/hid-tmff2/hid-tminit$ make
make -C /lib/modules/5.15.0-53-generic/build M=/home/christian/hid-tmff2/hid-tminit modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-5.15.0-53-generic »
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
make[1] : on quitte le répertoire « /usr/src/linux-headers-5.15.0-53-generic »

Kimplul, 2022-11-30:
> The question with "make" was where do I enter it? I have to open a terminal in the hid-tmff2 directory? that's what I did but it doesn't seem to work, I have error messages (gcc not found, error 127, error 2)

Oh, okay, you need to install gcc. On Ubuntu, this should be `sudo apt install build-essential`. This installs other stuff you might need when compiling stuff from source. You'll also need kernel headers, `sudo apt install linux-headers-$(uname -r)` should work.

And yes, you should run the commands in the directory where you downloaded this project, same place where there's a file called `Makefile`.

Thanks for the `dmesg` output, try installing the driver with the instructions above and check if there's any FFB. Otherwise, I'd like to see the `dmesg` again, though you don't need to copy all of it, just the bit that is generated when you plug in the wheel. In the output you posted, this would be 

> [ 940.462597] usb 3-3: new full-speed USB device number 8 using xhci_hcd
[ 940.642895] usb 3-3: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[ 940.642901] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 940.642904] usb 3-3: Product: Thrustmaster FFB Wheel
[ 940.642907] usb 3-3: Manufacturer: Thrustmaster
[ 940.691969] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:07.1/0000:0a:00.3/usb3/3-3/3-3:1.0/0003:044F:B65D.0006/input/input20
[ 940.692217] hid-thrustmaster 0003:044F:B65D.0006: input,hidraw2: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:0a:00.3-3/input0
[ 940.712878] hid-thrustmaster 0003:044F:B65D.0006: Wheel with model id 0x200 is a Thrustmaster T300RS (Missing Attachment)
[ 940.713997] usb 3-3: USB disconnect, device number 8
[ 940.714580] hid-thrustmaster 0003:044F:B65D.0006: URB to change wheel mode seems to have failed with error -2
[ 941.387017] usb 3-3: new full-speed USB device number 9 using xhci_hcd
[ 941.554887] usb 3-3: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[ 941.554893] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 941.554896] usb 3-3: Product: Thrustmaster T300RS Racing wheel
[ 941.554898] usb 3-3: Manufacturer: Thrustmaster
[ 941.572283] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:07.1/0000:0a:00.3/usb3/3-3/3-3:1.0/0003:044F:B66E.0007/input/input21
[ 941.572464] hid-generic 0003:044F:B66E.0007: input,hidraw2: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:0a:00.3-3/input0

Out of curiosity, have you compiled anything else from source before?

Kimplul, 2022-11-30:
Oh, you posted a new comment rights as I was finishing my last one. Fair enough.

> I have installed GCC
When I enter the hid-tmff2/hid-tminit directory, I still get this error message for the make command

You don't need to enter `hid-tminit`, the `Makefile` in `hid-tmff2` does everything you need.

Unless I'm reading the output wrong, it seems to just be a warning and can probably be ignored.

CD1728, 2022-11-30:
Well, already the wheel has reacted!

"apparmor_parser"
[ 2678.738607] usb 3-3: USB disconnect, device number 9
[ 2685.631012] usb 3-3: new full-speed USB device number 10 using xhci_hcd
[ 2685.811393] usb 3-3: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[ 2685.811400] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2685.811403] usb 3-3: Product: Thrustmaster FFB Wheel
[ 2685.811405] usb 3-3: Manufacturer: Thrustmaster
[ 2685.846627] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:07.1/0000:0a:00.3/usb3/3-3/3-3:1.0/0003:044F:B65D.0008/input/input22
[ 2685.846804] hid-thrustmaster 0003:044F:B65D.0008: input,hidraw2: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:0a:00.3-3/input0
[ 2685.869366] hid-thrustmaster 0003:044F:B65D.0008: Wheel with model id 0x200 is a Thrustmaster T300RS (Missing Attachment)
[ 2685.871008] usb 3-3: USB disconnect, device number 10
[ 2685.871826] hid-thrustmaster 0003:044F:B65D.0008: URB to change wheel mode seems to have failed with error -2
[ 2685.874992] hid_tminit: loading out-of-tree module taints kernel.
[ 2685.875046] hid_tminit: module verification failed: signature and/or required key missing - tainting kernel
[ 2686.543025] usb 3-3: new full-speed USB device number 11 using xhci_hcd
[ 2686.711385] usb 3-3: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[ 2686.711392] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2686.711395] usb 3-3: Product: Thrustmaster T300RS Racing wheel
[ 2686.711397] usb 3-3: Manufacturer: Thrustmaster
[ 2686.728759] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:07.1/0000:0a:00.3/usb3/3-3/3-3:1.0/0003:044F:B66E.0009/input/input23
[ 2686.728934] hid-generic 0003:044F:B66E.0009: input,hidraw2: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:0a:00.3-3/input0
[ 2686.847025] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:07.1/0000:0a:00.3/usb3/3-3/3-3:1.0/0003:044F:B66E.0009/input/input24
[ 2686.847214] tmff2 0003:044F:B66E.0009: input,hidraw2: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:0a:00.3-3/input0
[ 2686.851386] tmff2 0003:044F:B66E.0009: force feedback for T300RS

CD1728, 2022-11-30:
Progress!
But I only feel strength for the 0 or 1 effect, nothing for the others !

oystick opened
Features:
  * Absolute axes: X, Y, Z, RZ, Hat 0 X, Hat 0 Y, 
    [27 00 03 00 00 00 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: Constant, Periodic, Ramp, Spring, Friction, Damper, Rumble, Inertia, Gain, Autocenter, 
    Force feedback periodic effects: Square, Triangle, Sine, Saw up, Saw down, 
    [00 00 00 00 00 00 00 00 00 00 FF 1F 03 00 00 00 ]
  * Number of simultaneous effects: 16

Setting master gain to 75% ... OK
Uploading effect #0 (Periodic sinusoidal) ... OK (id 0)
Uploading effect #1 (Constant) ... OK (id 1)
Uploading effect #2 (Spring) ... OK (id 2)
Uploading effect #3 (Damper) ... OK (id 3)
Uploading effect #4 (Strong rumble, with heavy motor) ... OK (id 4)
Uploading effect #5 (Weak rumble, with

Kimplul, 2022-11-30:
Great, you got the driver installed!

0 and 1 are passive effects, 2 and 3 change how the wheel responds to rotation. Spring makes the wheel want to center itself and damper makes turning more stiff. 4 and 5 are sort of weird as their force direction is 0, so I'm not entirely sure how they should be handled.

Anycase, try out some games, see how they feel.

CD1728, 2022-11-30:
ok, to modify the forces (e.g. 75%) how do I do? Do I have to install "Oversteer" as well?

Kimplul, 2022-11-30:
I recommend Oversteer, yes. There's a public API that can be accessed from the command line if you really want to, but Oversteer is easier.

CD1728, 2022-11-30:
ok, thanks a lot for the help. I will try to do some tests...

CD1728, 2022-11-30:
Sorry, just a little more help,
I don't understand how to install oversteer !
there are links but not specially for ubuntu !
I managed to download and open the Zip file, but I can't install it!

Kimplul, 2022-11-30:
Which part of the instructions are you having trouble with?

https://github.com/berarma/oversteer#build-and-install

fuzunspm, 2022-11-30:
> Sorry, just a little more help, I don't understand how to install oversteer ! there are links but not specially for ubuntu ! I managed to download and open the Zip file, but I can't install it!

`apt install python3 python3-distutils python3-gi python3-pyudev python3-xdg python3-evdev gettext meson appstream-util desktop-file-utils python3-matplotlib python3-scipy`

then

`pip install --user oversteer` or

Kimplul, 2022-11-30:
Sorry, but where are you getting that you should run `pip`? I can't find it in the README.

Did you try the install instructions I linked?

fuzunspm, 2022-11-30:
yeah you are right, I though @CD1728 wants a quick installation method but should not be using pip since it's outdated

CD1728, 2022-11-30:
yes, I did what you said above!
But in the meantime I managed to install but it seems to be an old version that doesn't even recognize the steering wheel! And of course it's not as easy for me as uninstalling a windows program!

CD1728, 2022-11-30:
In fact for me the explanations on the page are not very understandable, as said @fuzunspm , I would need the (simple) procedure from a to Z...;)

Kimplul, 2022-11-30:
Sorry, closed by mistake.

Kimplul, 2022-11-30:
Reboot?

CD1728, 2022-11-30:
the installation I found earlier installed this version 0.1.3 (which does not recognize the wheel!) https://pypi.org/project/oversteer/

I will restart ....

CD1728, 2022-11-30:
christian@chrisPC:~$ meson build

ERROR: Neither directory contains a build file meson.build.
christian@chrisPC:~$

CD1728, 2022-11-30:
Well, being in the directory, the "meson" command seems to work !
Only I don't see any way to launch oversteer ! The only icon I have is the one I installed earlier (see the link :  https://pypi.org/project/oversteer/
How can I uninstall this ?

Kimplul, 2022-11-30:
> Only I don't see any way to launch oversteer ! The only icon I have is the one I installed earlier (see the link : 

You can run `oversteer` on the command line, though you should first uninstall the `pip` version if you haven't already. I suspect the icon will fix itself after a reboot.

> How can I uninstall this ?

Assuming you mean the oversteer you installed though `pip`, `pip uninstall oversteer`. Otherwise, the uninstall instructions are in the README right under the installation instructions.

Kimplul, 2024-11-03:
Doing some winter cleaning, forgot to close this issue way back when so doing it now.

closed 2024-11-03 by Kimplul