Kimplul/hid-tmff2 #53: Rumble effects do not work

peter-vasut, 2023-01-15:
Hello,

I've noticed that Dirt Rally 2 doesn't allow me to set "Tyre Slip" and "Engine" options in "vibration & feedback" menu. The **steering effects work**, but there is no **vibration**.

I've tried `fftest` on the device. It uploaded all effects without problem, but only "periodic sinusoidal" and "constant" do anything (they move the steering wheel), but nothing else (strong/weak rumble etc.) works. (There is no error, it's just nothing happens.)

Are these effects known to work or is this known issue?

Wheel firmware version: **34** (this is the one provided by the updater I could download from thrustmaster website, not sure if there is way to downgrade).

Kimplul, 2023-01-15:
> I've tried fftest on the device. It uploaded all effects without problem, but only "periodic sinusoidal" and "constant" do anything (they move the steering wheel), but nothing else (strong/weak rumble etc.) works. (There is no error, it's just nothing happens.)

Hello, the `fftest` rumble effects not working is a known 'issue', I suppose. The problem is that `fftest` [sets the direction of the rumbling effect to zero](https://github.com/flosse/linuxconsole/blob/master/utils/fftest.c#L308) (or, rather, doesn't set any value to it and is implicitly zero). The direction of the effect determines the actual strength as a proportion of the requested effect strength, more or less following [`actual_strength = strength * sin(direction)`](https://github.com/Kimplul/hid-tmff2/blob/master/hid-tmt300rs.c#L632). Or, this is how all effects I've observed have worked, I could be missing something.

There is an additional detail in that this driver itself doesn't report supporting rumble effects, rather the Linux subsystem [tries to emulate rumbling with periodic effects](https://elixir.bootlin.com/linux/latest/source/drivers/input/ff-core.c#L53), which the driver does report supporting. The subsystem doesn't do anything to the direction, though.

The damper and spring should also work in `fftest`, but they're passive effects, meaning that you start to feel them only when moving the wheel.

> I've noticed that Dirt Rally 2 doesn't allow me to set "Tyre Slip" and "Engine" options in "vibration & feedback" menu. The steering effects work, but there is no vibration.

DiRT Rally 2 not doing anything with some options is sort of known, iirc the game only ever uploads one effect to the wheel and the steering effects are all done with the single effect. This seems to be pretty common, most games I've tried seem to do this (at least with this driver), but in the case of DiRT I don't know if there's some internal condition that is failing and some effects aren't uploaded to the wheel because of it or what's going on, really very sorry.

peter-vasut, 2023-01-15:
Thank you for reply. You are right, all the affects are working, the friction and spring are just subtle. I've also tried to change the direction in the code you linked, it's as you say, when you change the direction, the rumbling effects work.

So you are right, it's not hardware or driver issue, it's probably something to do with Dirt Rally 2.

peterlopen, 2023-01-16:
@peter-vasut  please, what is this fftest utility you are using? where could I get it? thank you.

Kimplul, 2024-11-03:
Doing some winter cleaning, closing this as completed even though the resolution was maybe a bit unfortunate.