Kimplul/hid-tmff2 #55: Can you add the T-GT II please

riourik, 2023-01-28:
Hi I have a T-GT II and I would like to use it on linux (I use pop os)
I install the driver after figure it out for the ssl problem My wheel is not reconize 
dmesg

> [  384.552140] usb 5-1: USB disconnect, device number 12
> [  384.552145] usb 5-1.1: USB disconnect, device number 13
> [  386.192095] usb 5-1: new high-speed USB device number 14 using xhci_hcd
> [  386.343422] usb 5-1: New USB device found, idVendor=05e3, idProduct=0610, bcdDevice=32.98
> [  386.343427] usb 5-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
> [  386.343429] usb 5-1: Product: USB2.0 Hub
> [  386.394712] hub 5-1:1.0: USB hub found
> [  386.395172] hub 5-1:1.0: 2 ports detected
> [  386.524294] hub 5-1:1.0: USB hub found
> [  386.524791] hub 5-1:1.0: 2 ports detected
> [  386.875771] usb 5-1.1: new full-speed USB device number 15 using xhci_hcd
> [  386.992678] usb 5-1.1: New USB device found, idVendor=044f, idProduct=b66d, bcdDevice= 7.00
> [  386.992681] usb 5-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [  386.992682] usb 5-1.1: Product: Thrustmaster Racing Wheel FFB
> [  386.992683] usb 5-1.1: Manufacturer: Thrustmaster
> [  387.071842] input: Thrustmaster Thrustmaster Racing Wheel FFB as /devices/pci0000:00/0000:00:08.1/0000:2f:00.3/usb5/5-1/5-1.1/5-1.1:1.0/0003:044F:B66D.0010/input/input36
> [  387.071948] tmff2 0003:044F:B66D.0010: input,hiddev2,hidraw2: USB HID v1.10 Gamepad [Thrustmaster Thrustmaster Racing Wheel FFB] on usb-0000:2f:00.3-1.1/input0
> [  387.074675] tmff2 0003:044F:B66D.0010: firmware version 10 is too old, please update.
> [  387.074679] tmff2 0003:044F:B66D.0010: note: this has to be done through Windows.
> [  387.074680] tmff2 0003:044F:B66D.0010: failed initializing T300RS
> [  387.074681] tmff2 0003:044F:B66D.0010: init failed
> [  387.168318] tmff2: probe of 0003:044F:B66D.0010 failed with error -22

lsusb

> Bus 005 Device 017: ID 044f:b66d ThrustMaster, Inc. Thrustmaster Racing Wheel FFB

last firmware for the tgt2 base is the v10

So What do you need to add the T-GT II ? I have a laptop with windows on it if need information from windows side.
Thanks for reading me :)

Kimplul, 2023-01-28:
Hi, I'm somewhat surprised that your wheel is using the same USB product ID as the T300RS in PS4 mode. I don't have the T-GT II myself, is there a mode switch button on it and which mode are you in, if there is one?

Anycase, if you suspect your wheel works with the same set of FFB commands that a T300RS does, you can try temporarily commenting out https://github.com/Kimplul/hid-tmff2/blob/master/hid-tmt300rs.c#L1486 and skip the firmware version check. If it works, we can figure out how to differentiate with your wheel and a T300 later.

If it doesn't work, I have a short overview of what needs to be done to get a new device working with this driver in the wiki:
https://github.com/Kimplul/hid-tmff2/wiki#how-to-add-in-support-for-a-new-t-series-wheel

I won't lie, collecting and reverse-engineering the data is pretty tedious, but if you think you're up for it feel free to give it a go. I'll help best I can, but without a physical device to test with that help is likely somewhat limited.

Be warned, Thrustmaster seems to have a tendency to not really reuse code from one driver to the next, and I got lucky that the T248 I bought worked almost identically. There was a case where we thought a TX wheel would behave like a T300, but that turned out to be false: https://github.com/Kimplul/hid-tmff2/issues/48

riourik, 2023-01-29:
hi thank for the quick answerd so
`you can try temporarily commenting out https://github.com/Kimplul/hid-tmff2/blob/master/hid-tmt300rs.c#L1486`
I did it and the steering, gas, brake, and clutch work I have to check I have a dead zone for the first couple degres
triangle cercle square X work directionnal button  L2 R2  SH OP L3 R3 work

for the not working button 
I have the 2 stick and the 4 button (rotation and click)

I guess the tgt have more button

I have a GT mode on the steering not working on windows

lsusb:

> Bus 005 Device 006: ID 044f:b681 ThrustMaster, Inc. Thrustmaster Racing Wheel FFB T-GT

Im up to make my tgt works at 100%  (if possible) I make 1 full week and lots of hour to install the driver on an other distro but I finish to try pop os and its working

I will use fedit on my windows laptop and check what I need to do

riourik, 2023-01-29:
not sure If i did it correctly
[tgt2.zip](https://github.com/Kimplul/hid-tmff2/files/10528658/tgt2.zip)

riourik, 2023-01-29:
hi I do the record on my laptop under windows it's for that I don't have the same IDs
for the noise i'm not sure how to reduce it... I did a record with ffb to 0 but I believe they some noise cause when i record and not moving that record something I dont what to do with fedit (dont find much online how to use it)
[tgt ffb mini.zip](https://github.com/Kimplul/hid-tmff2/files/10529967/tgt.ffb.mini.zip)

Kimplul, 2023-01-29:
What I meant by noise is that you can filter out anything that goes to `x.y.2`, as we're (probably) mostly just interested in `x.y.1`, where the force feedback commands go. It's fine to leave it in when sharing logs, as I can filter it out later, but keep it in mind when you're doing stuff yourself.

As for `fedit`, force commands are built up of different parameters, and `fedit` lets you see which effect types take which parameters. Probably the most important one is the direction, which essentially works as a scalar for the actual strength of an effect, approximately `actual_strength = strength * sin(direction)`. So 0 direction means zero force, and in `fedit` 0 is pointing straight up. Try moving the direction to the right, that should be maximum force. All the way to the left should invert the effect.

Each effect is first uploaded to the wheel, when you finish configuring it. After uploading it, press the play button to play the effect. Effects have a length, which is for how long the effect should be active, and a delay, which is how long the effect waits after pressing play to actually start. The delay is how far the effect is from the start of the timeline in the main view.

Try playing around with it for a while, I'm sure you'll get the hang of it.

riourik, 2023-01-29:
the steering go up to 1080 degre do you want I many register like first at 1080 second a 720 etc... and do you fwant like 1 register for like center to left and back to center then a second record for center to right to center or that can be in the same record ? 
and do you want like some leftrightleft like when we touch a border ? 
and do you want i record only the button  and 1 just the pedal ?
What will be the best to do?

Kimplul, 2023-01-29:
We probably don't need to worry about buttons, the HID interface handles them more or less automatically. What we might need to do is capture and modify the wheel's `rdesc`, though it should already be captured in one of your zip files. The `rdesc` contains the info that the HID subsysten in Linux uses to map the different buttons.

For now I would just focus on the FFB and `fedit`.

riourik, 2023-02-07:
Just a small update I have a lots of hard time to set up fedit I try to lunch it as win xp compatibility and its more easy now so probably this week end I will record the ffb just a bit busy last week and this week but I should be able to work on it this week end :)

Querzion, 2023-03-13:
"[querzion@vmc ~]$ lsusb | grep ThrustMaster
Bus 005 Device 018: ID 044f:b695 ThrustMaster, Inc. TSS Handbrake
Bus 005 Device 021: ID 044f:b66e ThrustMaster, Inc. Thrustmaster T300RS Racing wheel
Bus 005 Device 019: ID 044f:b660 ThrustMaster, Inc. T500 RS Gear Shift"

I have the T300RS GT version, there's a irritating bug where the throttle and clutch is 100% all the time, In some games:

Grid, Grid Legends & Project Cars... I could unbind them and bind the pedals again, and they get an z axel input instead. The wheel just worked in Dirt Rally & Dirt Rally 2.0 without any tinkering. 

T300RS has a 2 pedal, 3 pedal and a 3 pedal pro option.

I'm thinking, if I buy the Pro pedals, will the problems go away?

Kimplul, 2023-03-13:
@Querzion Yeah, it's a known issue. I have no idea why different games expect different pedal mappings, as far as I'm aware there's no way to account for this in the driver. Changing the default pedal mappings to fix one game would just make other games' defaults wrong.

Pro pedals are unlikely to fix the issue, as the wheel itself reports which pedal is mapped to each axis. It is up to the games to interpret the axes to gas/brake/clutch, and, as you've noticed, the axis values themselves. Changing pedals might change which axis corresponds to which pedal, but I don't see how all games would suddenly start interpreting the axes correctly, sorry.

If you do buy the Pro pedals, please report back if anything changed, I'd love to be wrong.

In case you're curious, I compiled a small list of default mappings in some games: https://github.com/Kimplul/hid-tmff2/issues/31#issuecomment-985543679

Querzion, 2023-03-13:
@Kimplul Thanks. I got Dirt 4 to work because of that link. ^^ With force feed back, all the other games only has like spring effect on the motor and not force feedback, which kind of caught me off guard to be honest. I need to fix the settings so that it's not as violent. 

Now I start to wonder if I can get FFB to work in other games too. 

Regarding the pedals, I'm not sure if I'm going to buy them, but it's probably second hand, since the pro doesn't seem to be sold anymore. The one's that is part of this kit, is the new 3 pedals, that isn't as robust as the old one's.

samsunix, 2024-12-12:
2024 And Thrustmaster still doesnt chip in with Linux drivers... Could someone please update this or let me know what need to Be done to fix this. I really want drop Windows and I really don't want drop $700 wheel for driver issue.

Kimplul, 2024-12-12:
Hello @samsunix, I don't have a T-GT II myself and have no idea if anything's been happening beyond this issue, so I presume we're effectively where this thread started.

I guess I never got a full confirmation if commenting out the firmware version check worked, but presumably it didn't, so we're down to making some USB packet captures of traffic between the wheel and the host computer. I have a short overview of how that would be done: https://github.com/Kimplul/hid-tmff2/wiki#how-to-add-in-support-for-a-new-t-series-wheel

To start with, it should be enough to capture a couple packets, for example just setting the wheel range in Windows. Depending on the format of the packets, there are a few ways forward. If the format matches other wheels this driver supports, we should be fairly close to a working wheel, possibly with some configuration tweaks to the T300 code. Alternatively, we might see a replication of what happened with the TMX, see https://github.com/Kimplul/hid-tmff2/issues/78. Worst case, format doesn't look familiar at all, and you'd need to go through all effects supported by the wheel and build up some model for what bits in a message do what, like in `docs/FFBEFFECTS.md`.

samsunix, 2024-12-13:
Hello Kimplul. 

I have recent firmware v.10 

Here's the output (35s) turning wheel far left and right
[Link](https://www.dropbox.com/scl/fi/us7hf2u7cd2tj0a22l6um/wheel-maxes.pcapng?rlkey=lrd5h8xnrtw4f1yj649tgv48o&st=c8cj1n1o&dl=0)

Kimplul, 2024-12-15:
I maybe should've specified that the capture should be of doing something related to FFB, like setting the gain or autocentering of the wheel from the Thrustmaster control panel. Just turning the wheel doesn't quite give me enough information, unfortunately.

samsunix, 2024-12-15:
Hello @Kimplul , Here you go.

https://youtu.be/MJA6-roUR7o
[data](https://www.dropbox.com/scl/fi/rbyf33e3aqofmjdelv33u/thrusmasterffbtest.pcapng?rlkey=04gs919p050wttfk8fjklwd4p&st=5fexgkaf&dl=0)

Kimplul, 2024-12-15:
Alright, thanks. Choosing an FFB packet, for example number 9289, seems to match the T300, with a `60 00 01 ...` format. This is good news, at least on a surface level the wheels seem to use the same command set.

At this point, maybe try just installing the driver on Linux (probably removing the firmware version check) and report back.

samsunix, 2024-12-15:
Will proceed. This will be done by tomorrow afternoon as I need to move from windows -> linux

samsunix, 2024-12-15:
Not sure which part to comment from code but taking educated guess:
```
        /* fetch firmware version */
/*      ret = usb_control_msg(t300rs->usbdev,
                        usb_rcvctrlpipe(t300rs->usbdev, 0),
                        t300rs_fw_request.bRequest,
                        t300rs_fw_request.bRequestType,
                        t300rs_fw_request.wValue,
                        t300rs_fw_request.wIndex,
                        fw_response,
                        t300rs_fw_request.wLength,
                        USB_CTRL_SET_TIMEOUT
                        );

        if (ret < 0) {
                hid_err(t300rs->hdev, "could not fetch firmware version: %i\n", ret);
        goto out;
        }

        /* educated guess */
/*      if (fw_response->fw_version < 31 && ret >= 0) {
                hid_err(t300rs->hdev,
                                "firmware version %i is too old, please update.\n",
                                fw_response->fw_version
                       );

                hid_info(t300rs->hdev, "note: this has to be done through Windows.\n");

                ret = -EINVAL;
                goto out;
        }

        /* everything OK */
        ret = 0;
```

samsunix, 2024-12-16:
just a little note while trying to understand the issue, kubuntu gives following error when:
```
smx@smxplays:~/hid-tmff2$ sudo ./dkms/dkms-install.sh
./dkms/dkms-install.sh: 23: dkms: not found
./dkms/dkms-install.sh: 24: dkms: not found
./dkms/dkms-install.sh: 25: dkms: not found
```

As a side note, this is now 3rd attempt as I already tried twice with popos (latest)
and for unknown reason after performing given steps (could be headers, don't know), after reboot it 100% broke the whole installation and OS wasn't usable anymore due any USB device is not responding.

Kimplul, 2024-12-16:
Sounds like you'd need to install `dkms`, so something like `sudo apt install dkms` on Ubuntu-based distros. You might also want to go with the manual installation method while testing things, but it shouldn't really matter.

As for breaking an installation, that's new to me. Can you give a detailed list of things you did so I can try to replicate?

samsunix, 2024-12-16:
yeap that did the trick... rebooted but maybe unsubrisingly wheel doesn't appear on steam... Continued with windows driver instructions but currently stuck on this:

`smx@smxplays:~$ protontricks -c 'wine ~/Downloads/2024_TTRS_1.exe' 1551360
Traceback (most recent call last):
  File "/usr/bin/protontricks", line 8, in <module>
    sys.exit(cli())
             ^^^^^
  File "/usr/lib/python3/dist-packages/protontricks/cli/main.py", line 32, in cli
    main(args)
  File "/usr/lib/python3/dist-packages/protontricks/cli/util.py", line 159, in wrapper
    return cli_func(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/protontricks/cli/main.py", line 339, in main
    proton_app = find_proton_app(
                 ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/protontricks/steam.py", line 848, in find_proton_app
    tool_app = find_steam_compat_tool_app(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/protontricks/steam.py", line 610, in find_steam_compat_tool_app
    section for section in iter_appinfo_sections(appinfo_path)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/protontricks/steam.py", line 531, in iter_appinfo_sections
    raise SyntaxError(
SyntaxError: Invalid file magic number. The appinfo.vdf version might not be supported by the current version of Protontricks - please check for updates.
`

I know this is extending your support but maybe someone knows as likely many here faced same issues....
"apparantly" this is protontricks issue and should work with beta but rabid hole just gets deeper... 

```
smx@smxplays:~$ flatpak install flathub-beta com.github.Matoking.protontricks

Note that the directories 

'/var/lib/flatpak/exports/share'
'/home/smx/.local/share/flatpak/exports/share'

are not in the search path set by the XDG_DATA_DIRS environment variable, so
applications installed by Flatpak may not appear on your desktop until the
session is restarted.

Looking for matches…
error: The application com.github.Matoking.protontricks/x86_64/beta requires the runtime org.freedesktop.Platform/x86_64/23.08 which was not found
```

samsunix, 2024-12-16:
I have done now hid-tmff2 and [driver](https://github.com/Kimplul/hid-tmff2/blob/master/docs/DRIVER.md) installation succesfully but wheel doesn't still pop up on steam.

Kimplul, 2024-12-16:
`1551360` seems to be Forza Horizon 5, generally it'd be a good idea to mention what games you're testing out since sometimes they can be a bit hit and miss. There's also a previous issue about FH5 that you might want to peruse through at some point: https://github.com/Kimplul/hid-tmff2/issues/31

Please post the output of `sudo dmesg -w` when plugging in the wheel, I'm interested in seeing if there are any error messages that stop initialization or something along those lines. If everything looks clear, you might want to have a look at running test programs like `evtest`, `fftest` or `jstest`. `evtest` should show input, check if all buttons show up etc. `fftest` is for force feedback, and takes as a command line argument the device to test, so probably something like `fftest /dev/input/by-id/usb-SOMETHING_WITH_THRUSTMASTER-event-joystick`, not entirely certain what the wheel might show up as (assuming it shows up at all).

samsunix, 2024-12-16:
sudo dmesg -w prints horrible spam of 
`NVRM: nvAssertFailedNoLog: Assertion failed: pEventNotificationList->pendingEventNotifyCount == 0 @ event_notification.c:289`

jstest responds to turning wheel and button presses.

```
smx@smxplays:~$ sudo fftest /dev/input/by-id/usb-Thrustmaster_Thrustmaster_Racing_Wheel_FFB-event-joystick
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/by-id/usb-Thrustmaster_Thrustmaster_Racing_Wheel_FFB-event-joystick opened
Features:
  * Absolute axes: X, Y, Z, RZ, Hat 0 X, Hat 0 Y, Misc ,
    [27 00 03 00 00 01 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: Constant, Periodic, Ramp, Spring, Friction, Damper, Rumble, Inertia, Gain, Autocenter, 
    Force feedback periodic effects: Square, Triangle, Sine, Saw up, Saw down, 
    [00 00 00 00 00 00 00 00 00 00 FF 1F 03 00 00 00 ]
  * Number of simultaneous effects: 16

Setting master gain to 75% ... OK
Uploading effect #0 (Periodic sinusoidal) ... OK (id 0)
Uploading effect #1 (Constant) ... OK (id 1)
Uploading effect #2 (Spring) ... OK (id 2)
Uploading effect #3 (Damper) ... OK (id 3)
Uploading effect #4 (Strong rumble, with heavy motor) ... OK (id 4)
Uploading effect #5 (Weak rumble, with light motor) ... OK (id 5)
```

They work! *happy-hopeful face* :)

Kimplul, 2024-12-16:
> NVRM: nvAssertFailedNoLog: Assertion failed: pEventNotificationList->pendingEventNotifyCount == 0 @ event_notification.c:289

Ah, that's something related to NVidia. I guess they're not really known for having the best drivers around, shame that it buries more useful information then.

> They work! happy-hopeful face :)
> (I mean, they are weak af but they do work!)

Oh cool. Interesting that we were that close all along. `fftest` doesn't use 100% force as far as I'm aware, but you might also want to have a look at setting the wheel's gain from Oversteer, could be that the wheel defaults to some low value. https://github.com/berarma/oversteer

Anycase, so now we know buttons and FFB works for stuff outside of Steam. I just recently started looking into an issue that might be related over in https://github.com/Kimplul/hid-tmff2/issues/132, would you mind checking if https://github.com/Grumbel/sdl-jstest shows any input for the wheel?

Steam (or, to be more precise, Proton, which is based on Wine) uses SDL2, which seems to have some issues with (at least some) Thrustmaster wheels at the moment. If `sdl2-jstest` doesn't show any input, it's very likely that you're running into the same issue. If `sdl2-jstest` does work, it might just be FH5 that doesn't know what to do with the wheel, and you might want to try some other games if possible.

samsunix, 2024-12-16:
[This fork](https://github.com/meleu/jstest-sdl) was easier to install but allegedly uses SDL2.
```
Joystick Name:   'Thrustmaster Racing Wheel FFB'
Joystick Number: 0

Axes  6:
   0:      0  [                                                                     #                                                                     ]
   1:      0  [                                                                     #                                                                     ]
   2:      0  [                                                                     #                                                                     ]
   3:      0  [                                                                     #                                                                     ]
   4:      0  [                                                                     #                                                                     ]
   5:      0  [                                                                     #                                                                     ]

Buttons 15:
   0: 0  [ ]
   1: 0  [ ]
   2: 0  [ ]
   3: 0  [ ]
   4: 0  [ ]
   5: 0  [ ]
   6: 0  [ ]
   7: 0  [ ]
   8: 0  [ ]
   9: 0  [ ]
  10: 0  [ ]
  11: 0  [ ]
  12: 1  [#]
  13: 0  [ ]
  14: 0  [ ]

Hats  0:

Balls  0:
```
Downloading WRC2 and DR 2.0 currently. will let you know.

samsunix, 2024-12-17:
just as side note. If one do

`PATH="$PATH:$HOME/.var/app/com.github.Matoking.protontricks/cache/protontricks/proton/Proton\ Experimental/bin" WINEPREFIX=$PWD wine control`

... And pick "game controllers" We can confirm wheel is detected by proton and working (by turning the wheel)

[screenshot](https://imgur.com/a/z7QijPN)

samsunix, 2024-12-17:
Dirt Rally 2.0 Works out of box

Kimplul, 2024-12-18:
Cool! There have been some other wheels that have reused product IDs, it might make sense to implement some kind of direct query and change driver properties based on that over just looking at USB VID/PID as is done now, though kind of annoying that Thrustmaster seems to make things more complicated than strictly necessary.
In any case, I might just start by changing the firmware check be a warning instead of an error, that should make the T-GT II usable by others as well, with some caveats but still. I'll make the change later today, so we should be fairly close to closing this issue. Please feel free to open up new issues if you encounter any problems with the wheel.
I'll also add some notes about the T-GT II, especially about disabling Steam Input.

If you're still interested in helping out with the input issue mentioned earlier in the thread, please open up a new issue with the steps you took that broke your installation.

On a sidenote, GitHub doesn't send out notifications when someone edits a comment, generally you might want to post a new comment when there's some new information and only use edits for typos and stuff like that.

Kimplul, 2024-12-18:
I made the firmware check a warning and added a few notes about the T-GT II to the README. Please have a go at installing the driver from the `main` branch, the change was small enough that I'm not expecting any issues but I like to be safe.

samsunix, 2024-12-20:
Awesome. Merry Christmas and Happy new year.

samsunix, 2025-01-18:
Hi Again, Just reporting in with fresh CachyOS install. 

```
sudo pacman -S linux-headers joyutils dkms
git clone --recurse-submodules https://github.com/Kimplul/hid-tmff2.git
cd hid-tmff2/
sudo ./dkms/dkms-install.sh
sudo make udev-rules
```

+Disable steam input and make sure correct mode is selected, did the trick. 

No additional windows drivers, software or steps needed. 

Tested with Forza 5, Dirt Rally 2.0

Kimplul, 2025-01-20:
Great, thanks for the info. I'll close this issue then, but please do open up new issue if/when you encounter any problems.

closed 2025-01-20 by Kimplul