Kimplul/hid-tmff2 #56: T300 always Autocentering in ACC

kalegood, 2023-02-05:
I’m running this driver, installed via the aur, in arch linux. i have blacklisted hid_thrustmaster. i have installed the windows drivers in the wine prefix. 

In Assetto Corza Competizione, while on race track, wheel auto enters regardless of whether oversteer is open or closed, or whether overdrive autocenter is on/off. 

When in ACC menu, wheel does not autocenter. 

In Project Cars 2, wheel behaves as expected.

kalegood, 2023-02-07:
problem is present in proton 7 and 6.3; i can’t get it to run in other proton versions. problem not present in proton experimental

kalegood, 2023-02-07:
whoops. meant to close.

kalegood, 2023-02-07:
eh, wheel wasn’t properly initiated and i didn’t test enough…. doesn’t work as expected in experimental either.

kalegood, 2023-02-07:
ok, in acc (where there is no autocentering in menus)

if the wheel is centered when selecting “drive” (to being racing session), wheel will autocenter immediately. 

if, however, wheel is not centered when selecting “drive”, wheel will not autocenter immediately. it will, however, begin to autocenter after starting the cars engine (via both keyboard and wheel), shifting gears, or pressing some (but not all) other buttons

Kimplul, 2023-02-08:
Hi, thanks for the report and sorry for the delay, bit of a rough week.

> wheel also auto enters in Wreckfest, as it does in ACC. When in desktop (no game open), wheel auto centers until Overdrive is opened (with autocenter off).

Autocentering on desktop is on purpose, the wheel does that on Windows as well and I've emulated it in the driver. Once a game starts, the wheel goes into a different 'mode' and the desktop autocentering is automatically turned off by the firmware. I'm guessing the idea is that between games the wheel automatically returns to center so you don't accidentally start your next game driving into the nearest ditch.

As for the games, I captured the game's FFB events with `ffbwrap` and ACC only uses two effects, `DAMPER` and `CONSTANT`.

```
...
000005653991 > UPLOAD id:-1 dir:0 length:0 delay:0 type:DAMPER right_saturation:65535 left_saturation:65535 right_coeff:32767 left_coeff:32767 deadband:0 center:-1
000005654000 < 0 id:0
000032122055 > UPLOAD id:-1 dir:16429 length:0 delay:0 type:CONSTANT level:-1 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000032122071 < 0 id:1
000032122133 > PLAY 1 1
000032122137 < 24
000038281869 > PLAY 0 1
000038281889 < 24
...
```

This would imply the autocentering is something the game does automatically by adjusting the `CONSTANT` effect and as such would not be adjustable externally. I couldn't find a setting for it in game.

I haven't tested the game on Windows, so I don't know how it's 'supposed' to feel. Do you have experience with ACC on Windows? Or Wreckfest, for that matter?

berarma, 2023-02-08:
> EDIT: In theory I could likely automatically set the wheel into the game mode, even on desktop. Right now the mode changing is kind of iffy anyway, sometimes closing the game doesn't return the wheel from game mode and I'm not entirely sure why. Could be something to do with device file caching or something in the kernel itself, as the mode switch is triggered on someone calling `open()` and `close()` on the wheel.

I remember DR2 had FFB issues because it was opening the device more than once and closed one of the instances while the game was playing, thus removing all effects. This was fixed in the kernel. The sequence could have been something like `open(1)`, `open(2)`, `close(1)`, playing the game, `close(2)`. In that case, what mode would the wheel be while playing the game?

I think a module parameter to disable this feature would be helpful to discard issues with it.

Kimplul, 2023-02-08:
> I remember DR2 had FFB issues because it was opening the device more than once and closed one of the instances while the game was playing, thus removing all effects. This was fixed in the kernel. [...]

Interesting. The last instance should be the dominant one, so in your case `close(1)` would in theory set the wheel to 'desktop' mode. I'm assuming DR2 means DiRT Rally 2?

berarma, 2023-02-08:
> Interesting. The last instance should be the dominant one, so in your case `close(1)` would in theory set the wheel to 'desktop' mode. I'm assuming DR2 means DiRT Rally 2?

Yes, that's how the error was discovered. In that case, the effects were uploaded before the first close so there were no effects. If the game would have uplodaded the effects after the first close no one would have noticed anything. I'm making assumptions about how it works based on the info I have about that bug.

Kimplul, 2023-02-09:
I added a `open_mode` module parameter, which when set to 0 disables the mode switching and just initializes the wheel into the game mode. `sudo modprobe hid-tmff-new open_mode=0`, basically. However, ACC behaves the same with or without the setting. I'm downloading DiRT Rally 2 at the moment.

berarma, 2023-02-09:
> I added a `open_mode` module parameter, which when set to 0 disables the mode switching and just initializes the wheel into the game mode. `sudo modprobe hid-tmff-new open_mode=0`, basically. However, ACC behaves the same with or without the setting. I'm downloading DiRT Rally 2 at the moment.

Bad call, I was pretty convinced it could work.

ztx-lyghters, 2024-04-09:
Does this mean ACC is unplayable currently and does it affect T300RS? I'm about to buy T300RS for ACC specifically, so I'd appreciate if someone would help me understand tge current state of this problem,

Kimplul, 2024-04-09:
> Does this mean ACC is unplayable currently and does it affect T300RS?

The game runs and you *can* play it, but the experience is nowhere near as good as on Windows. Generally all versions of the T300 behave the same so it's very likely that the RS GT is affected, though I don't have the GT version so I can't say for sure.