Kimplul/hid-tmff2 #57: Update

sayapinsergej, 2023-02-10:


Kimplul, 2023-02-10:
Hello, I've pushed some commits but they've been pretty minor. Why do you ask?

sayapinsergej, 2023-02-10:
well, maybe there were added effects or improvements in the feedback. By the way, I noticed the absence of the effect of losing grip on the tires, is that it?

Kimplul, 2023-02-10:
> well, maybe there were added effects or improvements in the feedback.

No new effects, there was one case where I slightly changed how effect timing worked which might've had some effect, but nothing major, no.

> By the way, I noticed the absence of the effect of losing grip on the tires, is that it?

In which game would this be? I didn't remove it on purpose, I swear :)

sayapinsergej, 2023-02-10:
GranTurismo7

Kimplul, 2023-02-10:
I see. I don't have it myself, could you pinpoint which commit causes the effect to disappear? Of particular interest might be 3953b2e7a24e2cf3926a0038a91acf8d9f9ee887, e2dd7bdc3a22ac8b8551db4154fa36fcc631b402, 0e47624ebbed48e922fe27170364aebf79217e7b and possibly 95680ede70a6c4d292f469b630e3c986620fd16a. Those span a bit more than six months.

sayapinsergej, 2023-02-10:
I'm ready to help, but I don't quite understand what needs to be done

Kimplul, 2023-02-10:
Right, my interpretation of the situation is that you believe the tire grip loss 'effect' to have disappeared in Gran Turismo 7, sometime within the last six months. I'm asking which commit is the last one that still worked for you. I highlighted some commits that could have had an impact, that may be a good starting point.

In practice, look at the commit history of this project on GitHub, run `git checkout <HASH>` on some commit and install that version of the driver. Go through commits until you find the commit where the game still works for you, but the next commit doesn't any more. This should be the commit that introduced the issue, and will hopefully give a better picture of what I might've broken exactly.

sayapinsergej, 2023-02-10:
OK, I'll do it a little later and I'll text you right away!

sayapinsergej, 2023-02-11:
I have tried all versions, there is no tire slippage effect anywhere Wheel Thrustmaster T300

Kimplul, 2023-02-11:
Alright, thanks for checking, but so you actually meant that the effect never existed for you?

sayapinsergej, 2023-02-11:
yes, it is, check with yourself if it's not difficult

Kimplul, 2023-02-11:
> yes, it is, check with yourself if it's not difficult

I'm sorry, but I'm having a hard time understanding you and I don't want to keep guessing at what you mean.

I just realized Gran Turismo 7 doesn't seem to have been ported to PC, so I don't even understand what you checked. Could you please state *very* clearly what the issue you're having is, and what you did to check?

sayapinsergej, 2023-02-11:
The main problem is the lack of tire slippage effect in Gran Turismo 7 on Playstation

Kimplul, 2023-02-11:
Okay, I'm sorry to hear that but this driver is for Linux, on PC. I have no control over what the drivers on PlayStation do.

If you have any issues relevant to Linux, feel free to open up a new issue.

closed 2023-02-11 by Kimplul