Kimplul/hid-tmff2 #60: Hard Crashes with TM 300RS

akleinf, 2023-03-22:
Hello,
with the hid-tmff2-driver most of the ffb-effects were implemented very well. But unfortunately when I race with Grand Prix Legends using this driver the whole system crashes after a while.  Other racing simulations I haven't tested yet. For me, this means that serious online races with GPL cannot be contested. I would be very happy if this problem could be solved. 

My System:
Linux Mint 21.1
Wine 6.0.3
Kernel 5.15.0-58
TM Firmware 34.0 (Package 2023)
Grand Prix Legends v.1.2 (F1 1967 and F1 1969)

Kimplul, 2023-03-22:
Hi, thanks for the report. I'll maybe try to replicate it with the demo, seems to be freely available.

Does `sudo journalctl --boot=-1` show any errors or warning?

akleinf, 2023-03-22:
yes, 2 warnings:
ACPI BIOS Warning (bug): Optional FADT field Pm2ControlBlock has valid
Length but zero Address: 0x0000000000000000/0x1 (20210730/tbfadt-615)
Mär 22 20:05:34 albrecht-System-Product-Name kernel: ACPI BIOS Warning
(bug): 32/64X length mismatch in FADT/Gpe0Block: 64/32 (20210730/tbfadt-564)

akleinf, 2023-03-22:
In the attachment there are more warnings and coloured messages out of the
journal.

Kimplul, 2023-03-24:
Tried downloading the demo, haven't exprienced a full system crash yet. The game itself has crashed a couple times and the text in game seems to be messed up. Could you specify under which situation you encountered the crash? I.e. are you running the game through Lutris, which Wine version are you using, which tracks and cars, etc?

Sidenote, no attachment seems to have come through. Not sure if interacting with GitHub through email supports attachments.

Kimplul, 2023-03-24:
> Sorry for the troubles I caused you.

Not at all, a crash is a pretty serious thing and I appreciate the report.

> The crashes are absolutely unpredictable.

Unfortunate, makes debugging that much more difficult.

Out of curiosity, did you accidentally close the issue or are you implying the system crashes even when not using the wheel? Unpredictability doesn't necessarily mean that this driver isn't the culprit, I could just have some edge case in my concurrent code that only occasionally does something dumb.

Thanks for the attachment, seems to have been included this time. Unfortunately doesn't seem to include anything too relevant to this situation.

akleinf, 2023-03-24:
No, the system crashes occure only while driving GPL and using the TM 300RS with the ffb-driver. Driving GPL without ffb-driver I had and have no problems.

Kimplul, 2023-03-24:
Right, I'd prefer keeping this issue open in that case. I'll do some more digging with your tips and see what happens.

akleinf, 2023-03-28:
Meanwhile I have tested the system for ca. 5 hours successfully without any crash. Actually I wanted to confirm you that everything is alright. But then it happens again! 
Now I have created a new log file from today. The log file I sent you before was accidentally made without the steering wheel plugged in. But now with the TM300 RS plugged in there is a new error message to be seen. I hope this will bring us nearer to the problem.  Thanks for all your efforts.
[Logdatei Kleinfeld-2023-03-28.txt](https://github.com/Kimplul/hid-tmff2/files/11092379/Logdatei.Kleinfeld-2023-03-28.txt)

Kimplul, 2023-03-28:
Seems to be pretty infrequent, interesting.

Thanks for the log file. It seems like you've filtered it somehow, as I would expect there to be a couple messages from this driver after `hid-thrustmaster`.  Could you post a complete log, please?

akleinf, 2023-03-29:
Too bad, the Login-journal from yesterday 17:41 isn't available any longer and the login from 19:29 no longer contains this error, see line 748.
[Logdatei Kleinfeld-2023-03-28-19-29.txt](https://github.com/Kimplul/hid-tmff2/files/11099661/Logdatei.Kleinfeld-2023-03-28-19-29.txt)
In the near future I will continue to observe all journals and let you know if this error occurs again.

akleinf, 2023-03-29:
Here the complete Log-Journal from today with the error-massage again. 
[Logdatei Kleinfeld-2023-03-29-12-03.txt](https://github.com/Kimplul/hid-tmff2/files/11101420/Logdatei.Kleinfeld-2023-03-29-12-03.txt)

Kimplul, 2023-03-31:
Thanks for the full logs, I had a quick look through and noticed that it seems like `hid-generic` picks up the device, as can be seen on line 1894:

> Mär 29 12:03:13 albrecht-System-Product-Name kernel: hid-generic 0003:044F:B66E.0006: input,hidraw4: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:00:12.0-2/input0

This would indicate that this driver isn't even being used for some reason. Did you uninstall the driver at some point, or did you maybe upgrade your kernel? If you installed the driver manually, you need to reinstall it after every kernel upgrade. If you installed the driver with `dkms`, it should've reinstalled the driver for you.

The error message is 'fine', the wheel abruptly restarts itself and the kernel subsystem typically reports an error to the driver about it. The wheel seems to still be initialized, as otherwise it wouldn't be picked up by `hid-generic`.

Here's the lines corresponding lines to the error:

https://github.com/scarburato/hid-tminit/blob/9375f6c7d83af5dd6c8b8fe30351d0f36043b20a/hid-tminit.c#L200-L204

As you can probably imagine, returning an error on an 'ok' action is easily pretty confusing.

Kimplul, 2023-03-31:
For reference, here's what it looks like on my machine when this driver picks up the wheel:

```
[ 4349.960305] usb 1-4: new full-speed USB device number 7 using xhci_hcd
[ 4350.223833] usb 1-4: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[ 4350.223842] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 4350.223845] usb 1-4: Product: Thrustmaster FFB Wheel
[ 4350.223847] usb 1-4: Manufacturer: Thrustmaster
[ 4350.269106] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-4/1-4:1.0/0003:044F:B65D.0006/input/input21
[ 4350.269212] hid-thrustmaster 0003:044F:B65D.0006: input,hidraw5: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:02:00.0-4/input0
[ 4350.286409] hid_tminit: loading out-of-tree module taints kernel.
[ 4350.286444] hid_tminit: module verification failed: signature and/or required key missing - tainting kernel
[ 4350.288778] hid-thrustmaster 0003:044F:B65D.0006: Wheel with model id 0x206 is a Thrustmaster T300RS
[ 4350.289957] usb 1-4: USB disconnect, device number 7
[ 4350.290511] hid-thrustmaster 0003:044F:B65D.0006: URB to change wheel mode seems to have failed with error -2
[ 4350.963950] usb 1-4: new full-speed USB device number 8 using xhci_hcd
[ 4351.294790] usb 1-4: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[ 4351.294798] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 4351.294801] usb 1-4: Product: Thrustmaster T300RS Racing wheel
[ 4351.294803] usb 1-4: Manufacturer: Thrustmaster
[ 4351.314949] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-4/1-4:1.0/0003:044F:B66E.0007/input/input22
[ 4351.315128] hid-generic 0003:044F:B66E.0007: input,hidraw5: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:02:00.0-4/input0
[ 4351.388223] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-4/1-4:1.0/0003:044F:B66E.0007/input/input23
[ 4351.388324] t300rs 0003:044F:B66E.0007: input,hidraw5: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:02:00.0-4/input0
[ 4351.389782] t300rs 0003:044F:B66E.0007: force feedback for T300RS
```

Note the last two lines, with confirmation from this driver that it's being loaded.

akleinf, 2023-03-31:
If I understand you right, it seems the driver may not be properly loaded when Linux is starting. 
But the fact is the driver works mostly well. All feedback effects were given fine to the steering wheel. I would be very satisfied if the system did not crash from time to time. But now, what can I do?
I was using the dkms instruction to install the modul. Meanwhile the kernel had changed twice or more and it was not necessary to reinstall it. I also replaced the AMD Athlon II X2 280 cpu to an AMD FX 6300 and it seems that the crashes occur less often now.
Suggestion: I could install the module on a second and third computer, which are also older, and show you the startup logs from these. Perhaps it is a hardware problem due to the outdated components.
Or do you have any other proposals?

PS: I am afraid I haven't understand completely your remarks in the EDIT. How can I create a better journal? It's a bit awkward to mark, copy and paste the whole output of the terminal to put it in a simple txt.

Kimplul, 2023-03-31:
> If I understand you right, it seems the driver may not be properly loaded when Linux is starting.

Sounds like it, yes. Although if I interpret your comment correctly, it seems like there is some possibility of the message simply being left out?

> I would be very satisfied if the system did not crash from time to time. But now, what can I do?

I understand, and I'm very sorry but so far I haven't been able to reproduce anything. The logs you've provided are greatly appreciated, but they haven't revealed anything to me, and I'm still not sure what the issue could be.

I haven't done any long testing sessions yet, just short ones every now and then when I've had the time, but I might try and reserve some time to try a more continuous test. If the game crashes every five hours, it might be that I've just missed a crash by luck or something. 

So far, your logs have shown that the system has been shut down in a semi-controlled manner, without any obvious kernel error messages. This could be because a lot of different reasons, for example X crashing, but if the logs are only semi-complete, I wouldn't trust on it.

> I was using the dkms instruction to install the modul.

Alright, that's good.

> Suggestion: I could install the module on a second and third computer, which are also older, and show you the startup logs from these. Perhaps it is a hardware problem due to the outdated components.

Yes please, that could be useful.

> PS: I am afraid I haven't understand completely your remarks in the EDIT. How can I create a better journal? It's a bit awkward to mark, copy and paste the whole output of the terminal to put it in a simple txt.

One way could be to run `sudo journalctl --boot=-1 > /tmp/whatever.txt`. This dumps the whole buffer into a file, which you can then include in a comment.

Kimplul, 2023-04-04:
> First I installed the driver with dkms accidentally without pulling out the usb-plug. The crash mentioned above happened then after 30 minutes. Later I removed the driver and installed it again without the T300RS plugged in. Since then no crash happened. But I have to test the system longer. The journal after first installation is here

I don't think having the wheel plugged in would cause a crash, I mainly included that to make sure the driver would get properly loaded.

From the journals I can see that the driver is being loaded, which is good. I have to slightly correct one of my previous comments, I just realized that the sample `journalctl` output I posted should have `t300rs` replaced with `tmff2`. I accidentally had an older version of the driver being loaded through `dkms`. An up to date version of the driver should show up in the `journalctl` log something like
```
[...]
Apr 04 20:19:36 KimiPC kernel: tmff2 0003:044F:B66E.0007: input,hidraw5: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:02:00.0-4/input0
Apr 04 20:19:36 KimiPC kernel: tmff2 0003:044F:B66E.0007: force feedback for T300RS
```

which is essentially what I can see from your logs. Although I still can't see any obvious error messages that would relate to a crash.

Out of curiosity, are you able to switch to a virtual terminal after a crash? `Ctrl + Alt + F1` for example? Because it does sort of seem like the kernel itself might still be alive, as the error logs still show a controlled shutdown.

> Possibly the previous processor (AMD Athlon II X2 280), which is now installed in the PC1, was the cause. The only strange thing is that the same processor with Windows did not produce any crashes for years.

I have to admit I don't understand why a specific processor model would cause such an issue, but the correlation looks like it might be there. Although if the processor is causing the crash, I'm not surprised it worked on Windows, as the code in Windows is completely different than Linux. Having to run apps through an emulation layer etc. probably doesn't help either.

Still haven't sat down for a longer gaming session. Looking forward to seeing if we can replicate the crash on some other processor.

akleinf, 2023-04-05:
Thank you so much. I hope this will be the solution to the riddle. Nevertheless I will continue to test the other computers (PC5 and PC9) and inform you about the result, even if no more crashes should happen.

> are you able to switch to a virtual terminal after a crash? Ctrl + Alt + F1 for example? 

Up till now I only tried the ctrl+alt+del combination to leave the desktop after a crash. Although it seems that no keyboard input is accepted I will try to open a virtual terminal then.

akleinf, 2023-04-15:
I have tested the driver now for many hours but with the FX-Prozessor no further crashes had happened. For me it seems to be clear that the outdated AMD-Prozessor Athlon II X2 280 and not the driver provided the cause.
Sorry I was wrong with my first guess.

closed 2023-04-15 by akleinf