Kimplul/hid-tmff2 #61: T300RS Not Detect in Steam & Games

yacinbm, 2023-03-23:
Just finished successfully installing the driver and I tested my wheel with both `jstest-gdk` and  `oversteer`. The wheel/pedals work in both test suites.

However, when I launch a game on Steam (have tried Assetto Corsa & F1 2019), the games simply do not recognize the wheel. Both games do not show the wheel in the controllers list and opening the controller settings is steam also does not show the wheel. 

I also tried installing the TM drivers via Proton but no dice. Do you have any pointers as to what to try next?

Kimplul, 2023-03-23:
Hello, thanks for the report. Good catch on the `dmesg` output, I would've asked for that first.  Seems to confirm that the driver is indeed being loaded and should be active.

This seems like it might be in the same league as  https://github.com/Kimplul/hid-tmff2/issues/54, as of yet not solved. Have you tried different Proton versions?

Kimplul, 2023-03-24:
> I'm not really sure what to try next. Any suggestions?

I'm afraid I don't have any direct solutions. I have a couple requests to hopefully help debug this below.

> P.S: If it's ok with you, I'll keep this issue going because it's not quite the same issue since even Proton 5.0-10 doesn't work

Yep, all good.

> Small note, I cannot disable steam input for my games, the Controller menu under "My game" > Properties just isn't there.

You mentioned that you're trying to use the wheel on a Steam Deck. I wonder if the Steam Deck has some custom Steam version or something, or possibly some Linux kernel config that messes things up? We had some issues getting the driver to be detected on a steam deck in #48, though that could be entirely unrelated.

Out of curiosity, would you mind trying out
```
fftest /dev/input/by-id/usb-Thrustmaster_Thrustmaster_T300RS_Racing_wheel-event-joystick
```
Try effect zero for instance, should feel like a rumble. If the command works and you get FFB effects we can at least narrow down where the issue might possibly be.

> [...] but this seems to point to Wine not being able to see my wheel.

What happens when you run `wine control`? Proton does use Wine internally, but it was mentioned here https://github.com/Kimplul/hid-tmff2/issues/54#issuecomment-1433853783 that there's some containerisation going on that might prevent Proton from detecting the wheel. If `wine control` detects the wheel, we might be seeing some aspect of the same bug that was supposedly fixed in the Steam Runtime.

yacinbm, 2023-03-24:
>You mentioned that you're trying to use the wheel on a Steam Deck. I wonder if the Steam Deck has some custom Steam version or something, or possibly some Linux kernel config that messes things up?
I was having the same thought, but didn't want to pollute my OS just yet, so I haven't tried installing another more standard version of steam. If all else fails, I'll try this out!

> Out of curiosity, would you mind trying out
>`fftest /dev/input/by-id/usb-Thrustmaster_Thrustmaster_T300RS_Racing_wheel-event-joystick`
The wheel vibrates as expected. 

> What happens when you run wine control? 
A-HA! I didn't have wine installed by itself, as I was using it through Proton as you mentioned. After I installed it through `pacman -Sy wine`, running `wine control` results in:
[image]

Out of curiosity, I tried replacing the wine binaries contained in `Proton 5.0/bin/` by simlinks to the new wine executable I just installed and run `proton control`. The new wine executable did run, but the issue remains. Now I'm not sure how to go about fixing this issue, since I'm not sure if it's related to a Proton behavior, my version of Steam or maybe a permission issue.

I think I'll try installing the Flatpak version of Steam and see if the behavior is the same. I'll report back.

Thanks again for your help!

yacinbm, 2023-03-26:
Finally managed to fix my issue!!! Seems like our intuition about the Steam Deck running a custom version of the Steam clien was right. I tried installing the default Arch Steam client through `pacman -Sy steam`, and pacman indicated that the current installed client was different from the one on the Arch repository (I think it was `steam-jupyter-neptune` or something of the sort).

After installing the client from the pacman repo, I could finally see the Controller sub-menu when inspecting the properties of a game. I disabled Steam Input for Assetto Corsa, **and the wheel is now detected!** 

**TLDR;**
So, for anyone with a similar issue with a racing wheel on the Steam Deck, install the default steam client with `sudo pacman -Sy steam`. Once installed, open your Steam library, right click on the game you wish to play with your racing wheel and click Properties. In the Controller tab, select disable Steam Input, and launch your game. Your racing wheel should now be detected!

@Kimplul This had nothing to do with your driver, thanks so much for your help, and good freaking work, the driver works beautifully :)

closed 2023-03-26 by yacinbm

Kimplul, 2023-03-26:
Heck yeah, always nice to hear good news. Thanks for your patience and cooperation, really appreciate it.

Still unsure if this is related to #54 or not, but reporting this as a bug to Valve might be a good idea. I had a quick look around https://github.com/ValveSoftware/steam-runtime but didn't find anything matching. Not even sure if the issue is with the runtime or some other part, but still. Would you mind opening up an issue over there, reporting steam versions etc?

yacinbm, 2023-03-27:
For future reference, [here](https://github.com/ValveSoftware/steam-runtime/issues/578) is the ticket I opened on the Steam Runtime repo.

yacinbm, 2023-03-29:
Hey @Kimplul, I got the pedals/buttons working in Dirt Rally using the default T300RS game settings, but I get no steering. I get FFB, all the buttons/pedals work, but no steering. When trying to assign steering left/right, moving the wheel is just not recognized. Any idea what could be happening?

Kimplul, 2023-03-31:
Not sure. Just to check, you did rotate the wheel all the way to the left/right and back to centre? Some games seem to require it.

marlonrichert, 2023-08-12:
@yacinbm How did you manage to build `hid-tmff2` on your Steam Deck? I’m unable to make it work with the instructions in the Readme here.