Kimplul/hid-tmff2 #62: Loading hid-tmff with secure boot enabled

SnoutBug, 2023-03-25:
The hid_tmff_new module won't load when secure boot is enabled.

I'm not exactly sure how kernel modules work: Is it correct that you can prevent this by following the optional signing step you described with the manual installation procedure?

Kimplul, 2023-03-26:
> I'm not exactly sure how kernel modules work: Is it correct that you can prevent this by following the optional signing step you described with the manual installation procedure?

Yep, secure boot essentially disables loading untrusted modules. To make a module trusted, you have to sign it.

SnoutBug, 2023-03-26:
>Yep, secure boot essentially disables loading untrusted modules. To make a module trusted, you have to sign it.

Thanks  a bunch.

Signing the module was pretty straight forward with DKMS using [this gist](https://gist.github.com/dojoe/2a9e2dddca982c4f679552fc1ebb18df). The module is now working as intended!

closed 2023-03-26 by SnoutBug