Kimplul/hid-tmff2 #63: t248 with latest firmware but "firmware version 2 is too old, please update."  msg

jmbouzas, 2023-04-10:
Firmware version is checked as if device was a t300rs, but device is recognized as t248 (usb ID).

I cannot find the "t248" branch quoted in some issues.

Attached, dmesg, lsusb, lsmod and other info.

System is debian testing with current kernel.

Module was "manual installed" according to instructions.

[t248.txt](https://github.com/Kimplul/hid-tmff2/files/11193767/t248.txt)

Kimplul, 2023-04-10:
> I cannot find the "t248" branch quoted in some issues.

It's been merged into master a while back, shouldn't be needed.

> Firmware version is checked as if device was a t300rs, but device is recognized as t248 (usb ID).

Which T248 version do you have, Xbox or PS?

> System is debian testing with current kernel.

For future reference, my Debian testing currently says the kernel version is `6.1.0-7-amd64`. In general it's usually preferred to use the full version instead of 'current' or something, but no harm done.

Kimplul, 2023-04-11:
> PS version.

Alright, that's the version I have so I should be able to double check things.

> I think that I selected the PC mode in the mode button, but not completely sure I "confirm" the mode.
> 
> I do not know if related, the hid-tminit kernel module was not automatically load -but hid-tminit.c does not list the t248 model.

I'll have to do some more checking, but it does seem like the wheel is in some mode that I'm not currently handling in the driver.

At least on the T300, PS4 mode skips the `hid-thrustmaster` stage, and the same seems to be happening with your wheel. From the logs you've attached (thanks by the way, very thoughtful) it looks like your T248 is identifying itself with the same ID as the T300 uses in PS4 mode, `b66d`, and the driver treats your wheel as a T300 in PS4 mode. However, the response to the firmware version query seems to be in a different format than what the T300 uses, which would explain the error message at least.

I suspect an immediate fix would be changing the wheel mode, I'll check if I can replicate this on my own.

jmbouzas, 2023-04-15:
> Alright, that's the version I have so I should be able to double check things.
>
>
> I'll have to do some more checking, but it does seem like the wheel is in some mode that I'm not currently handling [...]

Thank you.

With the confirmation of PC mode it works correctly as you wrote.

For the record ,kernel version was:
uname -a
Linux6.1.0-7-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.20-1 (2023-03-19) x86_64 GNU/Linux

Attached lsusb with PS4 mode and PC mode.

Shall I close or rename as something more suitable the non-issue?

Maybe other hints in the log message or other checks to detect "false" USB id in non-PC mode are not worth to include in the driver but maybe a warning in the documentation about double check that device is in PC mode.

I am sorry, it was my fault not confirming  the PC-mode-

Thank you very much for your time and knowledge.

Regards

[t248_2.txt](https://github.com/Kimplul/hid-tmff2/files/11239920/t248_2.txt)

Kimplul, 2023-04-22:
Sorry for the late reply, apparently I missed your latest comment. I've opened up a separate issue to keep track of adding support to other wheel modes with the T248 in https://github.com/Kimplul/hid-tmff2/issues/64, and I'm closing this issue.

closed 2023-04-22 by Kimplul