Kimplul/hid-tmff2 #66: No FFB in BeamNG with T300RS GT

freduard, 2023-05-08:
Been at this trying to install everything for about 6 hours now. Running on little sleep so I might've done something wrong. I finally got the goddamn windows drivers to install. Running BeamNG native app has some basic force feedback but it's just self-centering and there's nothing else (no road bumps, details, etc). Running with ProtonGE the game detects my wheel, although the default assigned binds do nothing. After trying out the native version, it seems ProtonGE version has also lost access to the files or something, because there's no response curve anymore and all the settings were defaulted to 0. Can someone help me diagnose and fix? Losing hope.

Kimplul, 2023-05-08:
> Running BeamNG native app has some basic force feedback but it's just self-centering and there's nothing else (no road bumps, details, etc).

This sounds like what the wheel does with no driver installed. Could you post output of `dmesg` when you plug the wheel in?

freduard, 2023-05-08:
The wheel is listed:
```
[    1.892059] hid-generic 0003:046D:C539.0005: input,hiddev97,hidraw3: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-0000:00:14.0-3/input1
[    1.894060] hid-generic 0003:046D:C539.0007: hiddev98,hidraw5: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-3/input2
[    2.007980] usb 3-9: new full-speed USB device number 5 using xhci_hcd
[    2.134622] usb 3-9: descriptor type invalid, skip
[    2.136687] usb 3-9: New USB device found, idVendor=1038, idProduct=1122, bcdDevice= 2.40
[    2.136699] usb 3-9: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.136704] usb 3-9: Product: SteelSeries KLC
[    2.136708] usb 3-9: Manufacturer: SteelSeries
[    2.138973] hid-generic 0003:1038:1122.0008: hiddev99,hidraw6: USB HID v1.11 Device [SteelSeries SteelSeries KLC] on usb-0000:00:14.0-9/input0
[    2.139422] input: SteelSeries SteelSeries KLC as /devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9:1.1/0003:1038:1122.0009/input/input24
[    2.191416] hid-generic 0003:1038:1122.0009: input,hidraw7: USB HID v1.11 Device [SteelSeries SteelSeries KLC] on usb-0000:00:14.0-9/input1
[    2.304976] usb 3-14: new full-speed USB device number 6 using xhci_hcd
[    2.433677] usb 3-14: New USB device found, idVendor=8087, idProduct=0032, bcdDevice= 0.00
[    2.433692] usb 3-14: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.438709] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:044F:B65D.0001/input/input25
[    2.438754] hid-thrustmaster 0003:044F:B65D.0001: input,hidraw8: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:00:14.0-1/input0
[    2.457884] usb 3-1: USB disconnect, device number 2
[    2.501731] logitech-djreceiver 0003:046D:C539.0004: hidraw2: USB HID v1.11 Keyboard [Logitech USB Receiver] on usb-0000:00:14.0-3/input0
[    2.553890] logitech-djreceiver 0003:046D:C539.0007: hiddev98,hidraw5: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-3/input2
[    2.565809] logitech-djreceiver 0003:046D:C539.0005: hiddev97,hidraw3: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-0000:00:14.0-3/input1
[    2.608180] input: Logitech Wireless Mouse PID:4086 Keyboard as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.2/0003:046D:C539.0007/0003:046D:4086.000A/input/input26
[    2.608752] input: Logitech Wireless Mouse PID:4086 Mouse as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.2/0003:046D:C539.0007/0003:046D:4086.000A/input/input27
[    2.609551] hid-generic 0003:046D:4086.000A: input,hidraw8: USB HID v1.11 Keyboard [Logitech Wireless Mouse PID:4086] on usb-0000:00:14.0-3/input2:1
[    2.697714] input: Logitech G703 LS as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.2/0003:046D:C539.0007/0003:046D:4086.000A/input/input31
[    2.698651] logitech-hidpp-device 0003:046D:4086.000A: input,hidraw8: USB HID v1.11 Keyboard [Logitech G703 LS] on usb-0000:00:14.0-3/input2:1
[    3.080868] usb 3-1: new full-speed USB device number 7 using xhci_hcd
[    3.208039] usb 3-1: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[    3.208043] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.208044] usb 3-1: Product: Thrustmaster T300RS Racing wheel
[    3.208044] usb 3-1: Manufacturer: Thrustmaster
[    3.209891] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:044F:B66E.000B/input/input32
[    3.210051] hid-generic 0003:044F:B66E.000B: input,hidraw9: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:00:14.0-1/input0
[    4.296455] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:044F:B66E.000B/input/input33
[    4.296600] tmff2 0003:044F:B66E.000B: input,hidraw9: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:00:14.0-1/input0
[    4.414615] usbcore: registered new interface driver btusb

```

freduard, 2023-05-08:
I filtered only by USB. I'll get back to you on this tomorrow. Thanks so far!

freduard, 2023-05-09:
Alright, here's all the messages about FFB or T300 I could find. Everything looks sorta OK? It even recognized it as an Alcantara Edition which surprised me.
```
[    2.470491] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:044F:B65D.0001/input/input25
[    2.470528] hid-thrustmaster 0003:044F:B65D.0001: input,hidraw8: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:00:14.0-1/input0
[    2.488634] hid-thrustmaster 0003:044F:B65D.0001: Wheel with model id 0x204 is a Thrustmaster T300 Ferrari Alcantara Edition
[    2.488827] hid-thrustmaster 0003:044F:B65D.0001: Success?! The wheel should have been initialized!
[    2.488856] usb 3-1: USB disconnect, device number 2
[    4.357204] hid_tmff_new: loading out-of-tree module taints kernel.
[    4.357230] hid_tmff_new: module verification failed: signature and/or required key missing - tainting kernel
[    4.360901] Consider using thermal netlink events interface
[    4.406290] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:044F:B66E.000B/input/input33
[    4.406395] tmff2 0003:044F:B66E.000B: input,hidraw9: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:00:14.0-1/input0
[    4.406629] tmff2 0003:044F:B66E.000B: force feedback for T300RS
```

Kimplul, 2023-05-09:
> Alright, here's all the messages about FFB or T300 I could find. [...]

Yep, looks like the driver is being loaded correctly.

> After disconnecting and re-connecting. Are the "USB disconnect, device number x" messages correct even though I only disconnected and reconnected once?

Yep, they're fine. The wheel restarts itself after being initialized by `hid-thrustmaster` which makes it look like a device disconnected. In case you ever help develop a USB device, please don't do this, it is very dumb.

Anycase, looks like BeamNG FFB doesn't work for me with newer Proton versions. Proton 6.3-8 seems to work, did you already try different Proton versions?

freduard, 2023-05-09:
I do get some force feedback now, but again it's only pretty much a strong self-centering force like what happens in BeamNG native which I mentioned in the first comment.

freduard, 2023-05-09:
Alright I reinstalled Fedora and set everything up again. It seems to be better, but still oscillates really badly, even parked. Tried messing around with the settings in Oversteer to no avail. When drifting the wheel also oscillates a-lot mid slide and  it feels like the FFB updates only once or twice a second which causes it to be kind of choppy all the time and randomly catch up to where the wheel's actually supposed to be.

Kimplul, 2023-05-10:
Sounds like it's working about as well as it can work. BeamNG FFB is kind of iffy, there's a number of other reports about the game with some pretty different reports and some hacks to work around issues. I'm guessing you already found https://github.com/Kimplul/hid-tmff2/issues/34?

freduard, 2023-05-18:
I've read it yeah. Sorry for the late response since I've been away from home for a while now but I decided to go back to Windows for now.

Kimplul, 2023-05-18:
Understandable, I'll close this issue then.

closed 2023-05-18 by Kimplul