Kimplul/hid-tmff2 #71: hid-tmff2 in NixOS

RaySlash, 2023-09-08:
In order to use this module in NixOS, one would have to make a derivation for the module. I was going through the process and was able to successfully add it to NixOS. The following file is needed to use the module.
**Module *( hid-tmff2.nix )* :**
```nix
{ stdenv, lib, fetchFromGitHub, kernel, kmod }:

stdenv.mkDerivation rec {
	name = "hid-tmff2-${kernel.version}";
	version = "0.8";

	src = fetchFromGitHub {
		owner = "RaySlash";
		repo = "hid-tmff2";
		rev = "bab1afa2ecf5f9ebdac04d1bb1110f8f3111bb79";
		hash = "sha256-g29vB7uIyFUR4cifOCMtes0BCiEuwPfJQu3eVImj+jM=";
		fetchSubmodules = true;
	};

	nativeBuildInputs = kernel.moduleBuildDependencies ++ [ kmod ];

	makeFlags = kernel.makeFlags ++ [
		"KDIR=${kernel.dev}/lib/modules/${kernel.modDirVersion}/build"
	];

	installFlags = [
		"INSTALL_MOD_PATH=${placeholder "out"}"
	];

	meta = with lib; {
		description = "A linux kernel module for Thrustmaster T300RS and T248";
		homepage = "https://github.com/Kimplul/hid-tmff2";
		license = licenses.gpl3;
		maintainers = [ maintainers.rayslash ];
		platforms = platforms.linux;
	};
}
```

Add this to config to load the module ***( hardware-configuration.nix )* :**
```nix
{ config, pkgs, ... }:

let
    hid-tmff2 = config.boot.kernelPackages.callPackage ./hid-tmff2.nix {};
in
{
    boot.extraModulePackages = [ hid-tmff2 ];
    boot.kernelModules = [ "hid-tmff2" ];
    # Use boot.kernelModules = [ hid-tmff2 ]; to enable the module. Then, the user can load it using:
    # $ sudo modprobe hid-tmff2
    # OR use "" to autoload the module at startup
# Rest of config
};
```
Reference: https://nixos.wiki/wiki/Linux_kernel
Source: https://github.com/RaySlash/hid-tmff2
Linux kernel module for Thrustmaster T300RS and T248 wheels

I had been working around 3 days of my time to find this and sort this out (mostly because im a noob in NixOS). I am using my own fork of the driver as the ``` depmod -A ``` in the Makefile causes the rebuild to crash with error. I have made the change [here](https://github.com/RaySlash/hid-tmff2/commit/bab1afa2ecf5f9ebdac04d1bb1110f8f3111bb79). I hope the OP would add this information for NixOS users in the Wiki or someone can find this Issue to use this.

Kimplul, 2023-09-08:
Hello, thanks a lot, greatly appreciated. I'd be happy to add a page to the wiki outlining integration into different distros.
I think there are some other distros that require drivers to be wrapped in packages/modules/whatever the terminology is, could be useful to have a list of them.

I really only use Debian myself and I don't think I can promise to keep integration stuff up to date, but as long as people open up issues whenever something goes out of date I don't see much of an issue with the idea.

Just checking, but what do you mean that the rebuild crashes with error if `depmod -A` is executed? It was initially added because in some situations the kernel wouldn't immediately probe a newly installed module. Do you know if NixOS automatically refresh kernel modules on install or is there a risk that the behaviour might show up again?

RaySlash, 2023-09-08:
NixOS handles ```depmod``` diferently. It actually does it when the final set of modules are assembled. This would mean, it is something only needed for NixOS afaik. I was having an error such as the one below earlier without that change I made:
```nix
building Nix...
building the system configuration...
these 8 derivations will be built:
  /nix/store/xsq3h167wim46y23crky95s5j5nj9mc7-hid-tmff2-6.1.51.drv
  /nix/store/2sl2afffcjvqv5f9nhc9j5cbgxd05sm0-kernel-modules.drv
  /nix/store/4bbsf4jajjcpxgrfsz8ysfzcw315bgis-linux-6.1.51-modules.drv
  /nix/store/7zpxihzvlhr0zvc6vs37gmdd25krsgay-linux-6.1.51-modules-shrunk.drv
  /nix/store/sxvgk3wg0b5wnmd9gqi1qp1s5q649yjx-stage-1-init.sh.drv
  /nix/store/j2szwpvy2bbcajhc46wc2d985y207f12-initrd-linux-6.1.51.drv
  /nix/store/mwcx67bcxym09dfcwvmc9nksdwg5pn0v-boot.json.drv
  /nix/store/njplhrxcp0z4xdl3xz9l30qnimy95q2h-nixos-system-frost-23.11pre521611.e56990880811.drv
building '/nix/store/xsq3h167wim46y23crky95s5j5nj9mc7-hid-tmff2-6.1.51.drv'...
unpacking sources
unpacking source archive /nix/store/y5s2incxwz9vx5hdymvhpc49k0gx5pm1-source
source root is source
patching sources
updateAutotoolsGnuConfigScriptsPhase
configuring
no configure script, doing nothing
building
build flags: SHELL=/nix/store/r4vxljid3iq94jp7qvd639sps0fscwy3-bash-5.2-p15/bin/bash O=\$\(buildRoot\) CC=/nix/store/civqmfc14c8x7l59nfiwb3qggr8sghrg-gcc-wrapper-12.3.0/bin/cc HOSTCC=/nix/store/civqmfc14c8x7l59nfiwb3qggr8sghrg-gcc-wrapper-12.3.0/bin/cc HOSTLD=/nix/store/cyb4bb44krf4ghw8qasa03sxi2k4g6a4-binutils-wrapper-2.40/bin/ld ARCH=x86_64 KDIR=/nix/store/paa5039bzjz3afd451qah4bm3cx73a2l-linux-6.1.51-dev/lib/modules/6.1.51/build
make -C hid-tminit KDIR="/nix/store/paa5039bzjz3afd451qah4bm3cx73a2l-linux-6.1.51-dev/lib/modules/6.1.51/build"
make[1]: Entering directory '/build/source/hid-tminit'
make -C /nix/store/paa5039bzjz3afd451qah4bm3cx73a2l-linux-6.1.51-dev/lib/modules/6.1.51/build M=/build/source/hid-tminit modules
  CC [M]  /build/source/hid-tminit/hid-tminit.o
  MODPOST /build/source/hid-tminit/Module.symvers
  CC [M]  /build/source/hid-tminit/hid-tminit.mod.o
  LD [M]  /build/source/hid-tminit/hid-tminit.ko
  BTF [M] /build/source/hid-tminit/hid-tminit.ko
Skipping BTF generation for /build/source/hid-tminit/hid-tminit.ko due to unavailability of vmlinux
make[1]: Leaving directory '/build/source/hid-tminit'
make -C /nix/store/paa5039bzjz3afd451qah4bm3cx73a2l-linux-6.1.51-dev/lib/modules/6.1.51/build M=/build/source modules
  CC [M]  /build/source/hid-tmff2.o
  CC [M]  /build/source/hid-tmt300rs.o
  CC [M]  /build/source/hid-tmt248.o
  CC [M]  /build/source/hid-tmtx.o
  LD [M]  /build/source/hid-tmff-new.o
  MODPOST /build/source/Module.symvers
  CC [M]  /build/source/hid-tmff-new.mod.o
  LD [M]  /build/source/hid-tmff-new.ko
  BTF [M] /build/source/hid-tmff-new.ko
Skipping BTF generation for /build/source/hid-tmff-new.ko due to unavailability of vmlinux
installing
install flags: SHELL=/nix/store/r4vxljid3iq94jp7qvd639sps0fscwy3-bash-5.2-p15/bin/bash O=\$\(buildRoot\) CC=/nix/store/civqmfc14c8x7l59nfiwb3qggr8sghrg-gcc-wrapper-12.3.0/bin/cc HOSTCC=/nix/store/civqmfc14c8x7l59nfiwb3qggr8sghrg-gcc-wrapper-12.3.0/bin/cc HOSTLD=/nix/store/cyb4bb44krf4ghw8qasa03sxi2k4g6a4-binutils-wrapper-2.40/bin/ld ARCH=x86_64 KDIR=/nix/store/paa5039bzjz3afd451qah4bm3cx73a2l-linux-6.1.51-dev/lib/modules/6.1.51/build INSTALL_MOD_PATH=/nix/store/d1c7blhiji1clxa33v9by0j7bgi0f6k1-hid-tmff2-6.1.51 install
make -C hid-tminit KDIR="/nix/store/paa5039bzjz3afd451qah4bm3cx73a2l-linux-6.1.51-dev/lib/modules/6.1.51/build" install
make[1]: Entering directory '/build/source/hid-tminit'
make -C /nix/store/paa5039bzjz3afd451qah4bm3cx73a2l-linux-6.1.51-dev/lib/modules/6.1.51/build M=/build/source/hid-tminit modules_install
  INSTALL /nix/store/d1c7blhiji1clxa33v9by0j7bgi0f6k1-hid-tmff2-6.1.51/lib/modules/6.1.51/extra/hid-tminit.ko
  XZ      /nix/store/d1c7blhiji1clxa33v9by0j7bgi0f6k1-hid-tmff2-6.1.51/lib/modules/6.1.51/extra/hid-tminit.ko.xz
  DEPMOD  /nix/store/d1c7blhiji1clxa33v9by0j7bgi0f6k1-hid-tmff2-6.1.51/lib/modules/6.1.51
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/build/source/hid-tminit'
make -C /nix/store/paa5039bzjz3afd451qah4bm3cx73a2l-linux-6.1.51-dev/lib/modules/6.1.51/build M=/build/source modules_install
  INSTALL /nix/store/d1c7blhiji1clxa33v9by0j7bgi0f6k1-hid-tmff2-6.1.51/lib/modules/6.1.51/extra/hid-tmff-new.ko
  XZ      /nix/store/d1c7blhiji1clxa33v9by0j7bgi0f6k1-hid-tmff2-6.1.51/lib/modules/6.1.51/extra/hid-tmff-new.ko.xz
  DEPMOD  /nix/store/d1c7blhiji1clxa33v9by0j7bgi0f6k1-hid-tmff2-6.1.51/lib/modules/6.1.51
Warning: modules_install: missing 'System.map' file. Skipping depmod.
depmod -A
depmod: ERROR: could not open directory /lib/modules/6.1.51: No such file or directory
depmod: ERROR: could not open directory /lib/modules/6.1.51: No such file or directory
depmod: FATAL: could not search modules: No such file or directory
make: *** [Makefile:8: install] Error 1
error: builder for '/nix/store/xsq3h167wim46y23crky95s5j5nj9mc7-hid-tmff2-6.1.51.drv' failed with exit code 2
error: 1 dependencies of derivation '/nix/store/2sl2afffcjvqv5f9nhc9j5cbgxd05sm0-kernel-modules.drv' failed to build
error: 1 dependencies of derivation '/nix/store/4bbsf4jajjcpxgrfsz8ysfzcw315bgis-linux-6.1.51-modules.drv' failed to build
error: 1 dependencies of derivation '/nix/store/njplhrxcp0z4xdl3xz9l30qnimy95q2h-nixos-system-frost-23.11pre521611.e56990880811.drv' failed to build
```

Kimplul, 2023-09-10:
I added a page to the wiki with a reference to this issue. For now I think referencing is the better option, if the information goes out of date I might copy things over and add some note about this issue being out of date. Good to know that NixOS handles the `depmod` automatically.

Thanks again, closing as completed.

closed 2023-09-10 by Kimplul

RaySlash, 2023-09-20:
I have edited the derivation to use your repo and patch to remove the depmod lines so that it stays completely based of github/Kimplul/hid-tmff2. Any user that might want to use this might have to update the ``` rev = "#commit-rev"; ``` if they want a newer revision of the repo.