Kimplul/hid-tmff2 #72: Regarding Licensing of hid-tminit and hid-tmff2

RaySlash, 2023-09-23:
I would like to know further about why did the repository have GPL3 licensing when the [hid-tminit](https://github.com/scarburato/hid-tminit) have GPL2 license. Some files do explicitly say that they use GPL2 as mentioned [here](https://github.com/Kimplul/hid-tmff2/blob/e52d645750a885e44bc3cedab10b54da3626d521/hid-tmff2.c#L1C1-L1C36).
```c
// SPDX-License-Identifier: GPL-2.0
#include <linux/workqueue.h>
```

 I was trying to package this for [nixpkgs](https://github.com/nixos/nixpkgs/). The GPL2 and GPL3 license differs from one another.

According to nix, the GPL3 license seems to be deprecated and recommended to double check with maintainer of hid-tmff2 and hid-tminit. I would greatly appreciate if you can give me some help regarding this!

Kimplul, 2023-09-23:
Honestly, pretty much just an oversight on my part. `hid-tminit` and this project have some common history, `hid-tminit` is based on some code in this repository that scarburato then made more generic. I believe this was before I added a `LICENSE` file to the repository.

In any case, the module itself has always had `MODULE_LICENSE("GPL")` to get access to the HID subsystem. According to the kernel documentation, `"GPL"` means GPL-2.0-or-later, so the `SPDX-License-Identifier` comment should actually be `GPL-2.0-or-later`. I'll push a fix to update the comments. Having the `LICENSE` file be GPL3 complies with this 2-or-later, and I chose GPL3 more or less on a whim (newer is always better, right?). I'm not particularly interested in licensing stuff, and these decisions are honestly pretty much just based on the `MODULE_LICENSE("GPL")` macro that is required to make the driver work.

You edited out that GPL3 was deprecated, I don't really have any issues with using GPL2 if that makes packagers life easier though I'm curious why you edited that bit out?

eclairevoyant, 2023-09-23:
I think the point of confusion on their part was GPL3 _as an identifier_ is deprecated in favour of explicitly saying "-or-later" vs "-only". The license itself is still a valid license afaik.

RaySlash, 2023-09-24:
The only reason why I removed that was when I realised that was the case in for the nix and they have made a change in `` GPL3 `` to `` GPL3-only `` or `` GPL3-or-later `` and was asking me to use the non-deprecated license according to nix, if possible. Sorry, that was a confusion on my part which made me edit the post.

I was trying to communicate with nix team when they referred me this. They have deprecated just `` GPL-3 `` and one can read more about it [here](https://discourse.nixos.org/t/lib-licenses-gpl3-co-are-now-deprecated/8206).

RaySlash, 2023-09-24:
Myself, I am no expert in licensing. I have been trying to gather as much information regarding this. I really appreciate your openness towards users like me who can now play and enjoy these sim racing games with FFB on Linux systems. <3  I see that you have made necessary changes to the ``SPDX-License-Identifier`` already. In this context, If you could help me with whether to choose `` GPL3-only `` , `` GPL3-or-later `` or `` GPL2-or-later `` for the package I would be creating for this repo. I believe that it should be easier to process for reviewers of nix and myself if it was using the same LICENSE throughout the project. Cheers!

eclairevoyant, 2023-09-24:
@RaySlash based on
> the SPDX-License-Identifier comment should actually be GPL-2.0-or-later
I assume that's what they intend for the repo?

Kimplul, 2023-09-24:
> The only reason why I removed that was when I realised that was the case in for the nix and they have made a change in GPL3 to GPL3-only or GPL3-or-later and was asking me to use the non-deprecated [...]

Gotcha, thanks for the explanation.

> I believe that it should be easier to process for reviewers of nix and myself if it was using the same LICENSE throughout the project.

Makes sense.

I looked slightly closer at the GPL licenses, at least the SPDX docs at https://spdx.org/licenses/GPL-2.0-or-later.html show the 'or-later' part as something specified on a per-file basis. Not a lawyer, but to me this sounds like for each file you could say "you can use the code in this file in a project distributed with GPL-2.0-or-later", and with this project being GPL3 the requirement is fulfilled. There's no separate 'GPL-2.0-or-later' `LICENSE` file one could use, as far as I can tell. Still, I can switch the `LICENSE` file over to GPL2 if that makes things simpler?

> > the SPDX-License-Identifier comment should actually be GPL-2.0-or-later
>
> I assume that's what they intend for the repo?

I guess? The main point of truth here (in my opinion) is the `MODULE_LICENSE` macro, which specifies `GPL-2.0-or-later`, so I think `GPL-2.0-or-later` for the package would be the most fitting.

I realize this 'or-later' business is complicating things a bit, and using GPL2/3-only would probably be simpler. I don't have any strong opinions on GPL2 vs GPL3 vs possible future versions, but I know some people have and I would like to keep open the possibility of people extending this driver with their preferred version of the license. Sorry for the extra headache.

RaySlash, 2023-09-24:
No worries. That seems to be enough information regarding the license for now. Ill be following the ``GPL-2-or-later`` which would effectively mean that the repository can stay ``GPL-3``. I believe that should do it for now. Ill comment here back in case of any queries. Once again, Thank you for your response. 😄 ❤️

closed 2023-09-24 by RaySlash