Kimplul/hid-tmff2 #75: T128

Sylv1pons, 2023-11-16:
[image]
Lsusb for T128

Kimplul, 2023-11-16:
Hello, I assume you would like this driver to support the T128? There's already an open issue about it, see https://github.com/Kimplul/hid-tmff2/issues/67

I'll close this issue but feel free to participate over in #67. Out of curiosity, are you attempting to set up GIMX on a raspberry pi or something, or why are you connecting to Linux through putty?

By the way, posting screenshots/pictures of screens of text is usually frowned upon, please consider copypasting the text instead. Helps people who interact through email and so on.

closed 2023-11-16 by Kimplul

Sylv1pons, 2023-11-16:
I use Batocera on x86_64 on V39 with the steering wheel

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>