Kimplul/hid-tmff2 #76: Can we have a driver linux for the T128 Thrustmaster

Sylv1pons, 2023-11-27:
Please we want a linux driver for the Thrustmaster T128.
Thank you

Kimplul, 2023-11-27:
I would love to add support for the T128, but I don't own one, so I can't. Instead, I'm relying on someone in the community who owns one to come along and do the data captures/debugging/testing on their wheel, and then send a pull request. @Nighthawk604 offered to help with this over in #67, which was very nice, but seems to have lost interest or hasn't had time or something. Things happen, and that's fine. Feel free to have a crack at it, though.

This is the second issue you've opened about the T128, please keep the discussion in #67. Unnecessary issues clutter things up and make it more difficult to follow the discussion on a topic, so I'm going to close this.

closed 2023-11-27 by Kimplul

Sylv1pons, 2023-11-28:
OK.
I just want to help because i got a this steering wheel and at this time is not supported on batocera.
How can i help ? A friend talk to me about infos i could give from "lsusb" given by ssh.
Is it something useful ?
Sincerely

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

Kimplul, 2023-11-28:
Glad to hear you're interested in helping. Seeing that #67 hasn't really gone anywhere, the stuff I outlined in https://github.com/Kimplul/hid-tmff2/issues/67#issuecomment-1555889830 is still what would need to be done. There's also some docs related to this in https://github.com/Kimplul/hid-tmff2/wiki#how-to-add-in-support-for-a-new-t-series-wheel. I'm afraid the lsusb output isn't particularly useful at the moment, but good to know for future reference.