Kimplul/hid-tmff2 #78: Thrustmaster TMX Support

pastaq, 2023-12-11:
Hi. I'm working on Thurstmaster TMX support. Using the existing TX model as a baseline, and https://github.com/emtek995/TMX-driver to init the two state startup, I have got the module to load and detect the TMX. This has allowed me to load a corrected Report Descriptor and it also seems to enable auto-centering. So far I have not been able to play any FFB effects using fftest. 

```
fftest /dev/input/event7                                                                                                   ✔ 
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/event7 opened
Features:
  * Absolute axes: X, Y, Z, RZ, Hat 0 X, Hat 0 Y, 
    [27 00 03 00 00 00 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: Constant, Periodic, Ramp, Spring, Friction, Damper, Rumble, Inertia, Gain, Autocenter, 
    Force feedback periodic effects: Square, Triangle, Sine, Saw up, Saw down, 
    [00 00 00 00 00 00 00 00 00 00 FF 1F 03 00 00 00 ]
  * Number of simultaneous effects: 16

Setting master gain to 75% ... OK
Uploading effect #0 (Periodic sinusoidal) ... OK (id 0)
Uploading effect #1 (Constant) ... OK (id 1)
Uploading effect #2 (Spring) ... OK (id 2)
Uploading effect #3 (Damper) ... OK (id 3)
Uploading effect #4 (Strong rumble, with heavy motor) ... OK (id 4)
Uploading effect #5 (Weak rumble, with light motor) ... OK (id 5)
Enter effect number, -1 to exit
0
Now Playing: Sine vibration
Enter effect number, -1 to exit
1
Now Playing: Constant Force
Enter effect number, -1 to exit
2
Now Playing: Spring Condition
Enter effect number, -1 to exit
3
Now Playing: Damping Condition
Enter effect number, -1 to exit
4
Now Playing: Strong Rumble
Enter effect number, -1 to exit
5
Now Playing: Weak Rumble
Enter effect number, -1 to exit
-1
Stopping effects
```

Everything is detected but so far but nothing plays. 

I did some USBPcap captures in windows of each of the FFB effects in the config tool as well as changing configs for gain.

[tmx_ff_cap.zip](https://github.com/Kimplul/hid-tmff2/files/13641848/tmx_ff_cap.zip)

Current status is in my fork's main: https://github.com/pastaq/hid-tmff2/
What would you recommend as a next step? The TX and TMX devices seem very similar based on how they init and the needed Report Descriptor modifications. Reading through #48 didn't  provide any insight on where I need to go next.

Some more data:

# lsmod | grep tm
```
hid_tmff_new           73728  0
tmx_driver             16384  0
```
# udevadm info --attribute-walk --path=$(udevadm info --query=path --name=/dev/input/event7)
```
   looking at device '/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.1/3-6.1:1.0/0003:044F:B67F.0005/input/input37/event7':
    KERNEL=="event7"
    SUBSYSTEM=="input"
    DRIVER==""
[... 523 lines cut ...]
[ 1622.084659] usb 3-6.1: Product: Thrustmaster TMX GIP Racing Wheel
[ 1622.084663] usb 3-6.1: Manufacturer: Thrustmaster
[ 1622.084667] usb 3-6.1: SerialNumber: 000055AEDE480218
[ 1622.179675] probing for wheel
[ 1622.190680] Error maybe?
[ 1622.276480] usb 3-6.1: USB disconnect, device number 7
[ 1622.276552] disconnect the wheel
[ 1622.994927] usb 3-6.1: new full-speed USB device number 8 using xhci_hcd
[ 1623.104629] usb 3-6.1: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[ 1623.104638] usb 3-6.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1623.104643] usb 3-6.1: Product: Thrustmaster FFB Wheel
[ 1623.104647] usb 3-6.1: Manufacturer: Thrustmaster
[ 1623.176713] probing for wheel
[ 1623.181664] Error maybe?
[ 1623.300339] usb 3-6.1: USB disconnect, device number 8
[ 1623.348141] disconnect the wheel
[ 1624.018106] usb 3-6.1: new full-speed USB device number 9 using xhci_hcd
[ 1624.127613] usb 3-6.1: New USB device found, idVendor=044f, idProduct=b67f, bcdDevice= 1.00
[ 1624.127622] usb 3-6.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1624.127627] usb 3-6.1: Product: Thrustmaster TMX Racing Wheel
[ 1624.127631] usb 3-6.1: Manufacturer: Thrustmaster
[ 1624.201076] input: Thrustmaster Thrustmaster TMX Racing Wheel as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-6/3-6.1/3-6.1:1.0/0003:044F:B67F.000C/input/input38
[ 1624.201285] tmff2 0003:044F:B67F.000C: input,hidraw4: USB HID v1.11 Joystick [Thrustmaster Thrustmaster TMX Racing Wheel] on usb-0000:07:00.3-6.1/input0
[ 1624.226844] tmff2 0003:044F:B67F.000C: force feedback for TMX
```

Kimplul, 2023-12-12:
Hello, thanks for showing interest in this project.

I had a cursory look at the zip file, and it seems like the TMX and TX use different USB command sets. For example, in file `force_strength_100-0.pcapng`, packet number 39, the leftover capture data is `43 7f [...]`. From the name of the file, I'm guessing this would be setting the gain of the wheel, and from the other packets I would assume that `43` is the 'set force gain command' and `7f` in this case would be the value to set it to. However,  looking at the gain command in my [docs](https://github.com/Kimplul/hid-tmff2/blob/master/docs/FFBEFFECTS.md#ff_gain), the format that the T300 (and by extension, the TX) uses is `60 02 bf [...]`, meaning that it looks like the two wheels speak a different language.

I did notice that the packet looks a lot like what the T150 driver does for its gain setting, seems like the TMX and T150  might share a command set. See here for the T150 gain setting code:
https://github.com/scarburato/t150_driver/blob/bd543aa58c66cea7be76425eb7747de9bce15ea8/hid-t150/forcefeedback.c#L418

If that's the case, adding support for the T150 might allow adding support for the TMX (or vice versa) in a similar way to how T248 and TX wheels use the T300 functionality. Alternatively, you can try patching the T150 driver, would probably take fewer lines of code.

I have some rough outlines of what needs to be done to add support for a new wheel with a different command set, though nobody's done so yet: https://github.com/Kimplul/hid-tmff2/blob/master/docs/CONTRIBUTING.md#how-to-add-in-support-for-a-new-t-series-wheel

pastaq, 2023-12-12:
> Hello, thanks for showing interest in this project.
>
> I had a cursory look at the zip file, and it seems like the TMX and TX use different USB command sets. For example, in file [...]

Thanks for your quick reply. I'll look into how the t150 works and see if I can get something going here. If I'm honest I'd like to see a unification of TM drivers. It's a bit excessive that there are at least three init projects and two ffb projects all doing basically the same thing in radically different ways.

19Topgun93, 2024-06-27:
Hello :)

I'm not a complet Linux noob, but I don't understand the hid-timinit.c. How can I init my TMX Wheel to funtion properly. I have install the Drivers as it says here on Github, but it is only response as FFB Wheel. In F1 2020 I have the Wheel, but it says disconnected. I have also run proton tricks to install the windows driver in my steam prefix. 

I would love to has assist. I want learn this :D

Kimplul, 2024-06-27:
@19Topgun93 The TMX works with the T150 driver over in https://github.com/scarburato/t150_driver, try installing it and see what happens.

I suppose there's not much interest getting the T150/TMX wheel(s) added to this project since they already seem to work with the T150 driver, so I think it's fine to close this issue. Would still be cool if someone with access to a T150 looked into it.

closed 2024-06-27 by Kimplul

pastaq, 2024-06-27:
> Would still be cool if someone with access to a T150 looked into it.
Still on my to-do list, along with getting init done in hid-thrustmaster. Once I'm at a good place with my other input related project I'll be looking into it again. A single unified thrustmaster driver would be really nice.

19Topgun93, 2024-06-28:
Can I help? I have my TMX Wheel and Windows as dual boot. I have learned in my work python and javascript, but I'm not a pro. Can I assist you somewhere @pastaq ?

pastaq, 2024-06-28:
> Can I help? I have my TMX Wheel and Windows as dual boot. I have learned in my work python and javascript, but I'm not a pro. Can I assist you somewhere @pastaq ?

I'll ping you for testing when I'm ready for that. The data is available for everything, it just needs to be brought over.