Kimplul/hid-tmff2 #82: Instalation issue

Pirnox, 2024-02-01:
Starting the installation according to the instructions, unfortunately make returns an error about a folder that does not exist
``` ~/hid-tmff2   make
make -C deps/hid-tminit KDIR="/lib/modules/6.7.2-arch1-1/build"
make[1]: Directory entry '/home/pirnox/hid-tmff2/deps/hid-tminit'
make -C /lib/modules/6.7.2-arch1-1/build M=/home/pirnox/hid-tmff2/deps/hid-tminit modules
make[2]: Directory entry '/home/pirnox/hid-tmff2/deps/hid-tminit'
make[2]: *** /lib/modules/6.7.2-arch1-1/build: There is no such file or directory. Stop.
make[2]: Leaving the directory '/home/pirnox/hid-tmff2/deps/hid-tminit'
make[1]: *** [Makefile:5: all] Error 2
make[1]: Leaving the directory '/home/pirnox/hid-tmff2/deps/hid-tminit'
make: *** [Makefile:16: deps/hid-tminit] Error 2 ```
After looking deeper into the matter, the problem most likely involves different versions of linux-headers. I can't install the package are you able to help?

 ``` /l/modules  ls -ls
0 drwxr-xr-x 1 root root 518 02-01 13:20 6.7.2-arch1-2/ ```

Kimplul, 2024-02-01:
Compiling kernel modules needs the correct kernel headers, and even if you compile the module against the wrong headers Linux will refuse to load it, so the only solution I can offer is to somehow get the correct headers. Sometimes right after updating the system the running kernel might be older than the 'active' headers, in which case rebooting usually synchronizes the headers and kernel.

May I ask why you can't install the header package?

MmAaXx500, 2024-02-01:
My guess is that you updated the kernel and didn't reboot since then. A reboot may fix it.
The Makefile uses `uname -r` to determine the kernel version, and the old one is reported until a reboot.

Pirnox, 2024-02-03:
Right after the system reboot I rumbled to install the module correctly. I also got an ssl error but I have already dealt with that, thanks for your help

closed 2024-02-03 by Kimplul