Kimplul/hid-tmff2 #83: FBB stops working from the moment I launch beamng.

HetYoshiTeam, 2024-02-09:
I have the T300 Thrustmaster wheel base and the driver seems to work for the most part but I have still a tiny issue. If I  launch Beamng (have not tried other games) the FBB that was on the wheel dissapears completly and feels like it's not plugged in. I read a couple other issue reports but can't fully make sense of what i'm supposed to do with my specific error. i'm on arch linux using i3 as window manager and have a dual boot to windows I also have a usb connector thing to plug all my usb's in for the wheel that sometimes doesn't detect my wheel I have noticed.

Dmesg: 
```
[19716.190392] usb 3-2: USB disconnect, device number 16
[19716.190397] usb 3-2.2: USB disconnect, device number 17
[19716.342527] usb 3-2.4: USB disconnect, device number 18
[19732.148862] usb usb3-port2: Cannot enable. Maybe the USB cable is bad?
[19733.045508] usb usb3-port2: Cannot enable. Maybe the USB cable is bad?
[19733.045572] usb usb3-port2: attempt power cycle
[19734.018883] usb usb3-port2: Cannot enable. Maybe the USB cable is bad?
[19734.918830] usb usb3-port2: Cannot enable. Maybe the USB cable is bad?
[19734.918896] usb usb3-port2: unable to enumerate USB device
[19748.785247] usb 3-2: new high-speed USB device number 23 using xhci_hcd
[19748.931840] usb 3-2: New USB device found, idVendor=05e3, idProduct=0608, bcdDevice=85.32
[19748.931846] usb 3-2: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[19748.931848] usb 3-2: Product: USB2.0 Hub
[19748.982917] hub 3-2:1.0: USB hub found
[19748.983339] hub 3-2:1.0: 4 ports detected
[19749.265228] usb 3-2.2: new full-speed USB device number 24 using xhci_hcd
[19749.371350] usb 3-2.2: New USB device found, idVendor=044f, idProduct=b65d, bcdDevice= 1.00
[19749.371356] usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[19749.371358] usb 3-2.2: Product: Thrustmaster FFB Wheel
[19749.371360] usb 3-2.2: Manufacturer: Thrustmaster
[19749.468649] input: Thrustmaster Thrustmaster FFB Wheel as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.3/usb3/3-2/3-2.2/3-2.2:1.0/0003:044F:B65D.004B/input/input93
[19749.468809] hid-thrustmaster 0003:044F:B65D.004B: input,hidraw10: USB HID v1.00 Gamepad [Thrustmaster Thrustmaster FFB Wheel] on usb-0000:06:00.3-2.2/input0
[19749.488347] hid-thrustmaster 0003:044F:B65D.004B: Wheel with model id 0x204 is a Thrustmaster T300 Ferrari Alcantara Edition
[19749.492394] hid-thrustmaster 0003:044F:B65D.004B: Success?! The wheel should have been initialized!
[19749.558557] usb 3-2.4: new full-speed USB device number 25 using xhci_hcd
[19749.669355] usb 3-2.4: New USB device found, idVendor=044f, idProduct=b660, bcdDevice= 1.00
[19749.669364] usb 3-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[19749.669368] usb 3-2.4: Product: T500 RS Gear Shift
[19749.669371] usb 3-2.4: Manufacturer: Thustmaster
[19749.757537] input: Thustmaster T500 RS Gear Shift as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.3/usb3/3-2/3-2.4/3-2.4:1.0/0003:044F:B660.004C/input/input94
[19749.757635] hid-generic 0003:044F:B660.004C: input,hidraw11: USB HID v1.11 Joystick [Thustmaster T500 RS Gear Shift] on usb-0000:06:00.3-2.4/input0
[19749.759216] usb 3-2.2: USB disconnect, device number 24
[19749.962372] usb 3-2: USB disconnect, device number 23
[19749.981901] usb 3-2.4: USB disconnect, device number 25
[19751.171865] usb usb3-port2: Cannot enable. Maybe the USB cable is bad?
[19751.708507] usb 3-2: new high-speed USB device number 27 using xhci_hcd
[19751.858713] usb 3-2: New USB device found, idVendor=05e3, idProduct=0608, bcdDevice=85.32
[19751.858719] usb 3-2: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[19751.858721] usb 3-2: Product: USB2.0 Hub
[19751.894939] hub 3-2:1.0: USB hub found
[19751.895464] hub 3-2:1.0: 4 ports detected
[19751.903638] Error: Driver 'hid-thrustmaster' is already registered, aborting...
[19751.981628] usb 3-2: USB disconnect, device number 27
[19752.251914] usb 3-2: new high-speed USB device number 28 using xhci_hcd
[19752.398839] usb 3-2: New USB device found, idVendor=05e3, idProduct=0608, bcdDevice=85.32
[19752.398845] usb 3-2: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[19752.398848] usb 3-2: Product: USB2.0 Hub
[19752.439645] hub 3-2:1.0: USB hub found
[19752.440088] hub 3-2:1.0: 4 ports detected
[19752.731823] usb 3-2.2: new full-speed USB device number 29 using xhci_hcd
[19752.848357] usb 3-2.2: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[19752.848362] usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[19752.848364] usb 3-2.2: Product: Thrustmaster T300RS Racing wheel
[19752.848366] usb 3-2.2: Manufacturer: Thrustmaster
[19752.924795] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.3/usb3/3-2/3-2.2/3-2.2:1.0/0003:044F:B66E.004D/input/input95
[19752.924985] tmff2 0003:044F:B66E.004D: input,hidraw10: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:06:00.3-2.2/input0
[19752.930351] tmff2 0003:044F:B66E.004D: force feedback for T300RS
[19753.001832] usb 3-2.4: new full-speed USB device number 30 using xhci_hcd
[19753.121354] usb 3-2.4: New USB device found, idVendor=044f, idProduct=b660, bcdDevice= 1.00
[19753.121360] usb 3-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[19753.121362] usb 3-2.4: Product: T500 RS Gear Shift
[19753.121364] usb 3-2.4: Manufacturer: Thustmaster
[19753.218592] input: Thustmaster T500 RS Gear Shift as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.3/usb3/3-2/3-2.4/3-2.4:1.0/0003:044F:B660.004E/input/input96
[19753.218706] hid-generic 0003:044F:B660.004E: input,hidraw11: USB HID v1.11 Joystick [Thustmaster T500 RS Gear Shift] on usb-0000:06:00.3-2.4/input0
[19753.434873] usb 3-2: USB disconnect, device number 28
[19753.434879] usb 3-2.2: USB disconnect, device number 29
[19753.526544] usb 3-2.4: USB disconnect, device number 30
[19753.951804] usb 3-2: new high-speed USB device number 31 using xhci_hcd
[19754.094714] usb 3-2: New USB device found, idVendor=05e3, idProduct=0608, bcdDevice=85.32
[19754.094719] usb 3-2: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[19754.094722] usb 3-2: Product: USB2.0 Hub
[19754.134907] hub 3-2:1.0: USB hub found
[19754.135337] hub 3-2:1.0: 4 ports detected
[19754.418881] usb 3-2.2: new full-speed USB device number 32 using xhci_hcd
[19754.528355] usb 3-2.2: New USB device found, idVendor=044f, idProduct=b66e, bcdDevice= 1.00
[19754.528364] usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[19754.528368] usb 3-2.2: Product: Thrustmaster T300RS Racing wheel
[19754.528371] usb 3-2.2: Manufacturer: Thrustmaster
[19754.620804] input: Thrustmaster Thrustmaster T300RS Racing wheel as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.3/usb3/3-2/3-2.2/3-2.2:1.0/0003:044F:B66E.004F/input/input97
[19754.620975] tmff2 0003:044F:B66E.004F: input,hidraw10: USB HID v1.11 Joystick [Thrustmaster Thrustmaster T300RS Racing wheel] on usb-0000:06:00.3-2.2/input0
[19754.626361] tmff2 0003:044F:B66E.004F: force feedback for T300RS
[19754.698469] usb 3-2.4: new full-speed USB device number 33 using xhci_hcd
[19754.808355] usb 3-2.4: New USB device found, idVendor=044f, idProduct=b660, bcdDevice= 1.00
[19754.808365] usb 3-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[19754.808369] usb 3-2.4: Product: T500 RS Gear Shift
[19754.808372] usb 3-2.4: Manufacturer: Thustmaster
[19754.909672] input: Thustmaster T500 RS Gear Shift as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.3/usb3/3-2/3-2.4/3-2.4:1.0/0003:044F:B660.0050/input/input98
[19754.909834] hid-generic 0003:044F:B660.0050: input,hidraw11: USB HID v1.11 Joystick [Thustmaster T500 RS Gear Shift] on usb-0000:06:00.3-2.4/input0
```

Kimplul, 2024-02-09:
> I have the T300 Thrustmaster wheel base and the driver seems to work for the most part but I have still a tiny issue. If I launch Beamng (have not tried other games) the FBB that was on the wheel dissapears completly and feels like it's not plugged in.

This sounds like what should happen when the wheel is opened by some program. The driver sends a message to the wheel to enable FFB, which disables the autocentering that the wheel has on when not in active use. Have you driven around in BeamNG or just launched it?

As for the `dmesg` log, it kind of looks like the hub  (I assume you mean a hub by 'connector thing'?) might be flakey, products shouldn't be showing up twice when you plug them in. Not sure if the wheel is maybe drawing too much current on startup and the hub resets itself or what the issue might be, but I suggest you don't use it if you can avoid it.

HetYoshiTeam, 2024-03-27:
Also unplugging the hub has helped in detecting my wheel easier but didn't fix the overal issue.

Kimplul, 2024-03-27:
> EDIT: While reading my dmesg and comparing with someone elses that also had some troubles I noticed that I have "[19751.903638] Error: Driver 'hid-thrustmaster' is already registered, aborting..."
Is this something I should worry about?

Sorry, missed your question as GitHub doesn't notify people when you edit a comment. Anycase, that's nothing to worry about, `hid-thrustmaster` is just the in-kernel counterpart of `hid-tminit`, Linux is just reporting that one is being used over the other.

> Also unplugging the hub has helped in detecting my USB's easier but didn't fix the overal issue.

Alright, unfortunate but thanks for the info. Have you already looked at https://github.com/Kimplul/hid-tmff2/issues/68, https://github.com/Kimplul/hid-tmff2/issues/34 or https://github.com/Kimplul/hid-tmff2/issues/66? For whatever reason, BeamNG does not really play nice with FFB. You could maybe try running it via Proton, 6.3-8 seems to generally work but it might be worth it to try a couple different versions.

HetYoshiTeam, 2024-03-31:
Alright so switching to proton 6.3-8 seems top have  fixed the FBB from completely dissapearing so that's good already, I also tried Assetta Corsa Competzione and this seems to work aswell. The only weird feeling I get from the wheel is that it's not 100% realistic with what's happening in the game for example: in both games that I played and different cars it always felt like the same feedback. I'm very happy that i'm getting some FBB after all this time though so thank you already.

Yes I looked into #68 and #66 but don't quite think that #34 is the problem. (I also don't understand everything that is being said there since I am quite new to linux)

Any ideas what the other 'issue' might be?

Kimplul, 2024-03-31:
> that https://github.com/Kimplul/hid-tmff2/issues/34 is the problem. (I also don't understand everything that is being said there since I am quite new to linux)

That's OK, I mostly just included it to show that BeamNG has had other issues with FFB that you might run into even with a 'working' Proton version, probably should've been more clear. Doesn't sound like you're experiencing anything mentioned in that issue, though.

> The only weird feeling I get from the wheel is that it's not 100% realistic with what's happening in the game for example: in both games that I played and different cars it always felt like the same feedback.

Not sure I fully understand what you mean, could you be a bit more specific? Try to include things like, do you mean that all cars feel the exact same in all games you've tried, and what do they feel like? Have you tried tweaking the FFB profile in Oversteer? Is there any specific thing in the game that demonstrates this issue the clearest? Which games, which cars, etc? Anything else you can come up with to help pin-point the issue is useful as well. If you can show what some game does on Windows is different from Linux, that would also be great.

I can understand that game input feeling wrong is frustrating, but it can unfortunately be very difficult to pinpoint.

HetYoshiTeam, 2024-04-01:
I made some videos showcases the exact problem, how do you want me to share them with you? Uploading them into github doesn't work since they are bigger then 10MB.

Kimplul, 2024-04-01:
A YouTube or Google Drive link is fine. If you don't want to share it publically, you can send me an email directly, kimi.h.kuparinen@gmail.com.

HetYoshiTeam, 2024-04-01:
I will send you a mail, in video "windows-1" you can see how far the wheel goes on my windows without beamng being launched, in windows-2 you can see that while launching/loading inthe wheel doesn't give FBB but once loaded in it does (pretty normal stuff). You can also see if I drive the wheel react to what's happening inside of the game. On both OS's I drove a 'straight' line with just using my gas-pedal. On the windows you can see that the wheel react and on the linux nothing happens. I did the same showcase for the linux: wheel had FBB while loading in the game (not normal) and wheel doesn't react on driving normal or crashing.

If there is something that isn't clear let me know and i'll try to explain better.

Kimplul, 2024-04-01:
Thanks, I got your videos. Good idea to record the behaviour.
From what I can tell, it looks like BeamNG doesn't seem to try to activate FFB, meaning the packet that disables the autocentering never gets sent out (mentioned in my first response), which you describe as 'wheel had FFB'. Just so that signals don't get crossed in the future, when I say 'FFB', I mean like bumps, engine rumble, stuff like that, and right now it seems that you're not actually experiencing any FFB, only autocentering. Autocentering is on by default, and turns off only once a game tries to use FFB (the game can optionally enable autocentering again, but I actually can't remember is BeamNG does this).

I'm currently away from home, but I should be able to check if I can replicate the behaviour tomorrow. In the meantime, there are a couple checks you could do:

1. Try running
```
fftest /dev/input/by-id/usb-Thrustmaster_Thrustmaster_T300RS_Racing_wheel-event-joystick
```
(I'm not entirely sure if the Arcantara edition shows up under another name, tab around `/dev/input/by-id/` a bit until you find something similar) You should be prompted for a couple of different effects, the rumble effects are for gamepads and won't work but at least constant force should feel pretty obvious. If that works, at least we know that the driver is working properly.

2. Did you already try different Proton versions? I mentioned that 6.3-8 worked for me, but different versions might work for different systems.
3. From the `dmesg` logs, it looks like you have a shifter attached. Have you tried starting up BeamNG without it attached? Maybe a longshot, but receiving inputs from multiple devices might be confusing the game.

HetYoshiTeam, 2024-04-01:
Oh alright, sorry for the confusion about the 'FBB' before this my wheel didn't do any auto-centering at all so i'm sorry.

**1.** I installed fftest (linuxconsole) since I didn't have this installed and the output shows as follows:
`user@yoshi ~ % fftest /dev/input/by-id/usb-Thrustmaster_Thrustmaster_T300RS_Racing_wheel-event-joystick

Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/by-id/usb-Thrustmaster_Thrustmaster_T300RS_Racing_wheel-event-joystick opened
Features:
  * Absolute axes: X, Y, RZ, Throttle, Hat 0 X, Hat 0 Y,
    [63 00 03 00 00 00 00 00 ]
  * Relative axes:
    [00 00 ]
  * Force feedback effects types:
    Force feedback periodic effects:
    [00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ]
  * Number of simultaneous effects: 0

Uploading effect #0 (Periodic sinusoidal) ... Error:: Function not implemented
Uploading effect #1 (Constant) ... Error: Function not implemented
Uploading effect #2 (Spring) ... Error: Function not implemented
Uploading effect #3 (Damper) ... Error: Function not implemented
Uploading effect #4 (Strong rumble, with heavy motor) ... Error: Function not implemented
Uploading effect #5 (Weak rumble, with light motor) ... Error: Function not implemented
Enter effect number, -1 to exit
1
Now Playing: Constant Force
Enter effect number, -1 to exit`

My wheel didn't react on any of this.

**2.** Haven't tried other protons yet, I will try this tonight.
**3** Same goes for this I will aslo try this tonight.

Kimplul, 2024-04-01:
> Oh alright, sorry for the confusion about the 'FBB' before this my wheel didn't do any auto-centering at all so i'm sorry.

My fault really, should've been more clear.

> 1. I installed fftest (linuxconsole) since I didn't have this installed and the output shows as follows: [...]

Wow, that really shouldn't look like that. The driver should report that all functions are implemented, so it really seems like something is going wrong here way before we even get to BeamNG. What does `dmesg` show when you plug the wheel in, and do any errors show up when you start `fftest`? Which distro are you using?

The two other checks I mentioned are likely irrelevant, as they assume the driver is working properly, which it apparently isn't.

Kimplul, 2024-04-01:
Oh, I have a working theory: I'm guessing you installed the driver with `make install`?  On Linux, drivers are closely tied with the kernel version, so when you install using `make install`, you're really only installing for the current kernel version. When you update your system (and you're on Arch so I presume you run `yay -Syu` or whatever religiously), the kernel is also updated, and the driver for the older kernel version stops being loaded, as we can see from the `dmesg` output.

What you probably want to do is install the driver with the DKMS method, as this will automatically install a new version of the driver whenever you update your kernel. Apologies, I guess I assumed everyone would be familiar with DKMS which was a massive oversight on my part, I'll amend the README.

So, in summary, you should now install with DKMS, check that you see `tmff2` somewhere in `dmesg` when you plug your wheel in, check that `fftest` works, and finally, try out a couple different Proton versions for BeamNG.

HetYoshiTeam, 2024-04-01:
I did know that it would work like that and didn't know what the DKMS exactly did so just chose the first option.
I installed the DKMS version now and restarted my computer since nothing changed when I didn't.

Alright so I did some testing now aswell, the **fftest** still gives  the same errors as before BUT FFB does seem to work.
I tried 2 games "Assetto Corsa Competizione" which worked perfectly (react to driving/crashing etc.), "Beamng" also does this but my wheel just randomly pulls to a certain directions which makes it undrivable.

I'm gonna try different Proton versions now in the hope that it fixed the issue.

Kimplul, 2024-04-01:
Sounds promising. Curious about `fftest`, but I guess that's less relevant than the games. In any case, sounds like you're running into stuff similar to #34. TL;DR: a couple tweaks to the BeamNG's settings you could try include inverting the FFB, turning softlock off and changing the update rate and type.

HetYoshiTeam, 2024-04-02:
Alright so good news, I inverted the wheel in the game itself and that indeed fixed it. I had to do a lot of tweaking of the FBB settings in the game since it felt like I was turning a tank but after that everything worked nice and react nice.

Kimplul, 2024-04-02:
Cheers, thanks for your patience and glad we got things working.

closed 2024-04-02 by Kimplul