Kimplul/hid-tmff2 #85: make failed

Querzion, 2024-03-11:
```
make -C deps/hid-tminit KDIR="/lib/modules/6.7.9-206.fsync.fc39.x86_64/build" 
make[1]: Entering directory '/var/home/querzion/Git/hid-tmff2/deps/hid-tminit'
make -C /lib/modules/6.7.9-206.fsync.fc39.x86_64/build M=/var/home/querzion/Git/hid-tmff2/deps/hid-tminit modules
make[2]: Entering directory '/usr/src/kernels/6.7.9-206.fsync.fc39.x86_64'
  CC [M]  /var/home/querzion/Git/hid-tmff2/deps/hid-tminit/hid-tminit.o
  MODPOST /var/home/querzion/Git/hid-tmff2/deps/hid-tminit/Module.symvers
  CC [M]  /var/home/querzion/Git/hid-tmff2/deps/hid-tminit/hid-tminit.mod.o
  LD [M]  /var/home/querzion/Git/hid-tmff2/deps/hid-tminit/hid-tminit.ko
/bin/sh: line 1: ld: command not found
make[4]: *** [scripts/Makefile.modfinal:56: /var/home/querzion/Git/hid-tmff2/deps/hid-tminit/hid-tminit.ko] Error 127
make[3]: *** [/usr/src/kernels/6.7.9-206.fsync.fc39.x86_64/Makefile:1865: modules] Error 2
make[2]: *** [Makefile:246: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/kernels/6.7.9-206.fsync.fc39.x86_64'
make[1]: *** [Makefile:5: all] Error 2
make[1]: Leaving directory '/var/home/querzion/Git/hid-tmff2/deps/hid-tminit'
make: *** [Makefile:16: deps/hid-tminit] Error 2
```

Querzion, 2024-03-11:
What is usually ld?

Kimplul, 2024-03-11:
`ld` is the linker, a program that replaces references to symbols (functions, global variables) to their addresses. It's an essential part of compilation, and usually any compiler you install will automatically install the linker, so it seems like one or more of your packages are broken for whatever reason. Maybe try reinstalling your compiler?

napaalm, 2024-03-20:
@Querzion I see that you probably have a [Bazzite](https://github.com/ublue-os/bazzite/) installation. I suggest you to open a feature request there to include `ld`.
Note that you still won't be able to build this kernel module because of ublue-os/bazzite#899, but I've already requested to layer all custom kernel modules for steering wheels in the image (see ublue-os/bazzite#901), so eventually we won't need to build them ourselves.

Kimplul, 2024-11-03:
Doing some winter cleaning, this sounds like it can be closed.

closed 2024-11-03 by Kimplul