Kimplul/hid-tmff2 #86: T80 support

TorokLev, 2024-03-30:
Hi, 

I have a Thrustmaster T80 which is a bit different than this.
I'd like to ask for help how to start with building support for T80.
Thank you in advance,
Lev

Kimplul, 2024-03-30:
Hello, there's a wiki section with a brief overview of what needs to be done for a new wheel: https://github.com/Kimplul/hid-tmff2/wiki#how-to-add-in-support-for-a-new-t-series-wheel

Generally Thrustmaster wheels use two different USB command sets, the wheels in this repository are compatible with the T300, and wheels in https://github.com/scarburato/t150_driver are compatible with the T150. After you've done a couple USB captures, it's probably worth checking if the packets match one or the other. If they do, great, you'll probably get your wheel working with a couple of project-dependent tweaks. Otherwise, a more major effort is required, which can probably be discussed if it becomes relevant. I can help read through rhe USB captures.

I'd recommend reading the whole wiki, it's pretty short. Feel free to ask for clarification on anything.

konchunas, 2024-04-04:
You could try switching it into a Gamepad mode to maybe get some mileage, since gamepads are better supported on Linux.
Here is how to do it:
https://ts.thrustmaster.com/download/accessories/manuals/t80rw/t80_wheel_gamepad_modes.pdf

developersdevelopersdevelopersdeveloper, 2024-04-12:
Thrustmaster T80 acquired today. Tested and didn't work by default, but it _does_ work in gamepad mode. 

To enable gamepad mode:
- Unplug
- Hold PS button
- Plug 
- When LEDs blinking, gamepad mode is on
- Map buttons 

Needed to update /etc/udev/rules.d/99-wheels.rules and /userdata/system/configs/emulationstation/es_input.cfg (files provided by two people in #steering-wheels on discord, I can upload them if desired)

Pedals and wheel worked

Batocera v39 on raspberry pi 5

TorokLev, 2024-04-14:
It looks like that evtest prints activity on T80 when switched in Gamepad mode. Thank you! This is great improvements now.

However oversteer doesn't seem to recognize it as a usb device now.

In Wireshark, there only one usb with activitity. But nothing came from T80 device.

Can somebody tell me how I can capture evtest like output in Python?

HungryHedgehog, 2025-05-14:
Has anyone here managed to get it to work? It only gets recognized in gamepad mode for me and there axis get recognized as buttons etc. There does not seem any support anywhere for this

HungryHedgehog, 2025-05-14:
> Thrustmaster T80 acquired today. Tested and didn't work by default, but it _does_ work in gamepad mode.
>
> To enable gamepad mode:
>
>     * Unplug
>
>     * Hold PS button
>
>     * Plug
>
>     * When LEDs [...]

please do I keep running into posts like that where someone claims they solved my issue but then doesnt actually say how